coolsnowwolf/lede #1868: 关于kvm安装于op内的可行性

CirnoSalt, 2019-10-25:
这个问题来源于隔壁某快固件在最近的版本中加入了KVM功能，虽然可用但极其简陋，
基本没有可操作性。
所以KVM是否能同样安装在openwrt上，
为类似J1900等不支持vt-d(无法直通网卡)不适合安装esxi、pve的用户既能拥有OP直装网络性能的同时还能利用起剩余系统资源来虚拟其他系统？
（J1900直接安装op的情况下，测试300M带宽满速下载CPU占用率不超过20%，并且本人的4G内存服务全开也只占用了300M左右，资源过剩严重）

slitilanber, 2019-10-25:
可以，本人就是这样干的。不过openwrt内的qemu源码依赖包不足，缺少部分功能如spice，usb重定向。另外要修改makefile以支持VNC和其他想要的功能。

coolsnowwolf, 2019-10-25:
其实你这样还不如 PVE 里面跑 OP 的 Docker 了

fo2w, 2019-10-25:
@CirnoSalt 我记得esxi, pve里面跑OP, 300M慢速CPU也不会高很多呀

Aes64X, 2019-10-25:
PVE里使用VMXNET3不会高太多的

slitilanber, 2019-10-25:
1.vm极少使用，家里全是linux电脑，装了个windows的vm在openwrt里，以便不时之需。
2.nas硬件，配置低，pve轻量化不够。
3.luci的web系统监控方便。
4.webdav+jellyfin+syncthing+V.P.N+qemu+lxc+docker...，差一点就是完美的nas路由二合一了。

什么时候如果把qemu源码升级到4.1以上的版本，说不定可以把黑的DSM虚拟出来，就更爽了。

CirnoSalt, 2019-10-25:
> 1.vm极少使用，家里全是linux电脑，装了个windows的vm在openwrt里，以便不时之需。
> 2.nas硬件，配置低，pve轻量化不够。
> 3.luci的web系统监控方便。
> 4.webdav+jellyfin+syncthing+V.P.N+qemu+lxc+docker...，差一点就是完美的nas路由二合一了。
> 
> 什么时候如果把qemu源码升级到4.1以上的版本，说不定可以把黑的DSM虚拟出来，就更爽了。

我目前的想法就是在op上虚拟出Windows，因为大部分linux应用有docker就OK了，但还是有部分应用需求只支持Windows，单独购置性能满足于Windows的服务器功耗又太高......
如果大雕这边能把相关功能整合进编译可选项就完美了。

closed 2019-11-02 by CirnoSalt