coolsnowwolf/lede #1878: 未生成mysys_ssl.dir/my_md5.cc.o文件

czb1994, 2019-10-26:
Scanning dependencies of target mysys_ssl
make[6]: Leaving directory '/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mariadb-10.1.35'
make[6]: Entering directory '/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mariadb-10.1.35'
[ 42%] Building CXX object mysys_ssl/CMakeFiles/mysys_ssl.dir/my_sha1.cc.o
[ 42%] Building CXX object mysys_ssl/CMakeFiles/mysys_ssl.dir/my_sha224.cc.o
[ 42%] Building CXX object mysys_ssl/CMakeFiles/mysys_ssl.dir/my_sha256.cc.o
[ 43%] Building CXX object mysys_ssl/CMakeFiles/mysys_ssl.dir/my_sha384.cc.o
[ 43%] Building CXX object mysys_ssl/CMakeFiles/mysys_ssl.dir/my_sha512.cc.o
[ 43%] Building CXX object mysys_ssl/CMakeFiles/mysys_ssl.dir/my_md5.cc.o //在这之后没有生成my_md5.cc.o文件导致下面报错，make clean过了，.config也重新生成过来，路由器型号是ac58u
/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mariadb-10.1.35/mysys_ssl/my_md5.cc: In function 'void md5_result(MD5_CONTEXT*, uchar*)':
/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mariadb-10.1.35/mysys_ssl/my_md5.cc:86:3: error: 'EVP_MD_CTX_cleanup' was not declared in this scope
   EVP_MD_CTX_cleanup(context);
   ^~~~~~~~~~~~~~~~~~
/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mariadb-10.1.35/mysys_ssl/my_md5.cc:86:3: note: suggested alternative: 'EVP_MD_CTX_create'
   EVP_MD_CTX_cleanup(context);
   ^~~~~~~~~~~~~~~~~~
   EVP_MD_CTX_create
/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mariadb-10.1.35/mysys_ssl/my_md5.cc: In function 'void my_md5(uchar*, const char*, size_t)':
/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mariadb-10.1.35/mysys_ssl/my_md5.cc:102:15: error: aggregate 'MD5_CONTEXT md5_context' has incomplete type and cannot be defined
   MD5_CONTEXT md5_context;
               ^~~~~~~~~~~
/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mariadb-10.1.35/mysys_ssl/my_md5.cc: In function 'void my_md5_multi(uchar*, ...)':
/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mariadb-10.1.35/mysys_ssl/my_md5.cc:127:15: error: aggregate 'MD5_CONTEXT md5_context' has incomplete type and cannot be defined
   MD5_CONTEXT md5_context;
               ^~~~~~~~~~~
/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mariadb-10.1.35/mysys_ssl/my_md5.cc: In function 'size_t my_md5_context_size()':
/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mariadb-10.1.35/mysys_ssl/my_md5.cc:140:28: error: invalid application of 'sizeof' to incomplete type 'MD5_CONTEXT {aka evp_md_ctx_st}'
   return sizeof(MD5_CONTEXT);
                            ^
mysys_ssl/CMakeFiles/mysys_ssl.dir/build.make:127: recipe for target 'mysys_ssl/CMakeFiles/mysys_ssl.dir/my_md5.cc.o' failed
make[6]: *** [mysys_ssl/CMakeFiles/mysys_ssl.dir/my_md5.cc.o] Error 1
make[6]: Leaving directory '/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mariadb-10.1.35'
CMakeFiles/Makefile2:1565: recipe for target 'mysys_ssl/CMakeFiles/mysys_ssl.dir/all' failed
make[5]: *** [mysys_ssl/CMakeFiles/mysys_ssl.dir/all] Error 2
make[5]: Leaving directory '/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mariadb-10.1.35'
Makefile:151: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mariadb-10.1.35'
Makefile:522: recipe for target '/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mariadb-10.1.35/.built' failed
make[3]: *** [/home/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mariadb-10.1.35/.built] Error 2
make[3]: Leaving directory '/home/openwrt/lede/feeds/packages/utils/mariadb'
time: package/feeds/packages/mariadb/compile#1.31#0.20#5.36
package/Makefile:107: recipe for target 'package/feeds/packages/mariadb/compile' failed
make[2]: *** [package/feeds/packages/mariadb/compile] Error 2
make[2]: Leaving directory '/home/openwrt/lede'
package/Makefile:103: recipe for target '/home/openwrt/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/openwrt/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/openwrt/lede'
/home/openwrt/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

Kinodi, 2019-10-26:
一样的报错

hcym, 2019-10-26:
看不懂，早上编译了四个机型都正常，其中一个ac58u应该和你的一样架构，也没问题，不过插件很少，只有ss＋和wireguard

czb1994, 2019-10-26:
#.config文件信息
#
# Automatically generated file; DO NOT EDIT.
# OpenWrt Configuration
#
CONFIG_MODULES=y
CONFIG_HAVE_DOT_CONFIG=y
# CONFIG_TARGET_sunxi is not set
# CONFIG_TARGET_apm821xx is not set
# CONFIG_TARGET_ath25 is not set
# CONFIG_TARGET_ar71xx is not set
# CONFIG_TARGET_ath79 is not set
# CONFIG_TARGET_brcm2708 is not set
# CONFIG_TARGET_bcm53xx is not set
# CONFIG_TARGET_brcm47xx is not set
# CONFIG_TARGET_brcm63xx is not set
# CONFIG_TARGET_cns3xxx is not set
# CONFIG_TARGET_octeon is not set
# CONFIG_TARGET_gemini is not set
# CONFIG_TARGET_mpc85xx is not set
# CONFIG_TARGET_imx6 is not set
# CONFIG_TARGET_mxs is not set
# CONFIG_TARGET_adm8668 is not set
# CONFIG_TARGET_adm5120 is not set
# CONFIG_TARGET_xburst is not set
# CONFIG_TARGET_ixp4xx is not set
# CONFIG_TARGET_lantiq is not set
# CONFIG_TARGET_malta is not set
# CONFIG_TARGET_pistachio is not set
# CONFIG_TARGET_mvebu is not set
# CONFIG_TARGET_kirkwood is not set
# CONFIG_TARGET_mediatek is not set
# CONFIG_TARGET_ramips is not set
# CONFIG_TARGET_at91 is not set
# CONFIG_TARGET_rb532 is not set
# CONFIG_TARGET_mcs814x is not set
# CONFIG_TARGET_layerscape is not set
# CONFIG_TARGET_octeontx is not set
# CONFIG_TARGET_oxnas is not set
# CONFIG_TARGET_armvirt is not set
CONFIG_TARGET_ipq40xx=y
# CONFIG_TARGET_ipq806x is not set
# CONFIG_TARGET_au1000 is not set
# CONFIG_TARGET_samsung is not set
# CONFIG_TARGET_arc770 is not set
# CONFIG_TARGET_archs38 is not set
# CONFIG_TARGET_ar7 is not set
# CONFIG_TARGET_omap is not set
# CONFIG_TARGET_uml is not set
# CONFIG_TARGET_zynq is not set
# CONFIG_TARGET_x86 is not set
# CONFIG_TARGET_x86_64 is not set
# CONFIG_TARGET_x86_generic is not set
# CONFIG_TARGET_x86_legacy is not set
# CONFIG_TARGET_x86_geode is not set
# CONFIG_TARGET_MULTI_PROFILE is not set
# CONFIG_TARGET_ipq40xx_Default is not set
# CONFIG_TARGET_ipq40xx_DEVICE_8dev_jalapeno is not set
# CONFIG_TARGET_ipq40xx_DEVICE_asus_map-ac2200 is not set
# CONFIG_TARGET_ipq40xx_DEVICE_avm_fritzbox-4040 is not set
CONFIG_TARGET_ipq40xx_DEVICE_asus_rt-ac58u=y
# CONFIG_TARGET_ipq40xx_DEVICE_asus_rt-acrh17 is not set
# CONFIG_TARGET_ipq40xx_DEVICE_meraki_mr33 is not set
# CONFIG_TARGET_ipq40xx_DEVICE_compex_wpj428 is not set
# CONFIG_TARGET_ipq40xx_DEVICE_engenius_eap1300 is not set
# CONFIG_TARGET_ipq40xx_DEVICE_glinet_gl-b1300 is not set
# CONFIG_TARGET_ipq40xx_DEVICE_hugo_ac1200 is not set
# CONFIG_TARGET_ipq40xx_DEVICE_linksys_ea6350v3 is not set
# CONFIG_TARGET_ipq40xx_DEVICE_netgear_ex6100v2 is not set
# CONFIG_TARGET_ipq40xx_DEVICE_netgear_ex6150v2 is not set
[... 4944 lines cut ...]
# CONFIG_PACKAGE_shadowsocks-libev-ss-server is not set
# CONFIG_PACKAGE_shadowsocks-libev-ss-tunnel is not set
# CONFIG_PACKAGE_sockd is not set
# CONFIG_PACKAGE_socksify is not set
# CONFIG_PACKAGE_spawn-fcgi is not set
# CONFIG_PACKAGE_squid is not set
# CONFIG_PACKAGE_tinyproxy is not set
CONFIG_PACKAGE_uhttpd=y
# CONFIG_PACKAGE_uhttpd-mod-lua is not set
CONFIG_PACKAGE_uhttpd-mod-ubus=y

#
# dial-in/up
#
# CONFIG_PACKAGE_rp-pppoe-common is not set
# CONFIG_PACKAGE_rp-pppoe-relay is not set
# CONFIG_PACKAGE_rp-pppoe-server is not set

#
# tcprelay
#
# CONFIG_PACKAGE_tcpbridge is not set
# CONFIG_PACKAGE_tcpcapinfo is not set
# CONFIG_PACKAGE_tcpliveplay is not set
# CONFIG

czb1994, 2019-10-26:
.config文件信息
`#
# Automatically generated file; DO NOT EDIT.
# OpenWrt Configuration
#
CONFIG_MODULES=y
CONFIG_HAVE_DOT_CONFIG=y
# CONFIG_TARGET_sunxi is not set
# CONFIG_TARGET_apm821xx is not set
# CONFIG_TARGET_ath25 is not set
# CONFIG_TARGET_ar71xx is not set
# CONFIG_TARGET_ath79 is not set
# CONFIG_TARGET_brcm2708 is not set
# CONFIG_TARGET_bcm53xx is not set
# CONFIG_TARGET_brcm47xx is not set
# CONFIG_TARGET_brcm63xx is not set
# CONFIG_TARGET_cns3xxx is not set
# CONFIG_TARGET_octeon is not set
# CONFIG_TARGET_gemini is not set
# CONFIG_TARGET_mpc85xx is not set
# CONFIG_TARGET_imx6 is not set
# CONFIG_TARGET_mxs is not set
# CONFIG_TARGET_adm8668 is not set
# CONFIG_TARGET_adm5120 is not set
# CONFIG_TARGET_xburst is not set
# CONFIG_TARGET_ixp4xx is not set
# CONFIG_TARGET_lantiq is not set
# CONFIG_TARGET_malta is not set
# CONFIG_TARGET_pistachio is not set
# CONFIG_TARGET_mvebu is not set
# CONFIG_TARGET_kirkwood is not set
# CONFIG_TARGET_mediatek is not set
# CONFIG_TARGET_ramips is not set
# CONFIG_TARGET_at91 is not set
# CONFIG_TARGET_rb532 is not set
# CONFIG_TARGET_mcs814x is not set
# CONFIG_TARGET_layerscape is not set
# CONFIG_TARGET_octeontx is not set
# CONFIG_TARGET_oxnas is not set
# CONFIG_TARGET_armvirt is not set
CONFIG_TARGET_ipq40xx=y
# CONFIG_TARGET_ipq806x is not set
# CONFIG_TARGET_au1000 is not set
# CONFIG_TARGET_samsung is not set
# CONFIG_TARGET_arc770 is not set
# CONFIG_TARGET_archs38 is not set
# CONFIG_TARGET_ar7 is not set
# CONFIG_TARGET_omap is not set
# CONFIG_TARGET_uml is not set
# CONFIG_TARGET_zynq is not set
# CONFIG_TARGET_x86 is not set
# CONFIG_TARGET_x86_64 is not set
# CONFIG_TARGET_x86_generic is not set
# CONFIG_TARGET_x86_legacy is not set
# CONFIG_TARGET_x86_geode is not set
# CONFIG_TARGET_MULTI_PROFILE is not set
# CONFIG_TARGET_ipq40xx_Default is not set
# CONFIG_TARGET_ipq40xx_DEVICE_8dev_jalapeno is not set
# CONFIG_TARGET_ipq40xx_DEVICE_asus_map-ac2200 is not set
# CONFIG_TARGET_ipq40xx_DEVICE_avm_fritzbox-4040 is not set
CONFIG_TARGET_ipq40xx_DEVICE_asus_rt-ac58u=y
# CONFIG_TARGET_ipq40xx_DEVICE_asus_rt-acrh17 is not set
# CONFIG_TARGET_ipq40xx_DEVICE_meraki_mr33 is not set
# CONFIG_TARGET_ipq40xx_DEVICE_compex_wpj428 is not set
# CONFIG_TARGET_ipq40xx_DEVICE_engenius_eap1300 is not set
# CONFIG_TARGET_ipq40xx_DEVICE_glinet_gl-b1300 is not set
# CONFIG_TARGET_ipq40xx_DEVICE_hugo_ac1200 is not set
# CONFIG_TARGET_ipq40xx_DEVICE_linksys_ea6350v3 is not set
# CONFIG_TARGET_ipq40xx_DEVICE_netgear_ex6100v2 is not set
# CONFIG_TARGET_ipq40xx_DEVICE_netgear_ex6150v2 is not set
[... 4944 lines cut ...]
# CONFIG_PACKAGE_shadowsocks-libev-ss-server is not set
# CONFIG_PACKAGE_shadowsocks-libev-ss-tunnel is not set
# CONFIG_PACKAGE_sockd is not set
# CONFIG_PACKAGE_socksify is not set
# CONFIG_PACKAGE_spawn-fcgi is not set
# CONFIG_PACKAGE_squid is not set
# CONFIG_PACKAGE_tinyproxy is not set
CONFIG_PACKAGE_uhttpd=y
# CONFIG_PACKAGE_uhttpd-mod-lua is not set
CONFIG_PACKAGE_uhttpd-mod-ubus=y

#
# dial-in/up
#
# CONFIG_PACKAGE_rp-pppoe-common is not set
# CONFIG_PACKAGE_rp-pppoe-relay is not set
# CONFIG_PACKAGE_rp-pppoe-server is not set

#
# tcprelay
#
# CONFIG_PACKAGE_tcpbridge is not set
# CONFIG_PACKAGE_tcpcapinfo is not set
# CONFIG_PACKAGE_tcpliveplay is not set
# CONFIG

czb1994, 2019-10-26:
这个是.config文件
[.config.txt](https://github.com/coolsnowwolf/lede/files/3774490/default.config.txt)

Mattaclp, 2019-10-27:
更新源碼後 樹莓3B+也是出這個錯

make[6]: Leaving directory '/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53_musl/mariadb-10.1.35'
make[6]: Entering directory '/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53_musl/mariadb-10.1.35'
[ 42%] Building CXX object mysys_ssl/CMakeFiles/mysys_ssl.dir/my_md5.cc.o
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53_musl/mariadb-10.1.35/mysys_ssl/my_md5.cc: In function 'void md5_result(MD5_CONTEXT*, uchar*)':
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53_musl/mariadb-10.1.35/mysys_ssl/my_md5.cc:86:3: error: 'EVP_MD_CTX_cleanup' was not declared in this scope
   EVP_MD_CTX_cleanup(context);
   ^~~~~~~~~~~~~~~~~~
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53_musl/mariadb-10.1.35/mysys_ssl/my_md5.cc:86:3: note: suggested alternative: 'EVP_MD_CTX_create'
   EVP_MD_CTX_cleanup(context);
   ^~~~~~~~~~~~~~~~~~
   EVP_MD_CTX_create
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53_musl/mariadb-10.1.35/mysys_ssl/my_md5.cc: In function 'void my_md5(uchar*, const char*, size_t)':
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53_musl/mariadb-10.1.35/mysys_ssl/my_md5.cc:102:15: error: aggregate 'MD5_CONTEXT md5_context' has incomplete type and cannot be defined
   MD5_CONTEXT md5_context;
               ^~~~~~~~~~~
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53_musl/mariadb-10.1.35/mysys_ssl/my_md5.cc: In function 'void my_md5_multi(uchar*, ...)':
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53_musl/mariadb-10.1.35/mysys_ssl/my_md5.cc:127:15: error: aggregate 'MD5_CONTEXT md5_context' has incomplete type and cannot be defined
   MD5_CONTEXT md5_context;
               ^~~~~~~~~~~
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53_musl/mariadb-10.1.35/mysys_ssl/my_md5.cc: In function 'size_t my_md5_context_size()':
/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53_musl/mariadb-10.1.35/mysys_ssl/my_md5.cc:140:28: error: invalid application of 'sizeof' to incomplete type 'MD5_CONTEXT {aka evp_md_ctx_st}'
   return sizeof(MD5_CONTEXT);
                            ^
mysys_ssl/CMakeFiles/mysys_ssl.dir/build.make:127: recipe for target 'mysys_ssl/CMakeFiles/mysys_ssl.dir/my_md5.cc.o' failed
make[6]: *** [mysys_ssl/CMakeFiles/mysys_ssl.dir/my_md5.cc.o] Error 1
make[6]: Leaving directory '/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53_musl/mariadb-10.1.35'
CMakeFiles/Makefile2:1565: recipe for target 'mysys_ssl/CMakeFiles/mysys_ssl.dir/all' failed
make[5]: *** [mysys_ssl/CMakeFiles/mysys_ssl.dir/all] Error 2
make[5]: Leaving directory '/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53_musl/mariadb-10.1.35'
Makefile:151: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53_musl/mariadb-10.1.35'
Makefile:522: recipe for target '/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53_musl/mariadb-10.1.35/.built' failed
make[3]: *** [/home/ubuntu/lede/build_dir/target-aarch64_cortex-a53_musl/mariadb-10.1.35/.built] Error 2
make[3]: Leaving directory '/home/ubuntu/lede/feeds/packages/utils/mariadb'
time: package/feeds/packages/mariadb/compile#0.54#0.18#0.69
package/Makefile:107: recipe for target 'package/feeds/packages/mariadb/compile' failed
make[2]: *** [package/feeds/packages/mariadb/compile] Error 2
make[2]: Leaving directory '/home/ubuntu/lede'
package/Makefile:103: recipe for target '/home/ubuntu/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile' failed
make[1]: *** [/home/ubuntu/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
/home/ubuntu/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

closed 2019-10-27 by coolsnowwolf