coolsnowwolf/lede #1884: 编译报错好像是mysys_ssl和mariadb-10.1.35的问题。

lzx2009vb, 2019-10-26:
mysys_ssl/CMakeFiles/mysys_ssl.dir/build.make:127: recipe for target 'mysys_ssl/CMakeFiles/mysys_ssl.dir/
make[6]: *** [mysys_ssl/CMakeFiles/mysys_ssl.dir/my_md5.cc.o] Error 1
make[6]: Leaving directory '/home/zcb/4.9/lede/build_dir/target-x86_64_musl/mariadb-10.1.35'
CMakeFiles/Makefile2:1565: recipe for target 'mysys_ssl/CMakeFiles/mysys_ssl.dir/all' failed
make[5]: *** [mysys_ssl/CMakeFiles/mysys_ssl.dir/all] Error 2
make[5]: Leaving directory '/home/zcb/4.9/lede/build_dir/target-x86_64_musl/mariadb-10.1.35'
Makefile:151: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/zcb/4.9/lede/build_dir/target-x86_64_musl/mariadb-10.1.35'
Makefile:522: recipe for target '/home/zcb/4.9/lede/build_dir/target-x86_64_musl/mariadb-10.1.35/.built' failed
make[3]: *** [/home/zcb/4.9/lede/build_dir/target-x86_64_musl/mariadb-10.1.35/.built] Error 2
make[3]: Leaving directory '/home/zcb/4.9/lede/feeds/packages/utils/mariadb'
time: package/feeds/packages/mariadb/compile#0.77#0.25#2.48
package/Makefile:107: recipe for target 'package/feeds/packages/mariadb/compile' failed
make[2]: *** [package/feeds/packages/mariadb/compile] Error 2
make[2]: Leaving directory '/home/zcb/4.9/lede'
package/Makefile:103: recipe for target '/home/zcb/4.9/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/zcb/4.9/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zcb/4.9/lede'
/home/zcb/4.9/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

Ts-z, 2019-10-26:
解决了吗，我这编译x64 也是

Her0R0cket, 2019-10-27:
昨天更新openssl之后就有好几包编译到加密部分出问题了,更新到官方源里的10.2.26就没问题了
[mariadb.zip](https://github.com/coolsnowwolf/lede/files/3775565/mariadb.zip)

lzx2009vb, 2019-10-27:
> 昨天更新openssl之后就有好几包编译到加密部分出问题了,更新到官方源里的10.2.26就没问题了
> [mariadb.zip](https://github.com/coolsnowwolf/lede/files/3775565/mariadb.zip)

更换feed下面的package么？刚替换了继续报错。。。

Her0R0cket, 2019-10-27:
更新下feeds吧,lean更新了

closed 2019-10-27 by coolsnowwolf