coolsnowwolf/lede #1885: 树莓派B+上运行不了酸酸乳plus

yuban10703, 2019-10-27:
貌似一直在重启emm
Sun Oct 27 09:56:17 2019 daemon.info pdnsd[12179]: pdnsd-1.2.9b-par starting.
Sun Oct 27 09:56:17 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Oct 27 09:56:17 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Oct 27 09:56:51 2019 user.notice shadowsocksr: ssr redir tcp error.restart!
Sun Oct 27 09:56:52 2019 daemon.warn pdnsd[12179]: Caught signal 15. Exiting.
Sun Oct 27 09:56:52 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Oct 27 09:56:52 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Oct 27 09:56:56 2019 cron.info crond[12480]: crond (busybox 1.30.0) started, log level 5
Sun Oct 27 09:57:00 2019 daemon.info pdnsd[12677]: pdnsd-1.2.9b-par starting.
Sun Oct 27 09:57:01 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Oct 27 09:57:01 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Oct 27 09:57:34 2019 user.notice shadowsocksr: ssr redir tcp error.restart!
Sun Oct 27 09:57:35 2019 daemon.warn pdnsd[12677]: Caught signal 15. Exiting.
Sun Oct 27 09:57:35 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Oct 27 09:57:35 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Oct 27 09:57:38 2019 cron.info crond[12978]: crond (busybox 1.30.0) started, log level 5
Sun Oct 27 09:57:43 2019 daemon.info pdnsd[13175]: pdnsd-1.2.9b-par starting.
Sun Oct 27 09:57:43 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Oct 27 09:57:43 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Oct 27 09:58:17 2019 user.notice shadowsocksr: ssr redir tcp error.restart!
Sun Oct 27 09:58:18 2019 daemon.warn pdnsd[13175]: Caught signal 15. Exiting.
Sun Oct 27 09:58:18 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Oct 27 09:58:18 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Oct 27 09:58:21 2019 cron.info crond[13476]: crond (busybox 1.30.0) started, log level 5
Sun Oct 27 09:58:26 2019 daemon.info pdnsd[13673]: pdnsd-1.2.9b-par starting.
Sun Oct 27 09:58:26 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Oct 27 09:58:26 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses

yuban10703, 2019-10-27:
SSR可以运行
V2RAY不能运行

coolsnowwolf, 2019-10-27:
需要 armv6 的 v2ray

yuban10703, 2019-10-27:
哦哦，我看看😭

closed 2019-10-27 by yuban10703