coolsnowwolf/lede #1887: 编译报错？求救！删了重新拉的码源，两次都一样错

Kinodi, 2019-10-27:
cc1: all warnings being treated as errors
make[5]: *** [objs/src/http/ngx_http_script.o] Error 1
make[5]: Leaving directory `/home/openwrt/lede/build_dir/target-x86_64_musl/nginx-1.12.2'
make[4]: *** [build] Error 2
make[4]: Leaving directory `/home/openwrt/lede/build_dir/target-x86_64_musl/nginx-1.12.2'
make[3]: *** [/home/openwrt/lede/build_dir/target-x86_64_musl/nginx-1.12.2/.built] Error 2
make[3]: Leaving directory `/home/openwrt/lede/feeds/packages/net/nginx'
time: package/feeds/packages/nginx/compile#14.88#1.55#17.75
make[2]: *** [package/feeds/packages/nginx/compile] Error 2
make[2]: Leaving directory `/home/openwrt/lede'
make[1]: *** [/home/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/openwrt/lede'
make: *** [world] Error 2
[image]

culyzhao, 2019-10-27:
差不多的错
`/opt/lede/build_dir/target-mips_24kc_musl/nginx-1.15.8/lua-nginx/src/ngx_http_lua_script.c: In function 'ngx_http_lua_script_add_capture_code':
/opt/lede/build_dir/target-mips_24kc_musl/nginx-1.15.8/lua-nginx/src/ngx_http_lua_script.c:402:18: error: cast between incompatible function types from 'size_t (*)(ngx_http_lua_script_engine_t *)' {aka 'unsigned int (*)(struct <anonymous> *)'} to 'void (*)(ngx_http_lua_script_engine_t *)' {aka 'void (*)(struct <anonymous> *)'} [-Werror=cast-function-type]
     code->code = (ngx_http_lua_script_code_pt)
                  ^
cc1: all warnings being treated as errors
objs/Makefile:1529: recipe for target 'objs/addon/src/ngx_http_lua_script.o' failed
make[5]: *** [objs/addon/src/ngx_http_lua_script.o] Error 1
make[5]: Leaving directory '/opt/lede/build_dir/target-mips_24kc_musl/nginx-1.15.8'
Makefile:8: recipe for target 'build' failed
make[4]: *** [build] Error 2
make[4]: Leaving directory '/opt/lede/build_dir/target-mips_24kc_musl/nginx-1.15.8'
Makefile:305: recipe for target '/opt/lede/build_dir/target-mips_24kc_musl/nginx-1.15.8/.built' failed
make[3]: *** [/opt/lede/build_dir/target-mips_24kc_musl/nginx-1.15.8/.built] Error 2
make[3]: Leaving directory '/opt/lede/feeds/packages/net/nginx'
time: package/feeds/packages/nginx/compile#46.14#3.27#50.90
package/Makefile:107: recipe for target 'package/feeds/packages/nginx/compile' failed
make[2]: *** [package/feeds/packages/nginx/compile] Error 2
make[2]: Leaving directory '/opt/lede'
package/Makefile:103: recipe for target '/opt/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/opt/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/opt/lede'
/opt/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
`

culyzhao, 2019-10-27:
https://github.com/openresty/lua-nginx-module/issues/1357
GCC7到8的兼容性问题

closed 2019-10-27 by coolsnowwolf