coolsnowwolf/lede #1889: 加入hdparm后编译失败，貌似和gcc8不兼容，还有unrar

happyzhang1995, 2019-10-27:
[image]

Yangtian1971, 2019-10-27:
我也是，不加这个hdparm就可以过，加了openssh也会报错，不加就OK

Mattaclp, 2019-10-27:
我編譯的是32位元的
同樣的錯誤

make[5]: Entering directory '/home/ubuntu/lede/build_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/gcc-8.3.0-initial/zlib'
true "AR_FLAGS=rc" "CC_FOR_BUILD=gcc" "CFLAGS=-O2 -I/home/ubuntu/lede/staging_dir/host/include " "CXXFLAGS=-g -O2" "CFLAGS_FOR_BUILD=" "CFLAGS_FOR_TARGET=-Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result" "INSTALL=/usr/bin/install -c" "INSTALL_DATA=/usr/bin/install -c -m 644" "INSTALL_PROGRAM=/usr/bin/install -c" "INSTALL_SCRIPT=/usr/bin/install -c" "LDFLAGS=-static-libstdc++ -static-libgcc " "LIBCFLAGS=-O2 -I/home/ubuntu/lede/staging_dir/host/include " "LIBCFLAGS_FOR_TARGET=-Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result" "MAKE=make" "MAKEINFO=/home/ubuntu/lede/build_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/gcc-8.3.0/missing makeinfo --split-size=5000000 " "PICFLAG=" "PICFLAG_FOR_TARGET=" "SHELL=/bin/bash" "EXPECT=expect" "RUNTEST=runtest" "RUNTESTFLAGS=" "exec_prefix=/home/ubuntu/lede/staging_dir/toolchain-i386_pentium4_gcc-8.3.0_musl" "infodir=/home/ubuntu/lede/staging_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/share/info" "libdir=/home/ubuntu/lede/staging_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/lib" "prefix=/home/ubuntu/lede/staging_dir/toolchain-i386_pentium4_gcc-8.3.0_musl" "tooldir=/home/ubuntu/lede/staging_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/i486-openwrt-linux-musl" "AR=ar" "AS=as" "CC=gcc" "CXX=g++" "LD=ld" "LIBCFLAGS=-O2 -I/home/ubuntu/lede/staging_dir/host/include " "NM=nm" "PICFLAG=" "RANLIB=ranlib" "DESTDIR=" DO=all multi-do # make
make[5]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/gcc-8.3.0-initial/zlib'
make[5]: Entering directory '/home/ubuntu/lede/build_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/gcc-8.3.0-initial/libbacktrace'
make  all-am
make[6]: Entering directory '/home/ubuntu/lede/build_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/gcc-8.3.0-initial/libbacktrace'
true  DO=all multi-do # make
make[6]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/gcc-8.3.0-initial/libbacktrace'
make[5]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/gcc-8.3.0-initial/libbacktrace'
make[5]: Entering directory '/home/ubuntu/lede/build_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/gcc-8.3.0-initial/libcpp'
test -f config.h || (rm -f stamp-h1 && make stamp-h1)
make[5]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/gcc-8.3.0-initial/libcpp'
make[5]: Entering directory '/home/ubuntu/lede/build_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/gcc-8.3.0-initial/libdecnumber'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/gcc-8.3.0-initial/libdecnumber'
make[5]: Entering directory '/home/ubuntu/lede/build_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/gcc-8.3.0-initial/fixincludes'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/gcc-8.3.0-initial/fixincludes'
make[5]: Entering directory '/home/ubuntu/lede/build_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/gcc-8.3.0-initial/gcc'
g++ -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings  -Wno-unused -DHAVE_CONFIG_H -I. -I. -I/home/ubuntu/lede/build_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/gcc-8.3.0/gcc -I/home/ubuntu/lede/build_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/gcc-8.3.0/gcc/. -I/home/ubuntu/lede/build_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/gcc-8.3.0/gcc/../include -I/home/ubuntu/lede/build_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/gcc-8.3.0/gcc/../libcpp/include -I/home/ubuntu/lede/staging_dir/host/include -I/home/ubuntu/lede/staging_dir/host/include -I/home/ubuntu/lede/staging_dir/host/include  -I/home/ubuntu/lede/build_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/gcc-8.3.0/gcc/../libdecnumber -I/home/ubuntu/lede/build_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/gcc-8.3.0/gcc/../libdecnumber/dpd -I../libdecnumber -I/home/ubuntu/lede/build_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/gcc-8.3.0/gcc/../libbacktrace   -o gimple-match.o -MT gimple-match.o -MMD -MP -MF ./.deps/gimple-match.TPo gimple-match.c
g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Makefile:1110: recipe for target 'gimple-match.o' failed
make[5]: *** [gimple-match.o] Error 4
make[5]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/gcc-8.3.0-initial/gcc'
Makefile:4250: recipe for target 'all-gcc' failed
make[4]: *** [all-gcc] Error 2
make[4]: Leaving directory '/home/ubuntu/lede/build_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/gcc-8.3.0-initial'
Makefile:36: recipe for target '/home/ubuntu/lede/build_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/gcc-8.3.0-initial/.built' failed
make[3]: *** [/home/ubuntu/lede/build_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/gcc-8.3.0-initial/.built] Error 2
[... cut ...]
make[2]: *** [toolchain/gcc/initial/compile] Error 2
make[2]: Leaving directory '/home/ubuntu/lede'
toolchain/Makefile:96: recipe for target '/home/ubuntu/lede/staging_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/stamp/.toolchain_compile' failed
make[1]: *** [/home/ubuntu/lede/staging_dir/toolchain-i386_pentium4_gcc-8.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
/home/ubuntu/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

outemancut, 2019-10-27:
[image]
早知道刚刚不更新源码了 更新了编译不过去了

coolsnowwolf, 2019-10-27:
fixed

closed 2019-10-27 by coolsnowwolf