coolsnowwolf/lede #1894: qt-everywhere-opensource-src-5.13.0.tar.gz下载地址失效

yinzhefuyun, 2019-10-27:
time: package/feeds/packages/mariadb/compile#0.50#0.66#2.35
make[3]: Entering directory '/home/openwrt/lede/package/lean/qt5'
mkdir -p /home/openwrt/lede/dl
SHELL= flock /home/openwrt/lede/tmp/.qt-everywhere-opensource-src-5.13.0.tar.gz.flock -c '      /home/openwrt/lede/scripts/download.pl "/home/openwrt/lede/dl" "qt-everywhere-opensource-src-5.13.0.tar.gz" "d66b1da335d0c25325fdf493e9044c38" "" "http://download.qt-project.org/archive/qt/5.13/5.13.0/single"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure http://download.qt-project.org/archive/qt/5.13/5.13.0/single/qt-everywhere-opensource-src-5.13.0.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:02 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.lede-project.org/qt-everywhere-opensource-src-5.13.0.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   185  100   185    0     0     62      0  0:00:02  0:00:02 --:--:--    62
  0     0    0     0    0     0      0      0 --:--:--  0:00:05 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/qt-everywhere-opensource-src-5.13.0.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:04 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://downloads.openwrt.org/sources/qt-everywhere-opensource-src-5.13.0.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   185  100   185    0     0     50      0  0:00:03  0:00:03 --:--:--    50
  0     0    0     0    0     0      0      0 --:--:--  0:00:07 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.
Makefile:226: recipe for target '/home/openwrt/lede/dl/qt-everywhere-opensource-src-5.13.0.tar.gz' failed
make[3]: *** [/home/openwrt/lede/dl/qt-everywhere-opensource-src-5.13.0.tar.gz] Error 2
make[3]: Leaving directory '/home/openwrt/lede/package/lean/qt5'
time: package/lean/qt5/compile#0.32#0.12#21.26
package/Makefile:107: recipe for target 'package/lean/qt5/compile' failed
make[2]: *** [package/lean/qt5/compile] Error 2
make[2]: Leaving directory '/home/openwrt/lede'
package/Makefile:103: recipe for target '/home/openwrt/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/openwrt/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/openwrt/lede'
/home/openwrt/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

ksong008, 2019-10-27:
qt5 pkgsource错了

zxlhhyccc, 2019-10-27:
> time: package/feeds/packages/mariadb/compile#0.50#0.66#2.35
> make[3]: Entering directory '/home/openwrt/lede/package/lean/qt5'
> mkdir -p /home/openwrt/lede/dl
> SHELL= flock [...]

PKG_SOURCE:=qt-everywhere-src-$(PKG_VERSION).$(PKG_RELEASE).tar.xz
PKG_SOURCE_URL:=http://download.qt-project.org/archive/qt/$(PKG_VERSION)/$(PKG_VERSION).$(PKG_RELEASE)/single
PKG_BUILD_DIR=$(BUILD_DIR)/qt-everywhere-src-$(PKG_VERSION).$(PKG_RELEASE)
PKG_BUILD_PARALLEL:=1
PKG_USE_MIPS16:=0
对比去改吧，不要什么事都找大雕。。。。

closed 2019-10-27 by coolsnowwolf