coolsnowwolf/lede #1900: 用了新源码编译X86_X64固件,刷入后不断重启,无法进入系统

lqhorochi, 2019-10-28:
用了新源码编译X86_X64固件,刷入后不断重启,无法进入系统
无法获取日志,无办法启动到系统

ljk4160, 2019-10-28:
同

ksong008, 2019-10-28:
你们是414固件吗

ljk4160, 2019-10-28:
4.19

ljk4160, 2019-10-28:
[image]

zcluo, 2019-10-28:
同样问题

ksong008, 2019-10-28:
414也是会无限重启

badgv, 2019-10-28:
最新源码在ESXI同样无限重启

gsblay, 2019-10-28:
会不会是因为升级安装分区大小不一致造成的，原固件默认90M大小，新固件默认是160M。
我猜的，还没试。

lqhorochi, 2019-10-28:
> 最新源码在ESXI同样无限重启

貌似是通病???你试下全新去刷入

badgv, 2019-10-28:
> > 最新源码在ESXI同样无限重启
> 
> 貌似是通病???你试下全新去刷入

我全新做了ESXI虚拟机磁盘，等于全新刷入了，一样无限重启

lqhorochi, 2019-10-28:
> > > 最新源码在ESXI同样无限重启
> > 
> > 
> > 貌似是通病???你试下全新去刷入
> 
> 我全新做了ESXI虚拟机磁盘，等于全新刷入了，一样无限重启

这个坑,等大雕来填吧

Aes64X, 2019-10-28:
PVE 4.9.196 内核，EXT4镜像，不勾选 Pad images to filesystem size (for JFFS2)  无限重启，勾选后没事。另，以前EXT4生成256mb大小的镜像正常，现在直接报错root空间不足，我加到512MB进去root占用才129+。。。。。

lqhorochi, 2019-10-28:
> PVE 4.9.196 内核，EXT4镜像，不勾选 Pad images to filesystem size (for JFFS2) 无限重启，勾选后没事。另，以前EXT4生成256mb大小的镜像正常，现在直接报错root空间不足，我加到512MB进去root占用才129+。。。。。

分区问题?必须重新分区了????

lqhorochi, 2019-10-28:
虚拟机上重新写入又可以启动

Aes64X, 2019-10-28:
大雕上手了

coolsnowwolf, 2019-10-28:
fixed

closed 2019-10-28 by coolsnowwolf