coolsnowwolf/lede #1901: 4.19内核编译出错，大神知道是什么原因么？

czjesda, 2019-10-28:
小弟有一个问题咨询一下，我用内核4.9编译L大的代码没有问题，但是我把内核改成4.19后编译就报错了，出错代码如下：
modules/crypto.mk:155: recipe for target '/home/jw/lede/bin/targets/x86/64/packages/kmod-crypto-deflate_4.19.79-1_x86_64.ipk' failed
make[3]: *** [/home/jw/lede/bin/targets/x86/64/packages/kmod-crypto-deflate_4.19.79-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/jw/lede/package/kernel/linux'
time: package/kernel/linux/compile#6.10#2.34#8.54
package/Makefile:107: recipe for target 'package/kernel/linux/compile' failed
make[2]: *** [package/kernel/linux/compile] Error 2
make[2]: Leaving directory '/home/jw/lede'
package/Makefile:103: recipe for target '/home/jw/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/jw/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jw/lede'
/home/jw/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

hcym, 2019-10-28:
轻易别折腾，内核补丁各种不同，性能差异微乎其微

czjesda, 2019-10-28:
> 
> 
> 轻易别折腾，内核补丁各种不同，性能差异微乎其微

好的，感谢，你的意思就是还是用4.9的内核对么？我记得前几个版本都可以改的，貌似L大24日以后的版本就不能改了

hcym, 2019-10-28:
内核没搞过，想用新的就用官方编译，大雕的我改过ss-lib的版本也是以前可以这次不行，老老实实用大雕默认的，也没感觉什么差异，呵呵

coolsnowwolf, 2019-10-28:
fixed

closed 2019-10-28 by coolsnowwolf