coolsnowwolf/lede #1903: SSR使用 绕过中国大陆ip模式经常出现网络故障

z826540272, 2019-10-28:
[image]
按理说 绕过中国大陆ip 模式应该不会再出现dns问题，但还是经常部分网页加载出现问题，无法显示图片内容或者整个网页无法打开，确认不是服务器问题，必须开关wifi才能解决，导致使用体验极差。
系统 Android 9.0。
开关wifi会重置dns缓存。究竟是什么原因导致了系统缓存的dns无法正常使用。

z826540272, 2019-10-28:
这时候查到图片无法加载的域名是
lh3.googleusercontent.com
在手机上的dig软件查询一次这个域名的dns，显示台湾省 Google的ip，然后整个系统对于 play.google.com 下的内容显示立即变得正常

coolsnowwolf, 2019-10-28:
不要编译 ipv6

z826540272, 2019-10-28:
确定没有编译ipv6

hcym, 2019-10-28:
我一直编译ipv6，两种模式都可以，手机更新要看节点和高峰时段，基本正常，用mtk的差点意思，用n1和蜗牛都跑满100M宽带，不过以前版本容易出差

z826540272, 2019-10-28:
和节点没关系，经常遇得到可以打开google.com却打不开google.com.hk
的问题
又或者可以打开google.com.hk却打不开google.com

rufengsuixing, 2019-10-30:
v2ray 域名嗅探几乎解决一切dns问题

closed 2019-11-28 by coolsnowwolf