coolsnowwolf/lede #1904: 求助，qBittorrent编译成功后无法启动。

SORUNCEX, 2019-10-28:
如题，小白初次尝试自编译并成功编译了qbt，但路由器管理页面中找不到。确定有编译luci。
通过ssh中输入qbittorrent-nox也无法开启，提示illegal instruction。
检查过挂载，是成功的。还重新编译了4.1.3的qbt版本，也是同样的问题。
十分感谢。
[image]
[image]
[image]

hong0980, 2019-10-28:
命令启动后别按y你试试，估计打不开WebUI，正常命令要带端口和配置路径。qbittorrent-nox -d --profile=$profile_dir

S0RUNCEX, 2019-10-28:
感谢指出。我尝试了下，如果不按Y的话根本没法进行下一步

尝试加入代码指定端口和指定目录（因为Linux版的并不需要特意加指令，只要打开webui后设置就可以。所以这点不懂。还有如果加-d就没有任何反应）：

> root@OpenWrt:~# qbittorrent-nox -d --profile=/mnt/sda1/qbittorrent
root@OpenWrt:~# qbittorrent-nox -d --webui-port=9999
root@OpenWrt:~# qbittorrent-nox -d --profile=/tmp
root@OpenWrt:~# qbittorrent-nox -d --profile=/mnt/sda1/qbittorrent --webui-port=8888
root@OpenWrt:~# qbittorrent-nox --webui-port=9999
*** Legal Notice ***
qBittorrent is a file sharing program. When you run a torrent, its data will be made available to others by means of upload. Any content you share is your sole responsibility.
No further notices will be issued.
Press 'y' key to accept and continue...
y
Illegal instruction
root@OpenWrt:~# qbittorrent-nox --profile=/mnt/sda1/qbittorrent --webui-port=8888
*** Legal Notice ***
qBittorrent is a file sharing program. When you run a torrent, its data will be made available to others by means of upload. Any content you share is your sole responsibility.
No further notices will be issued.
Press 'y' key to accept and continue...
y
Illegal instruction

虽然还是无法启动，但会创建一个qb的目录，目录中的log文件排除重复部分如下

> qBittorrent v4.1.3 started
qBittorrent is trying to listen on any interface port: 8999
Peer ID: -qB4130-
HTTP User-Agent is 'qBittorrent/4.1.3'
DHT support [ON]
Local Peer Discovery support [ON]
PeX support [ON]
Anonymous mode [OFF]
Encryption support [ON]
Embedded Tracker [OFF]
UPnP / NAT-PMP support [ON]
Couldn't load GeoIP database. Reason: No such file or directory

4.1.7和4.1.3显示的除了数字不同，其余部分相同
我尝试了这里给出的8999端口，我自己设置的8888和9999端口，还有linux版给出的webui的8080端口都无法打开webui，应该是在地理数据那里就终止操作了。

现在还有个问题是为什么没有luci，明明勾选安装了，软件包里也有...
希望能看看咋回事吧，谢谢。

coolsnowwolf, 2019-10-28:
你没有编译 vsftp 的 luci，所以不会出现

S0RUNCEX, 2019-10-28:
........好吧，那我之后重新编译下

还想请教下大雕，因为第一次编译多了些用不到的东西（也少了些要用到的），如果现在只编译这些，还有没有缺少会影响功能的编译项目？

> luci-app-aria2
luci-app-autoreboot
luci-app-firewall
luci-app-flowoffload
luci-app-nlbwmon
luci-app-ramfree
luci-app-qbittorrent
luci-app-samba
[image]
luci-app-transmission
luci-app-ttyd
luci-app-upnp
luci-app-vsftpd
luci-app-wrtbwmon

感谢

S0RUNCEX, 2019-10-28:
还有大雕，能否考虑qbt加一个分支
https://github.com/c0re100/qBittorrent-Enhanced-Edition
是用来屏蔽吸血客户端的，之前BT被吸血吸烦了，换成这个版本了，感觉不错。
谢谢