coolsnowwolf/lede #1911: qt还是编译不过，应该是和ssl有关

zcluo, 2019-10-28:
ssl/qsslsocket_openssl_symbols.cpp:595:23: note: suggested alternative: 'SSL_CTX_set_alpn_select_cb'   
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \                                
                       ^~~~~~~                                                                         
ssl/qsslsocket_openssl_symbols.cpp:1382:5: note: in expansion of macro 'RESOLVEFUNC'                   
     RESOLVEFUNC(SSL_CTX_set_alpn_select_cb)                                                           
     ^~~~~~~~~~~                                                                                       
ssl/qsslsocket_openssl_symbols.cpp:595:11: error: '_q_SSL_get0_alpn_selected' was not declared in this scope                                                                                                  
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \                                
           ^~~                                                                                         
ssl/qsslsocket_openssl_symbols.cpp:1383:5: note: in expansion of macro 'RESOLVEFUNC'                   
     RESOLVEFUNC(SSL_get0_alpn_selected)                                                               
     ^~~~~~~~~~~                                                                                       
ssl/qsslsocket_openssl_symbols.cpp:595:11: note: suggested alternative: 'SSL_get0_alpn_selected'       
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \                                
           ^~~                                                                                         
ssl/qsslsocket_openssl_symbols.cpp:1383:5: note: in expansion of macro 'RESOLVEFUNC'                   
     RESOLVEFUNC(SSL_get0_alpn_selected)                                                               
     ^~~~~~~~~~~                                                                                       
ssl/qsslsocket_openssl_symbols.cpp:595:23: error: '_q_PTR_SSL_get0_alpn_selected' was not declared in this scope                                                                                              
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \                                
                       ^~~~~~~                                                                         
ssl/qsslsocket_openssl_symbols.cpp:1383:5: note: in expansion of macro 'RESOLVEFUNC'                   
     RESOLVEFUNC(SSL_get0_alpn_selected)                                                               
     ^~~~~~~~~~~                                                                                       
ssl/qsslsocket_openssl_symbols.cpp:595:23: note: suggested alternative: 'SSL_get0_alpn_selected'       
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \                                
                       ^~~~~~~                                                                         
ssl/qsslsocket_openssl_symbols.cpp:1383:5: note: in expansion of macro 'RESOLVEFUNC'                   
     RESOLVEFUNC(SSL_get0_alpn_selected)                                                               
     ^~~~~~~~~~~                                                                                       
make[7]: *** [Makefile:26252: .obj/qsslsocket_openssl_symbols.o] Error 1                               
make[7]: Leaving directory '/root/lede/build_dir/target-x86_64_musl/qt-everywhere-src-5.13.1/qtbase/src/network'                                                                                              
make[6]: *** [Makefile:273: sub-network-make_first] Error 2                                            
make[6]: Leaving directory '/root/lede/build_dir/target-x86_64_musl/qt-everywhere-src-5.13.1/qtbase/src'                                                                                                      
make[5]: *** [Makefile:51: sub-src-make_first] Error 2                                                 
make[5]: Leaving directory '/root/lede/build_dir/target-x86_64_musl/qt-everywhere-src-5.13.1/qtbase'   
make[4]: *** [Makefile:71: module-qtbase-make_first] Error 2                                           
make[4]: Leaving directory '/root/lede/build_dir/target-x86_64_musl/qt-everywhere-src-5.13.1'          
make[3]: *** [Makefile:228: /root/lede/build_dir/target-x86_64_musl/qt-everywhere-src-5.13.1/.built] Error 2                                                                                                  
make[3]: Leaving directory '/root/lede/package/lean/qt5'                                               
time: package/lean/qt5/compile#1098.64#129.62#1577.78                                                  
make[2]: *** [package/Makefile:109: package/lean/qt5/compile] Error 2                                  
make[2]: Leaving directory '/root/lede'                                                                
make[1]: *** [package/Makefile:103: /root/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2                                                                                                 
make[1]: Leaving directory '/root/lede'                                                                
make: *** [/root/lede/include/toplevel.mk:218: world] Error 2                                          
Start compile at 2019-10-28 14:01:23                                                                   
End compile at 2019-10-28 14:53:04                                                                     
Run time 51 mins

coolsnowwolf, 2019-10-28:
qt5 默认不支持 openssl 1.1

yinzhefuyun, 2019-10-29:
> qt5 默认不支持 openssl 1.1

qbittorrent这样就和一堆需要openssl的组件冲突了，尴尬

zcluo, 2019-10-29:
放弃了用回tr