coolsnowwolf/lede #1912: Compile Error from qt-everywhere

smallprogram, 2019-10-28:
## log
```ts

Makefile:26252: recipe for target '.obj/qsslsocket_openssl_symbols.o' failed
make[7]: *** [.obj/qsslsocket_openssl_symbols.o] Error 1
make[7]: Leaving directory '/home/zhusir/lede/build_dir/target-x86_64_musl/qt-everywhere-src-5.13.1/qtbase/src/network'
Makefile:272: recipe for target 'sub-network-make_first' failed
make[6]: *** [sub-network-make_first] Error 2
make[6]: Leaving directory '/home/zhusir/lede/build_dir/target-x86_64_musl/qt-everywhere-src-5.13.1/qtbase/src'
Makefile:50: recipe for target 'sub-src-make_first' failed
make[5]: *** [sub-src-make_first] Error 2
make[5]: Leaving directory '/home/zhusir/lede/build_dir/target-x86_64_musl/qt-everywhere-src-5.13.1/qtbase'
Makefile:70: recipe for target 'module-qtbase-make_first' failed
make[4]: *** [module-qtbase-make_first] Error 2
make[4]: Leaving directory '/home/zhusir/lede/build_dir/target-x86_64_musl/qt-everywhere-src-5.13.1'
Makefile:225: recipe for target '/home/zhusir/lede/build_dir/target-x86_64_musl/qt-everywhere-src-5.13.1/.built' failed
make[3]: *** [/home/zhusir/lede/build_dir/target-x86_64_musl/qt-everywhere-src-5.13.1/.built] Error 2
make[3]: Leaving directory '/home/zhusir/lede/package/lean/qt5'
time: package/lean/qt5/compile#633.02#69.27#935.39
package/Makefile:107: recipe for target 'package/lean/qt5/compile' failed
make[2]: *** [package/lean/qt5/compile] Error 2
make[2]: Leaving directory '/home/zhusir/lede'
package/Makefile:103: recipe for target '/home/zhusir/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/zhusir/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zhusir/lede'
/home/zhusir/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
```

## source code 
Commits 1611

smallprogram, 2019-10-29:
fix in commits 1618

closed 2019-10-29 by smallprogram

qq9568488, 2019-11-03:
我也出现这个问题  求大佬指点怎么解决

smallprogram, 2019-11-03:
> 我也出现这个问题 求大佬指点怎么解决

在源码提交1618以后的版本已经修复该问题。请重新克隆源码按照步骤编译即可

zxlhhyccc, 2019-11-07:
> > 我也出现这个问题 求大佬指点怎么解决
> 
> 在源码提交1618以后的版本已经修复该问题。请重新克隆源码按照步骤编译即可

请问1618是哪个版本？

smallprogram, 2019-11-08:
> > > 我也出现这个问题 求大佬指点怎么解决
> > 
> > 
> > 在源码提交1618以后的版本已经修复该问题。请重新克隆源码按照步骤编译即可
> 
> 1618以后的版本也是不行的。。。。

也许你需要删除本地源码，删除lede文件夹，重新clone一下，并且在make menuconfig之后，使用make download 将dl包下载完毕后再进行编译工作。我觉得可能你的源码或者dl下载不全导致的。