coolsnowwolf/lede #1913: 编译4.19内核出错，麻烦大神帮忙看看

bkye, 2019-10-28:
make[3]: Entering directory `/opt/lede/package/kernel/mac80211'
mkdir -p /opt/lede/bin/targets/x86/64/packages /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.79-1/ipkg-x86_64/kmod-mac80211/CONTROL /opt/lede/staging_dir/target-x86_64_musl/pkginfo
ERROR: module '/opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.79-1/net/mac80211/mac80211.ko' is missing.
make[3]: *** [/opt/lede/bin/targets/x86/64/packages/kmod-mac80211_4.19.79+4.19.79-1-1_x86_64.ipk] Error 1
make[3]: Leaving directory `/opt/lede/package/kernel/mac80211'
time: package/kernel/mac80211/compile#0.08#0.01#0.18
make[2]: *** [package/kernel/mac80211/compile] Error 2
make[2]: Leaving directory `/opt/lede'
make[1]: *** [/opt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/opt/lede'
make: *** [world] Error 2

编译4.9的内核正常，换成4.19就提示这样的错误，

coolsnowwolf, 2019-10-28:
make clean

bkye, 2019-10-28:
> make clean

make clean也是提示这个错误...

closed 2019-10-28 by coolsnowwolf

82kg, 2019-10-29:
一样的报错。楼主后面怎么搞的

kktt007, 2019-11-19:
同问,刚从libgpg-error解决过来,用openwrt的patch就没问题