coolsnowwolf/lede #1922: Compile Error from samba4-server

smallprogram, 2019-10-28:
## log
```ts
Collected errors:
 * check_data_file_clashes: Package samba4-server wants to install file /home/zhusir/lede/build_dir/target-x86_64_musl/root-x86/usr/sbin/smbd
        But that file is already provided by package  * samba36-server
 * opkg_install_cmd: Cannot install package samba4-server.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/zhusir/lede'
package/Makefile:104: recipe for target '/home/zhusir/lede/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: *** [/home/zhusir/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/zhusir/lede'
/home/zhusir/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
```

Her0R0cket, 2019-10-28:
samba3和4只能选一个继承,其他的选m或取消选择

smallprogram, 2019-10-28:
> samba3和4只能选一个继承,其他的选m或取消选择

我并没有选择samba4

Her0R0cket, 2019-10-28:
但是的确是选上了的

smallprogram, 2019-10-29:
清理环境，重新克隆代码，重新编译成功了

closed 2019-10-29 by smallprogram