coolsnowwolf/lede #1932: 新版qemu-ga编译出错

zhaoxp520, 2019-10-29:
内核4.19
更新最新源码后，添加qemu-ga后固件编译出错，之前是可以的
install -d -m0755 /home/chris/lede/bin/packages/x86_64/packages
/home/chris/lede/scripts/ipkg-build -c -o 0 -g 0 /home/chris/lede/build_dir/target-x86_64_musl/qemu-4.1.0/ipkg-x86_64/virtio-console-helper /home/chris/lede/bin/packages/x86_64/packages
Packaged contents of /home/chris/lede/build_dir/target-x86_64_musl/qemu-4.1.0/ipkg-x86_64/virtio-console-helper into /home/chris/lede/bin/packages/x86_64/packages/virtio-console-helper_4.1.0-2_x86_64.ipk
mkdir -p /home/chris/lede/bin/targets/x86/64/packages /home/chris/lede/build_dir/target-x86_64_musl/qemu-4.1.0/ipkg-x86_64/qemu-ga/CONTROL /home/chris/lede/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /home/chris/lede/build_dir/target-x86_64_musl/qemu-4.1.0/ipkg-x86_64/qemu-ga/usr/bin
install -m0755 /home/chris/lede/build_dir/target-x86_64_musl/qemu-4.1.0/ipkg-install/usr/bin/qemu-ga /home/chris/lede/build_dir/target-x86_64_musl/qemu-4.1.0/ipkg-x86_64/qemu-ga/usr/bin
install: cannot stat '/home/chris/lede/build_dir/target-x86_64_musl/qemu-4.1.0/ipkg-install/usr/bin/qemu-ga': No such file or directory
Makefile:388: recipe for target '/home/chris/lede/bin/packages/x86_64/packages/qemu-ga_4.1.0-2_x86_64.ipk' failed
make[3]: *** [/home/chris/lede/bin/packages/x86_64/packages/qemu-ga_4.1.0-2_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/chris/lede/feeds/packages/utils/qemu'
time: package/feeds/packages/qemu/compile#6.47#4.33#11.47
package/Makefile:107: recipe for target 'package/feeds/packages/qemu/compile' failed
make[2]: *** [package/feeds/packages/qemu/compile] Error 2
make[2]: Leaving directory '/home/chris/lede'
package/Makefile:103: recipe for target '/home/chris/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/chris/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/chris/lede'
/home/chris/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

zhaoxp520, 2019-10-29:
试过整个目录删掉重新弄也是这样

Kinodi, 2019-10-29:
我刚刚编译的4.19没问题，一遍过

sunxun47, 2019-10-30:
qemu-ga 编译同样出错

guoyuc, 2019-11-01:
我设置加入qemu-ga也出现一样的编译错误

RainCat1998, 2019-11-22:
同样的问题

Driverorz, 2019-11-24:
同样的问题+1

RainCat1998, 2019-11-29:
解决方法很简单，出错了，就把编译过程中生成的qemu-ga文件放到target-x86_64_musl\ipkg-install\usr\bin 目录下，再继续编译就行了。

closed 2019-12-24 by coolsnowwolf