coolsnowwolf/lede #1934: wget-ssl 问题导致订阅链接更新失败

SuLingGG, 2019-10-29:
使用中午的某个 commit 编译 OpenWrt 固件，具体的哪个 commit 记不清了...

在SSR Plus+ 更新订阅链接时，点击更新按钮后未正常更新出节点。查阅日志发现以下错误：

```
shadowsocksr_subscribe[14004]: https://xxxx/modules/servers/V2raySocks/osubscribe.php?sid=xxx&token=xxxxxxxxxxxxxxx&ssr=1 订阅数据获取失败 错误代码: 127
2019-10-29 22:01:02:https://xxxx/modules/servers/V2raySocks/osubscribe.php?sid=xxx&token=xxxxxxxxxxxxxxx&ssr=1 订阅数据获取失败 错误代码: 127
Error relocating /usr/bin/wget-ssl: secure_getenv: symbol not found
2019-10-29 22:01:02:下载订阅失败，自动重试中...
Error relocating /usr/bin/wget-ssl: secure_getenv: symbol not found
2019-10-29 22:01:02:尝试下载订阅链接到本地临时文件，请稍等...
2019-10-29 22:01:02:开始更新在线订阅列表...
2019-10-29 22:01:02:从 https://xxxx/modules/servers/V2raySocks/osubscribe.php?sid=xxx&token=xxxxxxxxxxxxxxx&ssr=1 获取订阅
shadowsocksr_subscribe[13804]: https://xxxx/modules/servers/V2raySocks/osubscribe.php?sid=xxx&token=xxxxxxxxxxxxxxx&ssr=1 订阅数据获取失败 错误代码: 127
2019-10-29 22:01:00:https://xxxx/modules/servers/V2raySocks/osubscribe.php?sid=xxx&token=xxxxxxxxxxxxxxx&ssr=1 订阅数据获取失败 错误代码: 127
Error relocating /usr/bin/wget-ssl: secure_getenv: symbol not found
```
可以看出是 wget 的问题导致订阅无法正常进行，在终端中使用 wget 测试时：

```
root@OpenWrt:~# wget https://github.com
Error relocating /usr/bin/wget: secure_getenv: symbol not found
root@OpenWrt:~# wget http://github.com
Error relocating /usr/bin/wget: secure_getenv: symbol not found
```

出现同样问题。
望修复~

coolsnowwolf, 2019-10-29:
删除整个目录重新编译即可

closed 2019-10-29 by coolsnowwolf

dongcj, 2020-01-23:
同样的问题