coolsnowwolf/lede #1938: 建议添加SSD订阅支持

marcosteam, 2019-10-30:
项目地址：[TheCGDF/SSD](https://github.com/TheCGDF/SSD-Windows)
目前大多数用户使用的还是商家提供的服务，因此需要快速导入大量节点时十分需要订阅，但众所周知，支持订阅的SSR原作者已经停止开发，且不兼容SS最新的AEAD流加密。建议Lean可以添加SSD（依附于原版SS）订阅规则的解析支持！

coolsnowwolf, 2019-10-30:
SSR 为什么要兼容 SS ?

garypang13, 2019-10-30:
> 项目地址：[TheCGDF/SSD](https://github.com/TheCGDF/SSD-Windows)
> 目前大多数用户使用的还是商家提供的服务，因此需要快速导入大量节点时十分需要订阅，但众所周知，支持订阅的SSR原作者已经停止开发，且不兼容SS最新的AEAD流加密。建议Lean可以添加SSD（依附于原版SS）订阅规则的解析支持！

只支持SS,不支持SSR和V2Ray的飞机场就不要考虑了,

outemancut, 2019-10-30:
不是V2R跟SSR的 就差不多可以淘汰了

marcosteam, 2019-10-30:
> SSR 为什么要兼容 SS ?

并不是说要兼容SS，目前大多数的机场面板已经添加了SSD订阅，我更偏向于SS最新的AEAD加密，所以只是提个建议

closed 2019-10-30 by coolsnowwolf

adamw92, 2019-11-03:
> > SSR 为什么要兼容 SS ?
> 
> 并不是说要兼容SS，目前大多数的机场面板已经添加了SSD订阅，我更偏向于SS最新的AEAD加密，所以只是提个建议

一般能提供ssd订阅链接的，ss订阅链接都是sip002格式的，然后目前是支持sip002的订阅的…