coolsnowwolf/lede #1939: 編譯

Mattaclp, 2019-10-30:
( cd /home/ubuntu/lede/build_dir/hostpkg/go-1.12.1/src ; GOROOT_FINAL=/home/ubuntu/lede/staging_dir/hostpkg/lib/go-cross  GOROOT_BOOTSTRAP=/home/ubuntu/lede/build_dir/hostpkg/go-1.12.1/.go_bootstrap CC=gcc CXX=g++  bash make.bash --no-banner ; )
Building Go cmd/dist using /home/ubuntu/lede/build_dir/hostpkg/go-1.12.1/.go_bootstrap.
Building Go toolchain1 using /home/ubuntu/lede/build_dir/hostpkg/go-1.12.1/.go_bootstrap.
go build bootstrap/cmd/compile/internal/ssa: /home/ubuntu/lede/build_dir/hostpkg/go-1.12.1/.go_bootstrap/pkg/tool/linux_amd64/6g: signal: killed
go tool dist: FAILED: /home/ubuntu/lede/build_dir/hostpkg/go-1.12.1/.go_bootstrap/bin/go install -gcflags=-l -tags=math_big_pure_go compiler_bootstrap bootstrap/cmd/...: exit status 1
Makefile:271: recipe for target '/home/ubuntu/lede/build_dir/hostpkg/go-1.12.1/.built' failed
make[3]: *** [/home/ubuntu/lede/build_dir/hostpkg/go-1.12.1/.built] Error 2
make[3]: Leaving directory '/home/ubuntu/lede/feeds/packages/lang/golang/golang'
time: package/feeds/packages/golang/host-compile#221.97#49.67#158.98
package/Makefile:107: recipe for target 'package/feeds/packages/golang/host/compile' failed
make[2]: *** [package/feeds/packages/golang/host/compile] Error 2
make[2]: Leaving directory '/home/ubuntu/lede'
package/Makefile:103: recipe for target '/home/ubuntu/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile' failed
make[1]: *** [/home/ubuntu/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
/home/ubuntu/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
[image]

coolsnowwolf, 2019-10-30:
go tool dist 下载依赖失败了，没被墙吧？

Mattaclp, 2019-10-30:
我把他刪除了，然後他重新下載，也是編譯失敗，我這個是在甲骨文vps上編譯的，不知道是否vps環境問題導致

Keith2002, 2019-11-09:
[image]
我也出现了GoLang的问题  试了3次了还是报错

closed 2019-12-24 by coolsnowwolf