coolsnowwolf/lede #1940: BaiduPCS-web 在浏览器中无选项

qiangwushuang, 2019-10-30:
在软件包内能查到已经安装，但是luci中不显示选项
BaiduPCS-Web | 3.6.8-1
下面是相关编译配置
```
#
# 3. Applications
#
CONFIG_PACKAGE_luci-app-accesscontrol=y
# CONFIG_PACKAGE_luci-app-adblock is not set
CONFIG_PACKAGE_luci-app-adbyby-plus=y
# CONFIG_PACKAGE_luci-app-advanced-reboot is not set
# CONFIG_PACKAGE_luci-app-ahcp is not set
# CONFIG_PACKAGE_luci-app-amule is not set
# CONFIG_PACKAGE_luci-app-aria2 is not set
CONFIG_PACKAGE_luci-app-arpbind=y
# CONFIG_PACKAGE_luci-app-asterisk is not set
# CONFIG_PACKAGE_luci-app-attendedsysupgrade is not set
CONFIG_PACKAGE_luci-app-autoreboot=y
CONFIG_PACKAGE_luci-app-baidupcs-web=y
```
```
#
# wireless
#
# CONFIG_PACKAGE_aircrack-ng is not set
# CONFIG_PACKAGE_airmon-ng is not set
# CONFIG_PACKAGE_dynapoint is not set
# CONFIG_PACKAGE_horst is not set
# CONFIG_PACKAGE_kismet-client is not set
# CONFIG_PACKAGE_kismet-drone is not set
# CONFIG_PACKAGE_kismet-server is not set
# CONFIG_PACKAGE_pixiewps is not set
# CONFIG_PACKAGE_reaver is not set
# CONFIG_PACKAGE_wavemon is not set
CONFIG_PACKAGE_wifischedule=y
# CONFIG_PACKAGE_464xlat is not set
# CONFIG_PACKAGE_6in4 is not set
# CONFIG_PACKAGE_6rd is not set
# CONFIG_PACKAGE_6to4 is not set
CONFIG_PACKAGE_BaiduPCS-Web=y
```

coolsnowwolf, 2019-10-30:
不要去掉 vsftp 的 luci

qiangwushuang, 2019-10-30:
问题解决，谢谢！

closed 2019-10-30 by qiangwushuang