coolsnowwolf/lede #1949: 一选择绕过中国大陆IP模式 没一会就死掉了

moweiraul, 2019-10-30:
选择绕过中国大陆IP模式，过一会，管理页面就无法登陆，ssh也无法登陆。重启后马上ssh登陆，然后图形界面也能访问了，但是过一会图形界面又连不上了，这时只要在ssh中停止ssr服务就好了。请问怎么回事，gfw模式也有类似问题，但是概率较低。我的x86编译的和mtk7621都遇到这个问题

hyird, 2019-10-30:
我也是  随机性的死机  等待正常之后 负载爆炸 7.x   没网 luci 啥都打不开

moweiraul, 2019-10-30:
> 我也是 随机性的死机 等待正常之后 负载爆炸 7.x 没网 luci 啥都打不开

我的等不到正常，只能重启

jackytang, 2019-10-30:
我昨天晚上编译的版本，是正常的呀

sirfromprc, 2019-10-31:
大屌的还是比较稳定的，我这一切正常。出现奇怪问题，一般是编译的时候有问题。编译前git pull下，再./scripts/feeds update -a && ./scripts/feeds install -a 然后make clean && rm -rf ./tmp && rm -rf .config ，这样编译最新的一般没问题

moweiraul, 2019-10-31:
> 大屌的还是比较稳定的，我这一切正常。出现奇怪问题，一般是编译的时候有问题。编译前git pull下，再./scripts/feeds update -a && ./scripts/feeds install -a 然后make clean && rm -rf ./tmp && rm -rf .config ，这样编译最新的一般没问题

试了还是这个问题，重新pull然后clean都没用，感觉这个不是编译导致的问题，是不是某些规则冲突导致的。因为在死掉后，我在ssh窗口看CPU和内存的负载也都是正常的

hyird, 2019-10-31:
> 大屌的还是比较稳定的，我这一切正常。出现奇怪问题，一般是编译的时候有问题。编译前git pull下，再./scripts/feeds update -a && ./scripts/feeds install -a 然后make clean && rm -rf ./tmp && rm -rf .config ，这样编译最新的一般没问题

我都把源码删除 重新编译了

moweiraul, 2019-10-31:
> > 大屌的还是比较稳定的，我这一切正常。出现奇怪问题，一般是编译的时候有问题。编译前git pull下，再./scripts/feeds update -a && ./scripts/feeds install -a 然后make clean && rm -rf ./tmp && rm -rf .config ，这样编译最新的一般没问题
> 
> 我都把源码删除 重新编译了

然后好了吗，我全部重删重来，还是有问题

hyird, 2019-10-31:
> > > 大屌的还是比较稳定的，我这一切正常。出现奇怪问题，一般是编译的时候有问题。编译前git pull下，再./scripts/feeds update -a && ./scripts/feeds install -a 然后make clean && rm -rf ./tmp && rm -rf .config ，这样编译最新的一般没问题
> > 
> > 
> > 我都把源码删除 重新编译了
> 
> 然后好了吗，我全部重删重来，还是有问题

还是有问题 我盲猜luci-app-ssr-plus的问题 有空的话你退回到 上古版本试试 单独这个包

sirfromprc, 2019-10-31:
那我就不清楚了，你们可以打开htop，观察进程，看看有没有什么异常的，和大屌提出来。我曾经发现过前期有个版本有一次dnsmasq被重启了。

sirfromprc, 2019-10-31:
会不会近期机场被墙的比较厉害，ssr频繁寻找切换节点导致死了，你们用自用vps看看死不死

hyird, 2019-10-31:
> 会不会近期机场被墙的比较厉害，ssr频繁寻找切换节点导致死了，你们用自用vps看看死不死

锤子 就算ssr节点挂掉了 也不应该导致 整体 都没网   我的就会 卡住30s左右 没网 啥 都进不去 等正常之后 负载爆炸

hyird, 2019-10-31:
> 会不会近期机场被墙的比较厉害，ssr频繁寻找切换节点导致死了，你们用自用vps看看死不死

你看了设置没有 10分钟诶

moweiraul, 2019-10-31:
> > 会不会近期机场被墙的比较厉害，ssr频繁寻找切换节点导致死了，你们用自用vps看看死不死
> 
> 锤子 就算ssr节点挂掉了 也不应该导致 整体 都没网 我的就会 卡住30s左右 没网 啥 都进不去 等正常之后 负载爆炸

我的症状是负载正常，但是web登陆不进去，但是上网开始正常，过一会就不行了，ssh也登陆不进去

closed 2019-12-24 by coolsnowwolf