coolsnowwolf/lede #1953: 各位最新刷了最新固件有问题么，我已经随机性死机两次了。

82kg, 2019-10-31:
N3700的cpu。以前一直很稳。最近编译了两次新固件，开启了docker。

 随机性死机；

表现就是  所有局域网设备 断网，没有dhcp服务。外网也无法访问到路由。

今天早上又遇到了，这次的固件 是前两天刚编译的，用的全新环境。彻底清空了编译的

hyird, 2019-10-31:
我和你一样

82kg, 2019-10-31:
> 我和你一样

你开启docker没？  我是死机后，必须手动重启才行

hyird, 2019-10-31:
> > 我和你一样
> 
> 你开启docker没？ 我是死机后，必须手动重启才行

没有 我的是 过30s后正常

82kg, 2019-10-31:
> > > 我和你一样
> > 
> > 
> > 你开启docker没？ 我是死机后，必须手动重启才行
> 
> 没有 我的是 过30s后正常

你内核版本 4.9 ？

hyird, 2019-10-31:
> > > > 我和你一样
> > > 
> > > 
> > > 你开启docker没？ 我是死机后，必须手动重启才行
> > 
> > 
> > 没有 我的是 过30s后正常
> 
> 你内核版本 4.9 ？
7621的 就是4..9吧

82kg, 2019-10-31:
> > > > > 我和你一样
> > > > 
> > > > 
> > > > 你开启docker没？ 我是死机后，必须手动重启才行
> > > 
> > > 
> > > 没有 我的是 过30s后正常
> > 
> > 
> > 你内核版本 4.9 ？
> > 7621的 就是4..9吧

我是X86路由，我群里另一个人 刚刚也死机了

mjyhj, 2019-10-31:
N2600也一样，死机倒是只碰到过2，3次，我多数是自动重启。
昨晚已经换了4.19，再观察一下还会不会这样

hcym, 2019-10-31:
别是插件问题哦，我的没有，三种机型，n1，1900，5020u都ok非常ok，就是插件少，只有ss＋，都是实体运行

mjyhj, 2019-10-31:
> 别是插件问题哦，我的没有，三种机型，n1，1900，5020u都ok非常ok，就是插件少，只有ss＋，都是实体运行

有可能是，我开了adbyby 酸酸乳 网易云 docker

KFERMercer, 2019-10-31:
几天前我4.9也稳定运行了一段时间, 没有遇到这个问题, j1900

82kg, 2019-10-31:
> 别是插件问题哦，我的没有，三种机型，n1，1900，5020u都ok非常ok，就是插件少，只有ss＋，都是实体运行

我怀疑是docker问题，群里也有其他人反应 莫名死机

82kg, 2019-10-31:
> > 别是插件问题哦，我的没有，三种机型，n1，1900，5020u都ok非常ok，就是插件少，只有ss＋，都是实体运行
> 
> 有可能是，我开了adbyby 酸酸乳 网易云 docker

群里另一个N3710的u  今天也莫名死机了，也开启了docker

hyird, 2019-10-31:
> > > 别是插件问题哦，我的没有，三种机型，n1，1900，5020u都ok非常ok，就是插件少，只有ss＋，都是实体运行
> > 
> > 
> > 有可能是，我开了adbyby 酸酸乳 网易云 docker
> 
> 群里另一个N3710的u 今天也莫名死机了，也开启了docker

我没有用docker  还是  假死

gww014, 2019-10-31:
内核4.9.196，也开了docker，平均2天死机1次，N3150

82kg, 2019-11-01:
> 内核4.9.196，也开了docker，平均2天死机1次，N3150

我也是2天左右，莫名死机。。。。。

coolsnowwolf, 2019-11-01:
切换为 4.19 内核，目前定位是内核的 NS 和 SFE 功能冲突

82kg, 2019-11-01:
> 切换为 4.19 内核，目前定位是内核的 NS 和 SFE 功能冲突

4.19内核编译卡在这儿，麻烦看看少了什么不
ERROR: module '/home/sunadmin/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.79-1/net/mac80211/mac80211.ko' is missing.
Makefile:561: recipe for target '/home/sunadmin/openwrt/bin/targets/x86/64/packages/kmod-mac80211_4.19.79+4.19.79-1-1_x86_64.ipk' failed

coolsnowwolf, 2019-11-01:
已修复

closed 2019-11-01 by coolsnowwolf