coolsnowwolf/lede #1955: 如何为编译环境全局代理

smallprogram, 2019-10-31:
如何在没有翻墙路由的情况下，为虚拟机里的ubuntu系统全局翻墙？
宿主机带有v2ray客户端，并且有局域网1080端口http代理。但是不知道怎么这是虚拟机ubuntu走代理

z826540272, 2019-10-31:
开个VPN

hushanwei, 2019-10-31:
Ubuntu网络设置里设置下手动代理

smallprogram, 2019-10-31:
> Ubuntu网络设置里设置下手动代理

你指的Desktop图形化界面是吗？

hushanwei, 2019-10-31:
> > Ubuntu网络设置里设置下手动代理
> 
> 你指的Desktop图形化界面是吗？

是的，ubuntu系统中，系统设置 –> 网络 –> 网络代理 –> 在HTTP/HTTPs/socks代理填写主机的IP地址，端口设置为相应v2的端口 –> 点击应用带整个系统。
命令行应该也可以。
百度 “虚拟机 代理”有更多方案

Her0R0cket, 2019-11-02:
要代理的主要是wget和git的代理,你可以百度下,还有一种笨办法就是是在vmware里再开一个虚拟机,跑openwrt,这个虚拟机第一网卡选lan区段模式并新建一个区段,第二个网卡选桥接或nat都可以,然后你的ubuntu里的网卡连接方式改为lan区段,并选择和openwrt里设置的相同名称.
这样就相当于你的ubuntu的上级是个openwrt的路由

ujincn, 2019-11-05:
ubuntu里装个proxychains4
配置代理为宿主ip和端口
然后proxychains4 make -j1 V=s

zeXtreme, 2019-11-08:
在虚拟机终端下输入
```
export http_proxy=http://宿主机IP:1080
export https_proxy=http://宿主机IP:1080
export ftp_proxy=http://宿主机IP:1080
```
然后在同一个终端中输入`make -j1 V=s`进行编译

closed 2019-11-12 by smallprogram

KnowledegAndAction, 2020-01-28:
和上门类似，我是这么操作的：
ubuntu安装proxychains
配置proxychains代理为host机的ip/V2的端口
最后proxychains make download
直接编译好像会遇到python相关的错误，另外即使make download完成后，真正编译的时候还是需要下载一些包，能不能完成下载看运气，我这里是试了几次后都编译完成了。

wuxiaocong, 2022-04-24:
> 要代理的主要是wget和git的连接,你可以百度下在linux怎么配置. 还有一种笨办法就是是在vmware里再开一个虚拟机,跑openwrt,这个虚拟机第一网卡选lan区段模式并新建一个区段,第二个网卡选桥接或nat都可以,然后你的ubuntu里的网卡连接方式改为lan区段,并选择和openwrt里设置的相同名称.这样就相当于你的ubuntu的上级是个openwrt的路由

大佬，这种方法在笔记本下尝试无效啊，要不就是op无网，要不就是虚拟机连的不是op的网

Her0R0cket, 2022-04-26:
> 大佬，这种方法在笔记本下尝试无效啊，要不就是op无网，要不就是虚拟机连的不是op的网

这个方法应该没什么人用吧,现在想想当时的想法好奇怪,就算用op的话,也只要给虚拟机op分配一个桥接的虚拟网卡,然后让op工作在旁路由模式下就好.

wuxiaocong, 2022-04-29:
> 

好像只能桥接物理网卡，并且无线网卡也没见人搞定，桥接物理网卡的话换了网络环境就要改子IP，比较麻烦。

Her0R0cket, 2022-04-30:
> > 
> 
> 好像只能桥接物理网卡，并且无线网卡也没见人搞定，桥接物理网卡的话换了网络环境就要改子IP，比较麻烦。

       什么叫无线网卡桥接没人搞定,vmware workstation的桥接本来就是自动的,有任何一个网络先连接的话就会桥接那个网络,想手动指定的话可以这样设置: 
                  把vmnet0的自动桥接改成有线网卡,再添加一个虚拟网络,模式改桥接,连接改到无线网卡
                  或者直接把原始的vmnet0的网卡连接改到无线网卡去.
       我想起了当时为什么告诉他那种方法了,怕他不会在命令行下改网络配置.

[image]