coolsnowwolf/lede #1956: 编译报错 帮忙看一下

w64139303, 2019-10-31:
checking for getaddrinfo... yes
checking getaddrinfo bug... yes
Fatal: You must get working getaddrinfo() function.
       or you can specify "--disable-ipv6".
Makefile:354: recipe for target '/home/wg/lede/build_dir/hostpkg/Python-2.7.17/.configured' failed
make[3]: *** [/home/wg/lede/build_dir/hostpkg/Python-2.7.17/.configured] Error 1
make[3]: Leaving directory '/home/wg/lede/feeds/packages/lang/python/python'
time: package/feeds/packages/python/host-compile#31.02#8.06#41.15
package/Makefile:107: recipe for target 'package/feeds/packages/python/host/compile' failed
make[2]: *** [package/feeds/packages/python/host/compile] Error 2
make[2]: Leaving directory '/home/wg/lede'
package/Makefile:103: recipe for target '/home/wg/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile' failed
make[1]: *** [/home/wg/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/wg/lede'
/home/wg/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

lyin888, 2019-10-31:
太片面

speedyworldclub, 2019-10-31:
Try with Disabled Ipv6

w64139303, 2019-10-31:
luci里面没有选ipv6 ，扩展包里面也没选

lyvivian0077, 2019-10-31:
### 我也是同样的问题

```
/home/xxx/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.4.0_musl/include/fortify/stdio.h:114:3: warning: format not a string literal, argument types not checked [-Wformat-nonliteral]
   __r = __orig_sprintf(__s, __f, __builtin_va_arg_pack());
   ^~~
lanplus_crypt_impl.c: In function 'lanplus_encrypt_aes_cbc_128':
lanplus_crypt_impl.c:167:17: error: storage size of 'ctx' isn't known
  EVP_CIPHER_CTX ctx;
                 ^~~
lanplus_crypt_impl.c:167:17: warning: unused variable 'ctx' [-Wunused-variable]
lanplus_crypt_impl.c: In function 'lanplus_decrypt_aes_cbc_128':
lanplus_crypt_impl.c:242:17: error: storage size of 'ctx' isn't known
  EVP_CIPHER_CTX ctx;
                 ^~~
lanplus_crypt_impl.c:242:17: warning: unused variable 'ctx' [-Wunused-variable]
Makefile:430: recipe for target 'lanplus_crypt_impl.lo' failed
make[8]: *** [lanplus_crypt_impl.lo] Error 1
make[8]: Leaving directory '/home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/ipmitool-1.8.18/src/plugins/lanplus'
Makefile:483: recipe for target 'all-recursive' failed
make[7]: *** [all-recursive] Error 1
make[7]: Leaving directory '/home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/ipmitool-1.8.18/src/plugins'
Makefile:582: recipe for target 'all-recursive' failed
make[6]: *** [all-recursive] Error 1
make[6]: Leaving directory '/home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/ipmitool-1.8.18/src'
Makefile:457: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory '/home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/ipmitool-1.8.18'
Makefile:388: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/ipmitool-1.8.18'
Makefile:53: recipe for target '/home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/ipmitool-1.8.18/.built' failed
make[3]: *** [/home/xxx/lede/build_dir/target-aarch64_cortex-a53_musl/ipmitool-1.8.18/.built] Error 2
make[3]: Leaving directory '/home/xxx/lede/feeds/packages/admin/ipmitool'
time: package/feeds/packages/ipmitool/compile#50.61#5.84#53.34
package/Makefile:107: recipe for target 'package/feeds/packages/ipmitool/compile' failed
make[2]: *** [package/feeds/packages/ipmitool/compile] Error 2
make[2]: Leaving directory '/home/xxx/lede'
package/Makefile:103: recipe for target '/home/xxx/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile' failed
make[1]: *** [/home/xxx/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xxx/lede'
/home/xxx/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
xxx@ubuntu:~/lede$
```

lyvivian0077, 2019-10-31:
我的是把IPMITOOL 去掉后,编译通过了.

w64139303, 2019-10-31:
这个错误我经常遇到，等下我在试试

AmadeusGhost, 2019-11-03:
这个我前几天遇到了，根据错误提示进build_dir/hostpkg/Python-2.7.17，然后改configure disable ipv6，具体怎么改忘了，改了之后就可以编译了。

AmadeusGhost, 2019-11-16:
是不是使用了proxychians代理来编译，如果是的话建议先用```proxychains make download V=s```来先下好dl包。

closed 2019-12-24 by coolsnowwolf