coolsnowwolf/lede #1966: 4.19的内核和xt_FULLCONENAT报内核warning

zcluo, 2019-11-02:
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.238330] WARNING: CPU: 1 PID: 8719 at __nf_unregister_net_hook+0x43/0x110
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.238695] Modules linked in: xt_FULLCONENAT ath9k ath9k_htc ath9k_common rt2800usb rt2800lib pppoe ppp_async ath9k_hw ath5k ath10k_pci ath10k_core ath snd_usb_audio rt2x00usb rt2x00lib pppox ppp_mppe ppp_generic nf_nat_pptp nf_conntrack_pptp mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 ax88179_178a asix xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_policy xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda tulip
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.241835]  ts_fsm ts_bm snd_usbmidi_lib slhc rtl8150 r8169 r8152 pcnet32 nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_proto_gre nf_nat_irc nf_nat_ipv4 nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan iptable_raw iptable_mangle iptable_filter ipt_ah ipt_ECN ip_tables i40evf i40e e1000e crc_ccitt compat br_netfilter bnx2 asn1_decoder fuse sch_cake tcp_bbr act_simple em_cmp sch_teql em_nbyte sch_prio sch_dsmark sch_pie act_ipt em_meta sch_gred act_pedit
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.245010]  act_csum libcrc32c sch_multiq act_gact em_text sch_codel sch_red sch_sfq act_police sch_fq cls_matchall act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw cls_basic sch_tbf sch_htb sch_hfsc sch_ingress configs evdev i2c_dev cryptodev xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_NPT ip6t_MASQUERADE ip6table_nat nf_nat_ipv6 nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables bonding ip_gre gre snd_via82xx
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.248108]  snd_mpu401_uart ixgbe igbvf igb i2c_algo_bit e1000 snd_ac97_codec ac97_bus ifb dummy sit mdio ipcomp6 xfrm6_tunnel xfrm6_mode_tunnel xfrm6_mode_transport xfrm6_mode_beet esp6 ah6 ipcomp xfrm4_tunnel xfrm4_mode_tunnel xfrm4_mode_transport xfrm4_mode_beet esp4 ah4 tunnel6 tunnel4 ip_tunnel veth tun snd_compress snd_pcm_oss snd_mixer_oss snd_pcm snd_timer snd_rawmidi snd_seq_device snd_hwdep snd soundcore af_key xfrm_user xfrm_ipcomp xfrm_algo vfat fat dm_crypt dm_mirror dm_region_hash dm_log dm_mod dax nls_utf8 nls_iso8859_1 nls_cp437 serpent_avx_x86_64 blowfish_x86_64 twofish_avx_x86_64 twofish_x86_64_3way twofish_x86_64 cast6_avx_x86_64 cast5_avx_x86_64 camellia_aesni_avx_x86_64 camellia_x86_64 serpent_generic blowfish_common wp512 twofish_common tgr192 tea khazad cast6_generic cast5_generic
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.251320]  cast_common camellia_generic anubis sha256_ssse3 sha256_generic sha1_ssse3 sha1_generic jitterentropy_rng drbg md5 hmac echainiv des_generic deflate zlib_deflate cbc authenc arc4 crypto_acompress exfat button_hotplug ptp realtek pps_core mii libphy
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.252418] CPU: 1 PID: 8719 Comm: kworker/u4:2 Not tainted 4.19.81 #0
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.252714] Hardware name: VMware, Inc. VMware Virtual Platform/440BX Desktop Reference Platform, BIOS 6.00 04/05/2016
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.253200] Workqueue: netns cleanup_net
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.253397] RIP: 0010:__nf_unregister_net_hook+0x43/0x110
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.253648] Code: fa ff ff 48 85 c0 0f 84 e0 00 00 00 49 89 c5 49 89 ff 48 c7 c7 60 9e 07 82 41 89 f6 e8 c6 3b 0e 00 4d 8b 65 00 4d 85 e4 75 13 <0f> 0b 48 c7 c7 60 9e 07 82 e8 0f 36 0e 00 e9 af 00 00 00 41 0f b7
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.254454] RSP: 0018:ffffc90004417d88 EFLAGS: 00010246
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.254698] RAX: 0000000000000000 RBX: ffffffffa0738040 RCX: 0000000000000000
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.255061] RDX: ffff88807c39bc00 RSI: 0000000000000002 RDI: ffffffff82079e60
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.255380] RBP: ffffc90004417db0 R08: 0000000000000001 R09: ffff88807dd9f9c0
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.255698] R10: ffffc90004417d68 R11: 0000000000000000 R12: 0000000000000000
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.256047] R13: ffff88806fb48b80 R14: 0000000000000002 R15: ffff88806fb48000
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.256370] FS:  0000000000000000(0000) GS:ffff88807db00000(0000) knlGS:0000000000000000
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.256742] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.257006] CR2: 00007f625a2de000 CR3: 0000000002008005 CR4: 00000000001606e0
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.257387] Call Trace:
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.257531]  nf_unregister_net_hook+0x3b/0x40
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.257752]  nf_unregister_net_hooks+0x2a/0x40
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.257970]  0xffffffffa0736045
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.258134]  ops_exit_list.isra.9+0x39/0x60
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.258338]  cleanup_net+0x177/0x260
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.258531]  process_one_work+0x1d2/0x320
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.258729]  worker_thread+0x20d/0x3c0
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.258922]  kthread+0x113/0x120
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.259094]  ? process_one_work+0x320/0x320
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.259298]  ? kthread_create_worker_on_cpu+0x40/0x40
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.259549]  ret_from_fork+0x35/0x40
Sat Nov  2 09:42:20 2019 kern.warn kernel: [ 1860.259730] ---[ end trace b86c142ac74ab18e ]---

zcluo, 2019-11-02:
x86_64

tomerilery, 2019-11-02:
The latest recource to complie (commit f4e3590), And it failed
Note:
touch /home/tomery/x86/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/tomery/x86/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/.built
touch /home/tomery/x86/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/.built_check
make -j1 -C "/home/tomery/x86/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.81" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" SUBDIRS="/home/tomery/x86/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0" EXTRA_CFLAGS="" modules
make[4]: Entering directory '/home/tomery/x86/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.81'
make[5]: *** No rule to make target 'tools/objtool/objtool', needed by '/home/tomery/x86/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/xt_FULLCONENAT.o'.  Stop.
Makefile:1517: recipe for target '_module_/home/tomery/x86/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0' failed
make[4]: *** [_module_/home/tomery/x86/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0] Error 2
make[4]: Leaving directory '/home/tomery/x86/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.81'
Makefile:64: recipe for target '/home/tomery/x86/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/.built' failed
make[3]: *** [/home/tomery/x86/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/.built] Error 2
make[3]: Leaving directory '/home/tomery/x86/lede/package/lean/openwrt-fullconenat'
time: package/lean/openwrt-fullconenat/compile#0.94#0.35#1.22
package/Makefile:107: recipe for target 'package/lean/openwrt-fullconenat/compile' failed
make[2]: *** [package/lean/openwrt-fullconenat/compile] Error 2
make[2]: Leaving directory '/home/tomery/x86/lede'
package/Makefile:103: recipe for target '/home/tomery/x86/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/tomery/x86/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/tomery/x86/lede'
/home/tomery/x86/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
tomery@ubuntu:~/x86/lede$

czxinshui, 2019-11-02:
make[3]: *** [/home/water/lede/bin/targets/x86/64/packages/kmod-mac80211_4.19.81+4.19.79-1-1_x86_64.ipk] Error 1
make[3]: Leaving directory `/home/water/lede/package/kernel/mac80211'
time: package/kernel/mac80211/compile#20.84#1.74#30.45
make[2]: *** [package/kernel/mac80211/compile] Error 2
make[2]: Leaving directory `/home/water/lede'
make[1]: *** [/home/water/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/water/lede'
make: *** [world] 错误 2

Aes64X, 2019-11-02:
大雕主要优化4.9内核，如果你要使用4.19内核，建议串一些openwrt主线的一些修复，比如使用openwrt官方的mac80211

coolsnowwolf, 2019-11-03:
这个是 NS 内核和 4.19 的 Fullcone 的支持问题，初始化的时候会触发报错，但不会影响使用

Her0R0cket, 2019-11-03:
刚刚在虚拟机上测试,发现也有这个问题,这个只在docker启动时才有,目前还不玩docker,先关了再说,看到这个总归有点不舒服.

zcluo, 2019-11-05:
> 这个是 NS 内核和 4.19 的 Fullcone 的支持问题，初始化的时候会触发报错，但不会影响使用

确实不影响，使用了几天还算稳定谢谢大佬回复

closed 2019-11-05 by zcluo