coolsnowwolf/lede #1968: 艰难啊，这次是boost了

vincentchen, 2019-11-02:
[image]

Her0R0cket, 2019-11-02:
把boost退回到1.69,新boost下rblibtorrent编译失败

vincentchen, 2019-11-02:
不行啊，实在没脾气了
[image]

toni-l, 2019-11-02:
这个要配置编译环境变量，比较麻烦，别折腾了

Her0R0cket, 2019-11-02:
> 
> 
> 不行啊，实在没脾气了
> [image]

你是不是没隐藏patch?

Her0R0cket, 2019-11-02:
看日志好像已经编译完成,复制文件的时候缺失文件失败了

Her0R0cket, 2019-11-02:
1.69不能只改版本,makefile差别还是很大的

closed 2019-12-24 by coolsnowwolf