coolsnowwolf/lede #1971: 2019年11月2日代码x86的mac80211编译不过去

tinnlux, 2019-11-02:
ERROR: module '/home/ubuntu/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.79-1/net/mac80211/mac80211.ko' is missing.
Makefile:561: recipe for target '/home/ubuntu/lede/bin/targets/x86/64/packages/kmod-mac80211_4.19.81+4.19.79-1-1_x86_64.ipk' failed
make[3]: *** [/home/ubuntu/lede/bin/targets/x86/64/packages/kmod-mac80211_4.19.81+4.19.79-1-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/ubuntu/lede/package/kernel/mac80211'
time: package/kernel/mac80211/compile#0.17#0.04#0.18
package/Makefile:107: recipe for target 'package/kernel/mac80211/compile' failed
make[2]: *** [package/kernel/mac80211/compile] Error 2
make[2]: Leaving directory '/home/ubuntu/lede'
package/Makefile:103: recipe for target '/home/ubuntu/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/ubuntu/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
/home/ubuntu/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

liujixings, 2019-11-02:
我也遇到这问题，，且不只是这个还有好多无线驱动编译的时候有类似报错，都是没有ko文件，我能力有限找不到解决办法，只好从openwrt源里面下载对应ipk，解包提取ko文件放到对应目录里去

我这里有这样问题的都有
kmod-mac80211
kmod-ath5k
kmod-ath9k-common
kmod-ath9k-htc
kmod-ath10k
kmod-ath10k-ct
kmod-iwlwifi
kmod-rt2x00-lib
kmod-rt2x00-usb
kmod-rt2800-lib
kmod-rt2800-usb

tinnlux, 2019-11-02:
@liujixings 你是编译的x86的固件吗？你这个方法好啊。。我也试试

bkye, 2019-11-02:
这个问题我也遇到，换成4.9，或者4.14都可以正常编译，用4.19就会出现这个错误，把源码删了再重新编译也不行，换另外的系统版本也不行....

liujixings, 2019-11-02:
> @liujixings你是编译的x86的固件吗？你这个方法好啊。。我也试试

是的x86 64的固件

liujixings, 2019-11-02:
> 这个问题我也遇到，换成4.9，或者4.14都可以正常编译，用4.19就会出现这个错误，把二进制文件删除了再重新编译也不行，换另外的系统版本也不行....

貌似只有4.14才有对应的.ko文件

hcym, 2019-11-02:
这个累了，要么回退要么等等。两天前的正常使用，不急不急了，呵呵

Aes64X, 2019-11-02:
大雕主要优化4.9内核，如果你要使用4.19内核，建议串一些openwrt主线的一些修复，比如使用openwrt官方的mac80211

Aes64X, 2019-11-02:
包括ath10k驱动，都要跟openwrt官方的走，大雕的是跟4.9内核走

bkye, 2019-11-02:
> 大雕主要优化4.9内核，如果你要使用4.19内核，建议串一些openwrt主线的一些修复，比如使用openwrt官方的mac80211

感谢，替换了官方的文件已成功编译！

tinnlux, 2019-11-02:
@Aes64X 大佬的配置文件现在默认编译4.19的内核，如果要编译4.9的需要手动更改了。

tinnlux, 2019-11-02:
@bkye 请问替换哪些官方文件啊？

Her0R0cket, 2019-11-03:
删掉target-x86_64_musl/linux-x86_64文件夹,重来一次试试,我4.19的内核是过了的,如果你在升级openssl之后没有编译,也没有运行过make clean的话,必须make clean之后再来

bkye, 2019-11-03:
> @bkye 请问替换哪些官方文件啊？

mac80211和ath10k，反正我替换这两个就可以了

coolsnowwolf, 2019-11-03:
可以编译过去的，测试通过

closed 2019-11-03 by tinnlux