coolsnowwolf/lede #1976: 2019-11-3 make failed

Godseeler, 2019-11-03:
cc1: some warnings being treated as errors
scripts/Makefile.build:303: recipe for target '/home/godseeker/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-1.9.git-2017-10-04/ioctl.o' failed
make[6]: *** [/home/godseeker/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-1.9.git-2017-10-04/ioctl.o] Error 1
Makefile:1517: recipe for target '_module_/home/godseeker/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-1.9.git-2017-10-04' failed
make[5]: *** [_module_/home/godseeker/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-1.9.git-2017-10-04] Error 2
make[5]: Leaving directory '/home/godseeker/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.81'
Makefile:27: recipe for target 'build' failed
make[4]: *** [build] Error 2
make[4]: Leaving directory '/home/godseeker/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-1.9.git-2017-10-04'
Makefile:74: recipe for target '/home/godseeker/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-1.9.git-2017-10-04/.built' failed
make[3]: *** [/home/godseeker/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-1.9.git-2017-10-04/.built] Error 2
make[3]: Leaving directory '/home/godseeker/lede/feeds/packages/utils/cryptodev-linux'
time: package/feeds/packages/cryptodev-linux/compile#0.91#0.22#1.03
package/Makefile:107: recipe for target 'package/feeds/packages/cryptodev-linux/compile' failed
make[2]: *** [package/feeds/packages/cryptodev-linux/compile] Error 2
make[2]: Leaving directory '/home/godseeker/lede'
package/Makefile:103: recipe for target '/home/godseeker/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/godseeker/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/godseeker/lede'
/home/godseeker/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

iamggs, 2019-11-03:
给你一模一样。

Her0R0cket, 2019-11-03:
cryptodev-linux  20天前就已经更新到1.10了

Godseeler, 2019-11-03:
> 
> 
> 给你一模一样。

rm -rf lede 重新来就好了

iamggs, 2019-11-03:
> > 给你一模一样。
> 
> rm -rf lede 重新来就好了

我晕，就是不想推倒重来。

closed 2019-11-03 by Godseeler

whlsxl, 2020-01-12:
```
./scripts/feeds clean 
./scripts/feeds update -a 
./scripts/feeds install -a
```