coolsnowwolf/lede #1980: 你好L大，今天遇到了编译错误，麻烦您看一下。

raiuka, 2019-11-03:
之前的编译都非常成功，今天git pull以后，取消了全部的无线网卡驱动和跟无线相关的东西后，再次编译出现问题：

`make[4]: Entering directory `/home/user/lede/build_dir/target-x86_64_musl/hostapd-2019-08-08-ca8c2bd2/wpa_supplicant'
make[4]: `wpa_cli' is up to date.
make[4]: `wpa_supplicant_multi.a' is up to date.
make[4]: Leaving directory `/home/user/lede/build_dir/target-x86_64_musl/hostapd-2019-08-08-ca8c2bd2/wpa_supplicant'
export MAKEFLAGS=""; x86_64-openwrt-linux-musl-gcc -o /home/user/lede/build_dir/target-x86_64_musl/hostapd-2019-08-08-ca8c2bd2/wpad -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/user/lede/build_dir/target-x86_64_musl/hostapd-2019-08-08-ca8c2bd2:hostapd-2019-08-08-ca8c2bd2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections -flto ./files/multicall.c /home/user/lede/build_dir/target-x86_64_musl/hostapd-2019-08-08-ca8c2bd2/hostapd/hostapd_multi.a /home/user/lede/build_dir/target-x86_64_musl/hostapd-2019-08-08-ca8c2bd2/wpa_supplicant/wpa_supplicant_multi.a -L/home/user/lede/staging_dir/target-x86_64_musl/usr/lib -L/home/user/lede/staging_dir/target-x86_64_musl/lib -L/home/user/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/usr/lib -L/home/user/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib -znow -zrelro -Wl,--gc-sections -flto=jobserver -fuse-linker-plugin -lubox -lubus
/home/user/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/7.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/user/lede/tmp/cckXKaq7.ltrans10.ltrans.o: in function `ap_sta_add':
<artificial>:(.text.ap_sta_add+0x1d3): undefined reference to `sta_track_claim_taxonomy_info'
/home/user/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/7.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/user/lede/tmp/cckXKaq7.ltrans13.ltrans.o: in function `br_addif':
<artificial>:(.text.br_addif+0x59): undefined reference to `linux_br_add_if'
/home/user/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/7.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/user/lede/tmp/cckXKaq7.ltrans13.ltrans.o: in function `vlan_get_bridge':
<artificial>:(.text.vlan_get_bridge+0x74): undefined reference to `linux_br_add'
/home/user/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/7.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/user/lede/tmp/cckXKaq7.ltrans13.ltrans.o: in function `br_delif':
<artificial>:(.text.br_delif+0x5c): undefined reference to `linux_br_del_if'
/home/user/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/7.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/user/lede/tmp/cckXKaq7.ltrans13.ltrans.o: in function `vlan_put_bridge':
<artificial>:(.text.vlan_put_bridge+0xa5): undefined reference to `linux_br_del'
collect2: error: ld returned 1 exit status
make[3]: *** [/home/user/lede/build_dir/target-x86_64_musl/hostapd-2019-08-08-ca8c2bd2/.built] Error 1
make[3]: Leaving directory `/home/user/lede/package/network/services/hostapd'
time: package/network/services/hostapd/wpad-full-internal/compile#19.39#1.21#20.81
make[2]: *** [package/network/services/hostapd/compile] Error 2
make[2]: Leaving directory `/home/user/lede'
make[1]: *** [/home/user/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/user/lede'
make: *** [world] Error 2
`
我google了一下collect2: error: ld returned 1 exit status这个报错，好像是说源码存在错误。但是我去/home/user/lede/tmp/文件夹下寻找cckXKaq7.ltrans13.ltrans.o也没有找到这个文件。

核心版本4.9
编译主机Ubuntu 14.04

如果可以的话，请帮忙看一下

czjesda, 2019-11-04:
你吧文件都删除了，重新下载后试试，然后把内核改成4.19看看

tvrohd, 2019-11-05:
> 你吧文件都删除了，重新下载后试试，然后把内核改成4.19看看

内核怎么改回4.19

czjesda, 2019-11-05:
> 
> 
> > 你吧文件都删除了，重新下载后试试，然后把内核改成4.19看看
> 
> 内核怎么改回4.19

打开这个目录：\lede\target\linux\x86\Makefile文件，把里面的4.9改成4.19

closed 2019-11-07 by raiuka

kktt007, 2019-11-11:
> 
> 
> 你吧文件都删除了，重新下载后试试，然后把内核改成4.19看看

圆圆真漂亮

zengmiaokun, 2019-11-27:
> 之前的编译都非常成功，今天git pull以后，取消了全部的无线网卡驱动和跟无线相关的东西后，再次编译出现问题：
> 
> 编译错误信息：
> 
> make[4]: Entering directory `/home/user/lede/build_dir/target-x86_64_musl/hostapd-2019-08-08-ca8c2bd2/wpa_supplicant' make[4]: `wpa_cli' is up to date.
> make[4]: `wpa_supplicant_multi.a' is up to date. make[4]: Leaving directory `/home/user/lede/build_dir/target-x86_64_musl/hostapd-2019-08-08-ca8c2bd2/wpa_supplicant'
> export MAKEFLAGS=""; x86_64-openwrt-linux-musl-gcc -o /home/user/lede/build_dir/target-x86_64_musl/hostapd-2019-08-08-ca8c2bd2/wpad -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/user/lede/build_dir/target-x86_64_musl/hostapd-2019-08-08-ca8c2bd2:hostapd-2019-08-08-ca8c2bd2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections -flto ./files/multicall.c /home/user/lede/build_dir/target-x86_64_musl/hostapd-2019-08-08-ca8c2bd2/hostapd/hostapd_multi.a /home/user/lede/build_dir/target-x86_64_musl/hostapd-2019-08-08-ca8c2bd2/wpa_supplicant/wpa_supplicant_multi.a -L/home/user/lede/staging_dir/target-x86_64_musl/usr/lib -L/home/user/lede/staging_dir/target-x86_64_musl/lib -L/home/user/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/usr/lib -L/home/user/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib -znow -zrelro -Wl,--gc-sections -flto=jobserver -fuse-linker-plugin -lubox -lubus
> /home/user/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/7.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/user/lede/tmp/cckXKaq7.ltrans10.ltrans.o: in function `ap_sta_add': <artificial>:(.text.ap_sta_add+0x1d3): undefined reference to `sta_track_claim_taxonomy_info'
> /home/user/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/7.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/user/lede/tmp/cckXKaq7.ltrans13.ltrans.o: in function `br_addif': <artificial>:(.text.br_addif+0x59): undefined reference to `linux_br_add_if'
> /home/user/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/7.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/user/lede/tmp/cckXKaq7.ltrans13.ltrans.o: in function `vlan_get_bridge': <artificial>:(.text.vlan_get_bridge+0x74): undefined reference to `linux_br_add'
> /home/user/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/7.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/user/lede/tmp/cckXKaq7.ltrans13.ltrans.o: in function `br_delif': <artificial>:(.text.br_delif+0x5c): undefined reference to `linux_br_del_if'
> /home/user/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/7.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/user/lede/tmp/cckXKaq7.ltrans13.ltrans.o: in function `vlan_put_bridge': <artificial>:(.text.vlan_put_bridge+0xa5): undefined reference to `linux_br_del'
> collect2: error: ld returned 1 exit status
> make[3]: *** [/home/user/lede/build_dir/target-x86_64_musl/hostapd-2019-08-08-ca8c2bd2/.built] Error 1
> make[3]: Leaving directory `/home/user/lede/package/network/services/hostapd' time: package/network/services/hostapd/wpad-full-internal/compile#19.39#1.21#20.81 make[2]: *** [package/network/services/hostapd/compile] Error 2 make[2]: Leaving directory `/home/user/lede'
> make[1]: *** [/home/user/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
> make[1]: Leaving directory `/home/user/lede'
> make: *** [world] Error 2
> 
> 我google了一下collect2: error: ld returned 1 exit status这个报错，好像是说源码存在错误。但是我去/home/user/lede/tmp/文件夹下寻找cckXKaq7.ltrans13.ltrans.o也没有找到这个文件。
> 也没有google到相同的错误。尝试make clean后编译，问题依旧，暂时还没有尝试重拉代码，正在测试。
> 
> 核心版本4.9
> 编译主机Ubuntu 14.04
> 编译target：X86
> 
> 如果可以的话，请帮忙看一下

我也遇到了这个问题，你是怎么解决的。

zengmiaokun, 2019-11-27:
> 之前的编译都非常成功，今天git pull以后，取消了全部的无线网卡驱动和跟无线相关的东西后，再次编译出现问题：
>
> 编译错误信息：
>
> make[4]: Entering directory `/home/user/lede/build_dir/target-x86_64_musl/hostapd-2019-08-08-ca8c2bd2/wpa_supplicant' make[4]: [...]

我也遇到了这个问题，你是怎么解决的。

czjesda, 2019-11-27:
> 我也遇到了这个问题，你是怎么解决的。

你吧无线都取消了，还有2个地方要取消，都是network目录下，找到wpad和hostpad-command这个也取消了，然后运行make clean后再编译就成功了

zengmiaokun, 2019-11-27:
> 你吧无线都取消了，还有2个地方要取消，都是network目录下，找到wpad和hostpad-command这个也取消了，然后运行make clean后再编译就成功了

可是那样就没有无线了啊，还要刷了之后再安装。

czjesda, 2019-11-27:
> 可是那样就没有无线了啊，还要刷了之后再安装。

这个方法是针对软路由的，如果是硬路由的，你先git pull到最新的地方，然后运行./scripts/feeds update -a && ./scripts/feeds install -a和rm -rf ./tmp && rm -rf .config，接着运行make menuconfig, 都弄好了先运行make download V=s下载DL库，最后运行make -j1 V=s

zengmiaokun, 2019-11-27:
> 这个方法是针对软路由的，如果是硬路由的，你先git pull到最新的地方，然后运行./scripts/feeds update -a && ./scripts/feeds install -a和rm -rf ./tmp && rm -rf .config，接着运行make menuconfig, 都弄好了先运行make download V=s下载DL库，最后运行make -j1 V=s

好的，我试试，谢谢了

zengmiaokun, 2019-11-27:
> 这个方法是针对软路由的，如果是硬路由的，你先git pull到最新的地方，然后运行./scripts/feeds update -a && ./scripts/feeds install -a和rm -rf ./tmp && rm -rf .config，接着运行make menuconfig, 都弄好了先运行make download V=s下载DL库，最后运行make -j1 V=s

才发现，我们的错误不是一样的，我的是这个。
time: package/network/services/hostapd/wpad-basic/compile#0.05#0.38#1.91
make[3]: Entering directory `/home/mango/Desktop/lede/package/network/services/hostapd'
mkdir -p /home/mango/Desktop/lede/bin/targets/ramips/mt7621/packages /home/mango/Desktop/lede/build_dir/target-mipsel_24kc_musl/hostapd-2019-08-08-ca8c2bd2/ipkg-mipsel_24kc/wpa-supplicant/CONTROL /home/mango/Desktop/lede/staging_dir/target-mipsel_24kc_musl/pkginfo
install -d -m0755 /home/mango/Desktop/lede/build_dir/target-mipsel_24kc_musl/hostapd-2019-08-08-ca8c2bd2/ipkg-mipsel_24kc/wpa-supplicant/usr/sbin
install -m0755 /home/mango/Desktop/lede/build_dir/target-mipsel_24kc_musl/hostapd-2019-08-08-ca8c2bd2/wpa_supplicant/wpa_supplicant /home/mango/Desktop/lede/build_dir/target-mipsel_24kc_musl/hostapd-2019-08-08-ca8c2bd2/ipkg-mipsel_24kc/wpa-supplicant/usr/sbin/
install: cannot stat '/home/mango/Desktop/lede/build_dir/target-mipsel_24kc_musl/hostapd-2019-08-08-ca8c2bd2/wpa_supplicant/wpa_supplicant': No such file or directory
make[3]: *** [/home/mango/Desktop/lede/bin/packages/mipsel_24kc/base/wpa-supplicant_2019-08-08-ca8c2bd2-2_mipsel_24kc.ipk] Error 1
make[3]: Leaving directory `/home/mango/Desktop/lede/package/network/services/hostapd'
time: package/network/services/hostapd/supplicant-full-internal/compile#0.04#0.30#1.64
make[2]: *** [package/network/services/hostapd/compile] Error 2
make[2]: Leaving directory `/home/mango/Desktop/lede'
make[1]: *** [/home/mango/Desktop/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/mango/Desktop/lede'
make: *** [world] Error 2

czjesda, 2019-11-28:
还是hostapd有问题

dovegq, 2020-03-10:
> 你吧无线都取消了，还有2个地方要取消，都是network目录下，找到wpad和hostpad-command这个也取消了，然后运行make clean后再编译就成功了

 我故意取消了无线，去掉了wpad和hostpad-comm就好了 感谢