coolsnowwolf/lede #1982: 最新源码编译不过，cryptodev编译报错

greennyreborn, 2019-11-03:
报错如下
`shell
/home/michael/lede/build_dir/target-x86_64_musl/linux-x86_64/cryptodev-linux-1.9.git-2017-10-04/ioctl.c:831:4: error: implicit declaration of function 'sys_close'; did you mean 'ksys_close'? [-Werror=implicit-function-declaration]
    sys_close(fd);
    ^~~~~~~~~
    ksys_close
`
稍微研究了下，发现老版本的cryptodev在4.17+的kernel有问题，连接如下
https://patchwork.openembedded.org/patch/151867/

qingtiandalaoye, 2019-12-17:
可以绕过吗？

qingtiandalaoye, 2019-12-17:
找到sys_close(fd); 所在 ./build_dir/target-aarch64_cortex-a53_musl/linux-brcm2708_bcm2710/cryptodev-linux-1.9.git-2017-10-04/ioctl.c      的文件，修改如下：

#if (LINUX_VERSION_CODE < KERNEL_VERSION(4, 17, 0))
 			sys_close(fd);
#else
			ksys_close(fd);
#endif

closed 2019-12-24 by coolsnowwolf