coolsnowwolf/lede #1986: 编译出错

xinxijishuwyq, 2019-11-03:
ERROR: module '/home/ken/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.79-1/net/mac80211/mac80211.ko' is missing.
make[3]: *** [Makefile:563: /home/ken/lede/bin/targets/x86/64/packages/kmod-mac80211_4.19.81+4.19.79-1-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/ken/lede/package/kernel/mac80211'
time: package/kernel/mac80211/compile#0.26#0.08#0.33
make[2]: *** [package/Makefile:109: package/kernel/mac80211/compile] Error 2
make[2]: Leaving directory '/home/ken/lede'
make[1]: *** [package/Makefile:103: /home/ken/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ken/lede'
make: *** [/home/ken/lede/include/toplevel.mk:218: world] Error 2

hcym, 2019-11-03:
回滚吧，我也是退回三天前的4.9内核才行

yclin20, 2019-11-03:
我之前也出现了这个错误，然后我把openwrt的mac80211替换L大里面的编译成功了，但不知道会不会出现啥问题，目前运行半个月没出现问题。

iamggs, 2019-11-04:
推倒重来吧，我延续半年的二次编译也倒在了昨天，删除lede重新clone后花费3~4个小时重新编译成功。

czjesda, 2019-11-04:
make clean后在编译，我以前也是这个错误，用了make clean就好了

xinxijishuwyq, 2019-11-04:
> make clean后在编译，我以前也是这个错误，用了make clean就好了

我clean过了。。。还是不行。

czjesda, 2019-11-04:
> 
> 
> > make clean后在编译，我以前也是这个错误，用了make clean就好了
> 
> 我clean过了。。。还是不行。

你吧代码都删除了，重新重新git一下，结束后先用make clean，然后再编译

xinxijishuwyq, 2019-11-04:
> 我之前也出现了这个错误，然后我把openwrt的mac80211替换L大里面的编译成功了，但不知道会不会出现啥问题，目前运行半个月没出现问题。

666
用这个办法解决了

closed 2019-11-04 by xinxijishuwyq