coolsnowwolf/lede #1992: 新路由2 D1 OPENWRT 的TF卡驅動問題

hxhwing0522, 2019-11-04:
大佬  我想求助 新路由2 D1 OPENWRT 的TF卡驅動問題

編譯前已選上
blockd  automount
kmod-mmc  kmod-sdhci kmod-sdhci-mt7620
kmod-nls-cp437   kmod-nls-iso8859-1  kmod-nls-utf8

目前狀況:
已經能加載到TF卡 (FAT / EXT4)

問題:
第一次掛載TF卡(FAT32/EXT4)，都是能正常運作的。(aria2可以順暢下載到tf卡上)
但是一旦重啟，就會失去格式，無法使用。
> FAT: NTFS signature is missing.
> EXT4: 掛載點再找不到EXT4區
即便fdisk上都存在，但也再MOUNT不了。
如果用aria2下載到tf卡上，很快路由就會崩掉。

同樣的情況 在USB上就完全沒有出現。
不想浪費D1的TF卡槽

請問大佬 應該如何解決?
謝謝

supppig, 2019-11-05:
Extra packages 勾选 automount

hxhwing0522, 2019-11-05:
> Extra packages 勾选 automount

勾了
主要問題是重啟後就出現TF卡格式丟失  用不了

supppig, 2019-11-05:
> > Extra packages 勾选 automount
> 
> 勾了
> 主要問題是重啟後就出現TF卡格式丟失 用不了

那你参考下这个 https://blog.csdn.net/winux123/article/details/52081220
第一步不用做。
最后勾选automount就可以了。

我也是这个设备，我就是这么做的

supppig, 2019-11-05:
[1.默认备份，SD可用.zip](https://github.com/coolsnowwolf/lede/files/3808790/1.SD.zip)

@hxhwing0522 这个也给你做参考吧，是选了 newifi D1之后，再添加SD卡后的全新默认配置。

hxhwing0522, 2019-11-05:
> > > Extra packages 勾选 automount
> > 
> > 
> > 勾了
> > 主要問題是重啟後就出現TF卡格式丟失 用不了
> 
> 那你参考下这个 https://blog.csdn.net/winux123/article/details/52081220
> 第一步不用做。
> 最后勾选automount就可以了。
> 
> 我也是这个设备，我就是这么做的

你的是重啟路由也能正常掛載TF卡，和正常使用? 
你TF卡是什麼格式?

supppig, 2019-11-06:
> > > > Extra packages 勾选 automount
> > > 
> > > 
> > > 勾了
> > > 主要問題是重啟後就出現TF卡格式丟失 用不了
> > 
> > 
> > 那你参考下这个 https://blog.csdn.net/winux123/article/details/52081220
> > 第一步不用做。
> > 最后勾选automount就可以了。
> > 我也是这个设备，我就是这么做的
> 
> 你的是重啟路由也能正常掛載TF卡，和正常使用?
> 你TF卡是什麼格式?

正常啊，ext4格式的。

hxhwing0522, 2019-11-06:
> > > > > Extra packages 勾选 automount
> > > > 
> > > > 
> > > > 勾了
> > > > 主要問題是重啟後就出現TF卡格式丟失 用不了
> > > 
> > > 
> > > 那你参考下这个 https://blog.csdn.net/winux123/article/details/52081220
> > > 第一步不用做。
> > > 最后勾选automount就可以了。
> > > 我也是这个设备，我就是这么做的
> > 
> > 
> > 你的是重啟路由也能正常掛載TF卡，和正常使用?
> > 你TF卡是什麼格式?
> 
> 正常啊，ext4格式的。

我昨晚試了下，把ARIA2下載到TF卡里。
重啟也能識別到卡，不過容量1.8GB變成了TB了。
然後用不了。
我再看看怎弄

lovezm, 2019-11-11:
我想问一下我连fdisk都安装不上  如何解决啊。

lovezm, 2019-11-11:
能把你软件源发一下么

closed 2019-12-24 by coolsnowwolf