coolsnowwolf/lede #2003: r8152芯片组USB适配器间歇性断开连接

pV-nRT, 2019-11-05:
出现间歇性的LAN中断，内核日志输出如下：

[36266.457483] br-lan: port 1(eth0) entered disabled state
[36266.522508] r8152 2-1.1:1.0 eth0: carrier off
[36266.596951] atkbd serio0: Unknown key pressed (translated set 2, code 0x8d on isa0060/serio0).
[36266.597442] atkbd serio0: Use 'setkeycodes e00d <keycode>' to make it known.
[36272.661076] atkbd serio0: Unknown key pressed (translated set 2, code 0x8d on isa0060/serio0).
[36272.661611] atkbd serio0: Use 'setkeycodes e00d <keycode>' to make it known.
[36272.964106] atkbd serio0: Unknown key pressed (translated set 2, code 0x8d on isa0060/serio0).
[36272.964634] atkbd serio0: Use 'setkeycodes e00d <keycode>' to make it known.
[36276.506181] r8152 2-1.1:1.0 eth0: carrier on
[36276.506655] r8152 2-1.1:1.0 eth0: Promiscuous mode enabled
[36276.508172] br-lan: port 1(eth0) entered blocking state
[36276.508484] br-lan: port 1(eth0) entered forwarding state
[36277.624583] br-lan: port 1(eth0) entered disabled state
[36277.691608] r8152 2-1.1:1.0 eth0: carrier off
[36279.258318] r8152 2-1.1:1.0 eth0: carrier on
[36279.258672] r8152 2-1.1:1.0 eth0: Promiscuous mode enabled
[36279.260178] br-lan: port 1(eth0) entered blocking state
[36279.260488] br-lan: port 1(eth0) entered forwarding state

在以下网页中发现是r8153适配器的r8152驱动程序无法处理USB自动挂起导致的，但是该网页中提供的禁止该USB设备自动暂停方法是基于Ubuntu，在openwrt中不知道该如何实现，希望能提供一些思路。
https://askubuntu.com/questions/1044127/usb-ethernet-adapter-realtek-r8153-keeps-disconnecting

hcym, 2019-11-05:
这个只能用官方的源码，19.07稳稳的，8152 8153都不错，用大雕的直接没有网口

pV-nRT, 2019-11-24:
重新换了一条网线之后，已经连续一星期不再发生断开现象，看来网卡有问题还是得先检查接触问题。

closed 2019-11-24 by pV-nRT