coolsnowwolf/lede #2008: MT7621 WPA_SUPPLICANT编译失败

xinnanlyu, 2019-11-05:
近期的几个版本在勾选Network- wpa-supplicant后编译失败，提示找不到wpa-supplicant。
Target选择为Ramips - MT7621 - Netgear R6220.

日志中错误部分如下
echo "hostapd-common" >> /home/linux/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/hostapd.wpad-full-internal.install
make[3]: Leaving directory '/home/linux/lede/package/network/services/hostapd'
time: package/network/services/hostapd/wpad-full-internal/compile#1.49#0.88#2.27
make[3]: Entering directory '/home/linux/lede/package/network/services/hostapd'
mkdir -p /home/linux/lede/bin/targets/ramips/mt7621/packages /home/linux/lede/build_dir/target-mipsel_24kc_musl/hostapd-2019-08-08-ca8c2bd2/ipkg-mipsel_24kc/wpa-supplicant/CONTROL /home/linux/lede/staging_dir/target-mipsel_24kc_musl/pkginfo
install -d -m0755 /home/linux/lede/build_dir/target-mipsel_24kc_musl/hostapd-2019-08-08-ca8c2bd2/ipkg-mipsel_24kc/wpa-supplicant/usr/sbin
install -m0755 /home/linux/lede/build_dir/target-mipsel_24kc_musl/hostapd-2019-08-08-ca8c2bd2/wpa_supplicant/wpa_supplicant /home/linux/lede/build_dir/target-mipsel_24kc_musl/hostapd-2019-08-08-ca8c2bd2/ipkg-mipsel_24kc/wpa-supplicant/usr/sbin/
install: cannot stat '/home/linux/lede/build_dir/target-mipsel_24kc_musl/hostapd-2019-08-08-ca8c2bd2/wpa_supplicant/wpa_supplicant': No such file or directory
Makefile:609: recipe for target '/home/linux/lede/bin/packages/mipsel_24kc/base/wpa-supplicant_2019-08-08-ca8c2bd2-1_mipsel_24kc.ipk' failed
make[3]: *** [/home/linux/lede/bin/packages/mipsel_24kc/base/wpa-supplicant_2019-08-08-ca8c2bd2-1_mipsel_24kc.ipk] Error 1
make[3]: Leaving directory '/home/linux/lede/package/network/services/hostapd'
time: package/network/services/hostapd/supplicant-full-internal/compile#1.42#0.87#2.20
package/Makefile:107: recipe for target 'package/network/services/hostapd/compile' failed
make[2]: *** [package/network/services/hostapd/compile] Error 2
make[2]: Leaving directory '/home/linux/lede'
package/Makefile:103: recipe for target '/home/linux/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/linux/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/linux/lede'
/home/linux/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

Her0R0cket, 2019-11-06:
切换到wpad

xinnanlyu, 2019-11-06:
在桥接到802.1x无线热点时，需要同时使用wpad和wpa-supplicant。如果不勾选wpa-supplicant，在以wireless client模式配置时，会无法输入802.1x密码。
[image]

Her0R0cket, 2019-11-07:
加密不能选wpa2-psk?

xinnanlyu, 2019-11-07:
以上截图为9.7.6版本编译成功的结果，在勾选wpa-supplicant后，加密方式可以选择适合802.1X加密的WAP2-EAP类型。
而在较新的版本中，由于无法编译wap-supplicant，导致加密列表中不包含WPA2-EAP，无法连接802.1X网络。
我也尝试过编译官网原版openwrt，没有出现该问题出现。

Her0R0cket, 2019-11-08:
[image]
我没法帮你试mt7621的固件,我用的是x86的,但是我用wpad桥接时是可以选的,是不是之前wireless的配置冲突?

xinnanlyu, 2019-11-18:
经反复测试发现：wpad与wpa-supplicant产生冲突，导致无法安装wpa-supplicant。
由于该问题与本项目无关，因此关闭问题。
谢谢帮助

closed 2019-11-18 by xinnanlyu