coolsnowwolf/lede #2009: X64 mac80211编译错误

applepearq, 2019-11-06:
make[3]: Entering directory '/home/zh/lede/package/kernel/mac80211'
mkdir -p /home/zh/lede/bin/targets/x86/64/packages /home/zh/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.79-1/ipkg-x86_64/kmod-mac80211/CONTROL /home/zh/lede/staging_dir/target-x86_64_musl/pkginfo
ERROR: module '/home/zh/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.79-1/net/mac80211/mac80211.ko' is missing.
Makefile:561: recipe for target '/home/zh/lede/bin/targets/x86/64/packages/kmod-mac80211_4.19.81+4.19.79-1-1_x86_64.ipk' failed
make[3]: *** [/home/zh/lede/bin/targets/x86/64/packages/kmod-mac80211_4.19.81+4.19.79-1-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/zh/lede/package/kernel/mac80211'
time: package/kernel/mac80211/compile#0.18#0.04#0.20
package/Makefile:107: recipe for target 'package/kernel/mac80211/compile' failed
make[2]: *** [package/kernel/mac80211/compile] Error 2
make[2]: Leaving directory '/home/zh/lede'
package/Makefile:103: recipe for target '/home/zh/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/zh/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zh/lede'
/home/zh/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

MingmingJiang, 2019-11-06:
我也遇到了

Kennylaogu, 2019-11-06:
我只有我有这个错误。。。我都重新编译很多次了。。。

Kennylaogu, 2019-11-06:
而且把所有以来 mac80211 的组件去除 (如整个无线驱动)，然后清除 wpad 和 hostapd-common
是可以继续编下去，但到 package/install 的时候会提示 kmod 内核不符：

Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-fs-exfat:
 * 	kernel (= 4.19.81-1-76f1777cf11dcd17cf1d9fff7efa1755) * 
 * opkg_install_cmd: Cannot install package kmod-fs-exfat.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-button-hotplug:
 * 	kernel (= 4.19.81-1-76f1777cf11dcd17cf1d9fff7efa1755) * 
 * opkg_install_cmd: Cannot install package kmod-button-hotplug.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-cryptodev:
 * 	kernel (= 4.19.81-1-76f1777cf11dcd17cf1d9fff7efa1755) * 	kernel (= 4.19.81-1-76f1777cf11dcd17cf1d9fff7efa1755) * 	kernel (= 4.19.81-1-76f1777cf11dcd17cf1d9fff7efa1755) * 	kernel (= 4.19.81-1-76f1777cf11dcd17cf1d9fff7efa1755) * 	kernel (= 4.19.81-1-76f1777cf11dcd17cf1d9fff7efa1755) * 	kernel (= 4.19.81-1-76f1777cf11dcd17cf1d9fff7efa1755) * 
 * opkg_install_cmd: Cannot install package kmod-cryptodev.

bkye, 2019-11-06:
把openwrt官方源码的mac80211目录替换lean源码的mac80211目录即可编译成功

zhg-em, 2019-11-06:
x86_64 还是用4.9 的比较好，4.9是可以编译成功的，不需要替换！

applepearq, 2019-11-06:
> x86_64还是用4.9的比较好，4.9是可以​​编译成功的，不需要替换！

请问怎么改成4.9？

Kennylaogu, 2019-11-06:
> 把openwrt官方源码的mac80211目录替换lean源码的mac80211目录即可编译成功

亲测有用！多谢！

zhg-em, 2019-11-06:
> > x86_64还是用4.9的比较好，4.9是可以​​编译成功的，不需要替换！
> 
> 请问怎么改成4.9？

修改 target/linux/x86/Makefile

greennyreborn, 2019-11-06:
一样的问题，而且我看了其他已关闭的issue，删了重新clone 也一样

hyyz17200, 2019-11-06:
我也是这样，我暂时退老版本编译了

czjesda, 2019-11-18:
> 
> 
> > 把openwrt官方源码的mac80211目录替换lean源码的mac80211目录即可编译成功
> 
> 亲测有用！多谢！

你好，如何把openwrt的文件拷贝过去呢？

czjesda, 2019-11-18:
> 
> 
> 把openwrt官方源码的mac80211目录替换lean源码的mac80211目录即可编译成功

你好，请问一下如何把openwrt的文件拷贝过去呢？

kktt007, 2019-11-19:
> > 把openwrt官方源码的mac80211目录替换lean源码的mac80211目录即可编译成功
> 
> 你好，请问一下如何把openwrt的文件拷贝过去呢？

又是你,我们最近遇到的几个问题差不多.
拷贝就是你在线把那个文件夹下载下来,或者clone到本地,然后拷贝,感觉clone到本地方便,可能用到.

czjesda, 2019-11-20:
> 
> 
> > > 把openwrt官方源码的mac80211目录替换lean源码的mac80211目录即可编译成功
> > 
> > 
> > 你好，请问一下如何把openwrt的文件拷贝过去呢？
> 
> 又是你,我们最近遇到的几个问题差不多.
> 拷贝就是你在线把那个文件夹下载下来,或者clone到本地,然后拷贝,感觉clone到本地方便,可能用到.

哈哈，看来我们遇到的都是一样的问题。就是把lean和openwrt的代码都clone到本地。然后在吧openwrt的mac80211文件拷贝到lean的mac80211里面覆盖么？

bkye, 2019-11-20:
> > > > 把openwrt官方源码的mac80211目录替换lean源码的mac80211目录即可编译成功
> > > 
> > > 
> > > 你好，请问一下如何把openwrt的文件拷贝过去呢？
> > 
> > 
> > 又是你,我们最近遇到的几个问题差不多.
> > 拷贝就是你在线把那个文件夹下载下来,或者clone到本地,然后拷贝,感觉clone到本地方便,可能用到.
> 
> 哈哈，看来我们遇到的都是一样的问题。就是把lean和openwrt的代码都clone到本地。然后在吧openwrt的mac80211文件拷贝到lean的mac80211里面覆盖么？

可以clone，到本地替换，也可以把mac80211的驱动取消，因为我单纯编译用在软路由，所以所有无线驱动我都取消了。暂时没发现错误。

czjesda, 2019-11-20:
> 可以clone，到本地替换，也可以把mac80211的驱动取消，因为我单纯编译用在软路由，所以所有无线驱动我都取消了。暂时没发现错误。

你好，如何取消mac80211的驱动，我也是用在软路由上的，不需要无线驱动。

bkye, 2019-11-20:
> 你好，如何取消mac80211的驱动，我也是用在软路由上的，不需要无线驱动。

加QQ聊吧..843108617 这里聊太麻烦了.....

f3404923, 2019-11-28:
> > 把openwrt官方源码的mac80211目录替换lean源码的mac80211目录即可编译成功
> 
> 亲测有用！多谢！

你们幸运，我替换了也还是一样，mac80211 missing, 编译目标为x86

jzhou404, 2019-11-29:
这个方法管用， 我之前X86可以， 但X64编译不出来的问题， 完美解决了。谢谢

closed 2019-12-25 by coolsnowwolf