coolsnowwolf/lede #2014: 之前多次编译成功，但从10月27号至11月6号，每天至少编译一次，一直失败，不知道什么原因。

jy3344511, 2019-11-06:
设备：Nanopi NEO2

之前多次编译成功，但从10月27号至11月6号，每天至少编译一次，一直失败，不知道什么原因。中间机场换了3家，系统从ubuntu 14、18到Debian都重新安装过，仍然编译失败，肯定不是系统、机场的问题了，以下是报错代码，麻烦看看是怎么回事？最重要的是怎么解决，谢谢。

ssl/qsslsocket_openssl_symbols.cpp:1345:5: note: in expansion of macro 'RESOLVEFUNC'
     RESOLVEFUNC(SSL_set_alpn_protos)
     ^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:603:11: note: suggested alternative: 'SSL_set_alpn_protos'
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \
           ^
ssl/qsslsocket_openssl_symbols.cpp:1345:5: note: in expansion of macro 'RESOLVEFUNC'
     RESOLVEFUNC(SSL_set_alpn_protos)
     ^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:603:23: error: '_q_PTR_SSL_set_alpn_protos' was not declared in this scope
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \
                       ^
ssl/qsslsocket_openssl_symbols.cpp:1345:5: note: in expansion of macro 'RESOLVEFUNC'
     RESOLVEFUNC(SSL_set_alpn_protos)
     ^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:603:23: note: suggested alternative: 'SSL_set_alpn_protos'
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \
                       ^
ssl/qsslsocket_openssl_symbols.cpp:1345:5: note: in expansion of macro 'RESOLVEFUNC'
     RESOLVEFUNC(SSL_set_alpn_protos)
     ^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:603:11: error: '_q_SSL_CTX_set_alpn_select_cb' was not declared in this scope
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \
           ^
ssl/qsslsocket_openssl_symbols.cpp:1346:5: note: in expansion of macro 'RESOLVEFUNC'
     RESOLVEFUNC(SSL_CTX_set_alpn_select_cb)
     ^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:603:11: note: suggested alternative: 'SSL_CTX_set_alpn_select_cb'
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \
           ^
ssl/qsslsocket_openssl_symbols.cpp:1346:5: note: in expansion of macro 'RESOLVEFUNC'
     RESOLVEFUNC(SSL_CTX_set_alpn_select_cb)
     ^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:603:23: error: '_q_PTR_SSL_CTX_set_alpn_select_cb' was not declared in this scope
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \
                       ^
ssl/qsslsocket_openssl_symbols.cpp:1346:5: note: in expansion of macro 'RESOLVEFUNC'
     RESOLVEFUNC(SSL_CTX_set_alpn_select_cb)
     ^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:603:23: note: suggested alternative: 'SSL_CTX_set_alpn_select_cb'
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \
                       ^
ssl/qsslsocket_openssl_symbols.cpp:1346:5: note: in expansion of macro 'RESOLVEFUNC'
     RESOLVEFUNC(SSL_CTX_set_alpn_select_cb)
     ^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:603:11: error: '_q_SSL_get0_alpn_selected' was not declared in this scope
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \
           ^
ssl/qsslsocket_openssl_symbols.cpp:1347:5: note: in expansion of macro 'RESOLVEFUNC'
     RESOLVEFUNC(SSL_get0_alpn_selected)
     ^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:603:11: note: suggested alternative: 'SSL_get0_alpn_selected'
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \
           ^
ssl/qsslsocket_openssl_symbols.cpp:1347:5: note: in expansion of macro 'RESOLVEFUNC'
     RESOLVEFUNC(SSL_get0_alpn_selected)
     ^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:603:23: error: '_q_PTR_SSL_get0_alpn_selected' was not declared in this scope
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \
                       ^
ssl/qsslsocket_openssl_symbols.cpp:1347:5: note: in expansion of macro 'RESOLVEFUNC'
     RESOLVEFUNC(SSL_get0_alpn_selected)
     ^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:603:23: note: suggested alternative: 'SSL_get0_alpn_selected'
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \
                       ^
ssl/qsslsocket_openssl_symbols.cpp:1347:5: note: in expansion of macro 'RESOLVEFUNC'
     RESOLVEFUNC(SSL_get0_alpn_selected)
     ^~~~~~~~~~~
Makefile:26050: recipe for target '.obj/qsslsocket_openssl_symbols.o' failed
make[7]: *** [.obj/qsslsocket_openssl_symbols.o] Error 1
make[7]: Leaving directory '/home/q/lede/build_dir/target-aarch64_cortex-a53_musl/qt-everywhere-src-5.12.4/qtbase/src/network'
Makefile:272: recipe for target 'sub-network-make_first' failed
make[6]: *** [sub-network-make_first] Error 2
make[6]: Leaving directory '/home/q/lede/build_dir/target-aarch64_cortex-a53_musl/qt-everywhere-src-5.12.4/qtbase/src'
Makefile:50: recipe for target 'sub-src-make_first' failed
make[5]: *** [sub-src-make_first] Error 2
make[5]: Leaving directory '/home/q/lede/build_dir/target-aarch64_cortex-a53_musl/qt-everywhere-src-5.12.4/qtbase'
Makefile:70: recipe for target 'module-qtbase-make_first' failed
make[4]: *** [module-qtbase-make_first] Error 2
make[4]: Leaving directory '/home/q/lede/build_dir/target-aarch64_cortex-a53_musl/qt-everywhere-src-5.12.4'
Makefile:225: recipe for target '/home/q/lede/build_dir/target-aarch64_cortex-a53_musl/qt-everywhere-src-5.12.4/.built' failed
make[3]: *** [/home/q/lede/build_dir/target-aarch64_cortex-a53_musl/qt-everywhere-src-5.12.4/.built] Error 2
make[3]: Leaving directory '/home/q/lede/package/lean/qt5'
time: package/lean/qt5/compile#806.57#67.77#856.16
package/Makefile:107: recipe for target 'package/lean/qt5/compile' failed
make[2]: *** [package/lean/qt5/compile] Error 2
make[2]: Leaving directory '/home/q/lede'
package/Makefile:103: recipe for target '/home/q/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile' failed
make[1]: *** [/home/q/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/q/lede'
/home/q/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

hcym, 2019-11-06:
退回一个星期的吧，主要是内核换了419，我的x86也是死活过不去，用4.9内核正常编译

wxlost, 2019-11-06:
```
git reset --hard 61cfe1645c0680ba099627b07c8bed1a9e36f6f5
```
回退代码到内核升级前编译就无问题..升级就凉了....唉...
```
make -C "/home/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.81" ARCH="x86" CROSS_COMPILE="x86_64-openwrt-linux-musl-" M="/home/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5" CONFIG_EXFAT_FS=m modules
make[4]: Entering directory `/home/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.81'
make[5]: *** No rule to make target `/home/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/exfat_core.o', needed by `/home/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/exfat.o'.  Stop.
make[4]: *** [_module_/home/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5] Error 2
make[4]: Leaving directory `/home/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.81'
make[3]: *** [/home/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/.built] Error 2
make[3]: Leaving directory `/home/lede/lede/feeds/packages/kernel/exfat-nofuse'
time: package/feeds/packages/exfat-nofuse/compile#0.20#0.15#1.08
make[2]: *** [package/feeds/packages/exfat-nofuse/compile] Error 2
make[2]: Leaving directory `/home/lede/lede'
make[1]: *** [/home/lede/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/lede/lede'
make: *** [world] 错误 2
```

jy3344511, 2019-11-06:
> 退回一个星期的吧，主要是内核换了419，我的x86也是死活过不去，用4.9内核正常编译

具体怎么操作？ 内核是指openwrt的固件版本吗？这个回滚后，编译出来的固件，是老版本，还是最新的版本？

wenjn, 2019-11-06:
改target/linux/x86/Makefile这个文件，内核改成4.14或4.9都行

jy3344511, 2019-11-06:
> 改target/linux/x86/Makefile这个文件，内核改成4.14或4.9都行

改完后，还要执行git pull的操作吗？

jy3344511, 2019-11-06:
> 改target/linux/x86/Makefile这个文件，内核改成4.14或4.9都行

设备：Nanopi NEO2，不是x86的平台，是不是应该改另外目录的版本号？

wenjn, 2019-11-06:
> > 改target/linux/x86/Makefile这个文件，内核改成4.14或4.9都行
> 
> 设备：Nanopi NEO2，不是x86的平台，是不是应该改另外目录的版本号？

你这个问题好像这个是qt5和OpenSSL冲突了，昨天代码有更新qt5的东西，你有make clean过吗？

jy3344511, 2019-11-06:
> > > 改target/linux/x86/Makefile这个文件，内核改成4.14或4.9都行
> > 
> > 
> > 设备：Nanopi NEO2，不是x86的平台，是不是应该改另外目录的版本号？
> 
> 你这个问题好像这个是qt5和OpenSSL冲突了，昨天代码有更新qt5的东西，你有make clean过吗？

之前编译过的系统已经删除了，现在是全新安装的ubuntu，未编译过的系统，仍然无法编译成功。

jy3344511, 2019-11-06:
Makefile下显示为4.14，无法编译成功，如果改成4.9的话，需要执行git pull吗？

wenjn, 2019-11-06:
> Makefile下显示为4.14，无法编译成功，如果改成4.9的话，需要执行git pull吗？

你这个改内核没用，是qt5和OpenSSL冲突了。

jy3344511, 2019-11-06:
> > Makefile下显示为4.14，无法编译成功，如果改成4.9的话，需要执行git pull吗？
> 
> 你这个改内核没用，是qt5和OpenSSL冲突了。

那怎么搞呢？？？？

bin000000, 2019-11-06:
昨天更新的QT5一样不好使。取消全部QT5有关的选项就正常了。要么重新拉取源码编译，要么make clean后编译，但是都不要选择luci-app-qbittorrent这个插件就OK。

jy3344511, 2019-11-06:
> qbittorrent

qbittorrent这个插件是我用的最多的插件啊，不要怎么搞啊？

bin000000, 2019-11-06:
你这话问的，openssl默认不支持QT5，L大都放弃了，要么你是高手，自己改代码让支持，要么就不用，哪来这么多怎么搞。伸手党就不要这么多要求。不是还有docker

outemancut, 2019-11-07:
openssl用回1.0.2版本就是了 屁事就没那么多了

zxlhhyccc, 2019-11-07:
> openssl用回1.0.2版本就是了 屁事就没那么多了

应该不是冲突问题。。。。

zxlhhyccc, 2019-11-07:
> openssl用回1.0.2版本就是了 屁事就没那么多了

我openssl退回1.0.2版本也是一样，怀疑是libressl问题

outemancut, 2019-11-07:
本来就是openssl1.1.1版本跟qt5的冲突 退回1.0.2 make clean一遍就过来 至于你怀疑是哪个的问题 你有本事解决就自己解决好了 不用怀疑

zxlhhyccc, 2019-11-07:
> 本来就是openssl1.1.1版本跟qt5的冲突 退回1.0.2 make clean一遍就过来 至于你怀疑是哪个的问题 你有本事解决就自己解决好了 不用怀疑

感觉很牛逼的一直，说话真狂。。。。。

zxlhhyccc, 2019-11-07:
> 本来就是openssl1.1.1版本跟qt5的冲突 退回1.0.2 make clean一遍就过来 至于你怀疑是哪个的问题 你有本事解决就自己解决好了 不用怀疑

经测试，openssl退回到1.0.2q版本，同时做了make dirclean，qt5也是过不了。。。。。

jy3344511, 2019-11-07:
Nanopi NEO2也不支持Docker啊，Transmission这个下载种子，添加了Tracker后，下载速度都慢的要死，而Aria2编译完后，进入管理界面，添加种子文件后，所有的下载直接打把x，显示什么错误 16，没有一个种子可以下载的。。。。

nameless3721, 2019-11-09:
> Nanopi NEO2也不支持Docker啊，Transmission这个下载种子，添加了Tracker后，下载速度都慢的要死，而Aria2编译完后，进入管理界面，添加种子文件后，所有的下载直接打个x，显示什么错误 16，没有一个种子可以下载的。

[image]
[image]
支持的

closed 2019-12-24 by coolsnowwolf

norikohxb, 2020-02-15:
Include v2ray-plugin (Shadowsocks plugin)这个选项不可以加上去
加上去就无法编译通过

norikohxb, 2020-02-15:
我也是nano pi H5，编译的时候如果勾选Include v2ray-plugin (Shadowsocks plugin)就会出错。但是我又想要这个。