coolsnowwolf/lede #2016: [Feature Request] 服务器节点页面 node条目 增加“应用”按钮

breakersun, 2019-11-06:
痛点：
在服务器节点页面，测试得到所有node的延时后，需要手动记录下希望使用的节点名称；
然后切换到客户端页面，在主服务器下拉选单，选择希望使用的节点。

期望的修改：
在服务器节点页面，node列表中，每个node上增加一个"应用“，测试延时后，即时可以点击切换到想要的节点。

感谢Lean以及各位贡献者的付出。如果能帮忙修改一下最好了。
小白表示对LUCI的修改，摸不着门路。

再次感谢。

outemancut, 2019-11-07:
你想什么就满足你什么那就简单多了！你要是顺道发个PR多好

breakersun, 2019-11-08:
> 你想什么就满足你什么那就简单多了！你要是顺道发个PR多好

貌似要用到js，表示完全不具备前端知识，还在痛苦摸索中。
目前的思路：修改/usr/lib/lua/luci/view/cbi/tblsection.htm 在其中的htm部分增加一个button；同时追加一段js代码，调用shadowsocksr的change node方法（假如存在这个方法的话）。

然后目前死在第二步，加了一段抄来的js，执行后提示“$ is not defined”

我把把抄来的/usr/lib/lua/luci/view/cbi/tblsection.htm 的js部分贴在这里，希望大佬们给指点一下。小弟拜谢。

```
<script type="text/javascript">
    function apply_node(node){
            $("#cbi-apply-shadowsocksr1").show();
            $.get('<%=luci.dispatcher.build_url("admin", "services", "shadowsocksr","change")%>',
                    {
                        set:node
                    },
                    function (data, status) {
                        $("#cbi-apply-shadowsocksr1").hide();
                        
                        if (data.status ) {
                            var id = '#cbi-<%=self.config%>-'+node;
                            $(".cbi-section-table-row").removeClass("fast");
                            $(id).addClass("fast");
                        }
                    });
        }
</script>
```

breakersun, 2019-11-08:
经过一个上午的摸索，已经改了一个版本出来，可以activate node.
但是发现使用效果不太理想，仍需要再增加一些状态指示(指示切换过程，标识当前正在使用的节点)等一系列的动作。
issue先关闭了，先回去闭关修炼。

closed 2019-11-08 by breakersun