coolsnowwolf/lede #2018: Adbyby Plus+ 未运行

klever1988, 2019-11-06:
从[https://github.com/coolsnowwolf/lede/tree/2365f15b47c0c15732c81af54b148565ae05e9c0](url)编译的， 只选了ipv6helper和大雕套装，平台是nanopi neo2，刷了两次尝试了修改很多设置，都无法运行adbyby plus+

zxlhhyccc, 2019-11-07:
> 从[https://github.com/coolsnowwolf/lede/tree/2365f15b47c0c15732c81af54b148565ae05e9c0](url)编译的， 只选了ipv6helper和大雕套装，平台是nanopi neo2，刷了两次尝试了修改很多设置，都无法运行adbyby plus+

是不是选择v2ray不运行？如果是，那就是要换成其他的或自己单独编译一个v2ray二进制。。。

zxlhhyccc, 2019-11-07:
> 从[https://github.com/coolsnowwolf/lede/tree/2365f15b47c0c15732c81af54b148565ae05e9c0](url)编译的， 只选了ipv6helper和大雕套装，平台是nanopi neo2，刷了两次尝试了修改很多设置，都无法运行adbyby plus+

搞错，不好意思。

cnzgray, 2019-12-27:
在sh下根据/etc/init.d/adbyby的步骤进行跟踪了下，最后发现是`/usr/share/adbyby/adbyby`的编译版本问题。

运行后提示：
[image]

closed 2019-12-30 by coolsnowwolf