coolsnowwolf/lede #2020: 安装了编译后的openwrt，往里面托一个40M的文件显示没有剩余空间了

CXeon, 2019-11-07:
安装了编译后的openwrt，往里面托一个40M的文件显示没有剩余空间了，但是也刷过其它软路由没有这个问题

hcym, 2019-11-07:
分区挂载呗，我是怕麻烦

Her0R0cket, 2019-11-07:
带docker之后剩余空间就只有30m了,你不用docker的话可以把docker那些相关的包都取消,那样空间就够了

CXeon, 2019-11-07:
> 
> 
> 带docker之后剩余空间就只有30m了,你不用docker的话可以把docker那些相关的包都取消,那样空间就够了

尴尬，我需要在里面跑docker。我的情况是这样的，安装完openwrt之后，带的docker运行不起来，我下载了个docker要放到 /usr/bin目录下替换掉，然后就提示没有剩余空间了。挂载了个硬盘，但是docker不能放在其它地方吧

CXeon, 2019-11-07:
> 
> 
> 分区挂载呗，我是怕麻烦

挂载了一个硬盘，但是有些东西不能望着里面放

Her0R0cket, 2019-11-07:
运行不起来要找原因啊,大家都能运行的啊,再者docker也能放其他盘的,软链接到bin就可以

CXeon, 2019-11-07:
> 运行不起来要找原因啊,大家都能运行的啊,再者docker也能放其他盘的,软链接到bin就可以

我这里docker在openwrt上面运行报这个错,然后具体原因我还在查
root@OpenWrt:~# dockerd
WARN[2019-11-07T06:45:17.852696505Z] failed to load plugin io.containerd.snapshotter.v1.btrfs  error="path /opt/docker/containerd/daemon/io.containerd.snapshotter.v1.btrfs must be a btrfs filesystem to be used with the btrfs snapshotter"
WARN[2019-11-07T06:45:17.858913726Z] failed to load plugin io.containerd.snapshotter.v1.aufs  error="modprobe aufs failed: "": exit status 255"
WARN[2019-11-07T06:45:17.859257093Z] failed to load plugin io.containerd.snapshotter.v1.zfs  error="path /opt/docker/containerd/daemon/io.containerd.snapshotter.v1.zfs must be a zfs filesystem to be used with the zfs snapshotter"
WARN[2019-11-07T06:45:17.859290952Z] could not use snapshotter zfs in metadata plugin  error="path /opt/docker/containerd/daemon/io.containerd.snapshotter.v1.zfs must be a zfs filesystem to be used with the zfs snapshotter"
WARN[2019-11-07T06:45:17.859302660Z] could not use snapshotter btrfs in metadata plugin  error="path /opt/docker/containerd/daemon/io.containerd.snapshotter.v1.btrfs must be a btrfs filesystem to be used with the btrfs snapshotter"
WARN[2019-11-07T06:45:17.859313209Z] could not use snapshotter aufs in metadata plugin  error="modprobe aufs failed: "": exit status 255"
ERRO[2019-11-07T06:45:17.880126401Z] 'overlay2' is not supported over overlayfs    storage-driver=overlay2
ERRO[2019-11-07T06:45:17.886233966Z] AUFS was not found in /proc/filesystems       storage-driver=aufs
ERRO[2019-11-07T06:45:17.892289797Z] 'overlay' is not supported over overlayfs     storage-driver=overlay
ERRO[2019-11-07T06:45:17.892323666Z] Failed to built-in GetDriver graph devicemapper /opt/docker 
WARN[2019-11-07T06:45:17.898152606Z] Your kernel does not support cgroup blkio weight 
WARN[2019-11-07T06:45:17.898186779Z] Your kernel does not support cgroup blkio weight_device 
WARN[2019-11-07T06:45:17.899698230Z] Running modprobe nf_nat failed with message: `modprobe: unrecognized option: v`, error: exit status 255 
WARN[2019-11-07T06:45:17.900474014Z] Running modprobe xt_conntrack failed with message: `modprobe: unrecognized option: v`, error: exit status 255

CXeon, 2019-11-07:
> 
> 
> 运行不起来要找原因啊,大家都能运行的啊,再者docker也能放其他盘的,软链接到bin就可以

感谢。解决了，我把daemon.json的data-root先注释掉就好了。因为openwrt是刚安装的，daemon.json data-root是 /opt/docker 我先注释掉让docker使用默认的了

closed 2019-12-15 by coolsnowwolf