coolsnowwolf/lede #2025: x86最新版本，内核4.19编译出错，什么时候修复下

Subdue0, 2019-11-08:
x86最新版本，内核4.19编译出错，什么时候修复下

dolphinpaopao, 2019-11-08:
没有报错log如何定位问题，然后解决问题呢？编译报错可以首先尝试默认设置编译。最近频繁看到有人提交报错问题，我全新拉取最新代码，编译都正常无报错。系统ubuntu 14 x64 server virtualbox虚拟机。

Subdue0, 2019-11-08:
make[3]: *** [/home/zh/lede/bin/targets/x86/64/packages/kmod-mac80211_4.19.81+4.19.79-1-1_x86_64.ipk] Error 1

Subdue0, 2019-11-08:
最新的代码，直接编译，不出意外，肯定报错

dolphinpaopao, 2019-11-08:
[image]
刚刚拉取最新代码编译，正常通过。

Subdue0, 2019-11-08:
我昨晚编译的，可能更新过了吧，我今晚再试试

Subdue0, 2019-11-09:
mkdir -p /home/subdue/lede/bin/targets/x86/64/packages /home/subdue/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.79-1/ipkg-x86_64/kmod-mac80211/CONTROL /home/subdue/lede/staging_dir/target-x86_64_musl/pkginfo
ERROR: module '/home/subdue/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.79-1/net/mac80211/mac80211.ko' is missing.
Makefile:561: recipe for target '/home/subdue/lede/bin/targets/x86/64/packages/kmod-mac80211_4.19.81+4.19.79-1-1_x86_64.ipk' failed
make[3]: *** [/home/subdue/lede/bin/targets/x86/64/packages/kmod-mac80211_4.19.81+4.19.79-1-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/subdue/lede/package/kernel/mac80211'

Subdue0, 2019-11-09:
刚拉取最新代码编译的，还是那个问题

Subdue0, 2019-11-09:
ERROR: module '/home/subdue/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.79-1/net/mac80211/mac80211.ko' is missing.

zcluo, 2019-11-09:
灭有问题，
最新的内核
4.19.81

Subdue0, 2019-11-09:
我昨晚刚编译的

Subdue0, 2019-11-09:
而且不只是我一个人出现这样的问题，我看到好多个都是这样的问题了，都是kmod-mac80211_4.19.81+4.19.79-1-1_x86_64.ipk] Error 1

Subdue0, 2019-11-09:
老版本代码的不会出问题

d975xbx, 2019-11-09:
4.19的x64和d2都正常编译啊，昨天。

Subdue0, 2019-11-09:
我这个内核模块没有

star-o0O, 2019-11-09:
11月1号内核更新到4.19.81，你这出现4.19.79+4.19.81要么是自己改的，要么是没有make clean，建议删除整个lede文件夹重新开始。

Subdue0, 2019-11-09:
我这个是拉取的最新代码编译的，不用clean，不是旧的，也没改过内核，虚拟机环境都是新装的

star-o0O, 2019-11-09:
> 我这个是拉取的最新代码编译的，不用clean，不是旧的，也没改过内核，虚拟机环境都是新装的

那可能是虚拟机环境问题了，ubuntu19.10测试通过，看上面Ubuntu14也没问题。

Subdue0, 2019-11-09:
这个虚拟机，我以前编译过官方的openwrt和lean的openwrt，前天刚编译官方的openwrt也没问题

Subdue0, 2019-11-09:
https://github.com/coolsnowwolf/lede/issues/2009#issue-518175076

Subdue0, 2019-11-09:
我看了下，好多人都这样，问题可能不在我们本身

hcym, 2019-11-09:
改成4.9就好了，大雕忙着琢磨大事

Kinodi, 2019-11-09:
我这里x86-64 4.19 正常  正常通过

Kinodi, 2019-11-09:
试试  删库重新拉取码源  我每次都是全新编译  所以没见报错

Subdue0, 2019-11-09:
我刚用rm -f .config && rm -f .config.old把配置清掉了，直接编译默认的，可以编译成功，所以猜测，应该是插件冲突造成的

closed 2019-11-09 by Subdue0

kktt007, 2019-11-11:
我还是报错，，有人收把op的文件复制过来，还没试，不知道有什么影响。什么时候删库，跑路，我也就不用折腾了