coolsnowwolf/lede #2040: qBittorrent能否提供最新版本的编译。

jy3344511, 2019-11-09:
qBittorrent目前版本为4.1.7-1，能否提供最新的版本。

1.首先，很多人是无法使用Docker的。
2.我的设备：Nanopi NEO2，编译完Aria2后，进入管理界面下载种子文件，所有的下载直接打个x，显示什么错误 16，没有一个种子可以下载的。
3.编译完Transmission，可以正常使用，但是即便通过Transmission Remote GUI对种子文件添加Tracker，过了几分钟，下载速度也经常是0或几十KB的速度，而且要添加Tracker，需要把每个种子都添加一次。
4.同样的种子，用qBittorrent下载的话，几秒到十几秒，速度就上去了。添加Tracker也只需要添加一次就完了，以后所有的种子都不用再次添加。qBittorrent我最喜欢的一个功能是监控文件夹，只要把种子文件的下载地址，改成监控文件夹，下载种子后，qBittorrent自动就开始下载了。
5.所以，qBittorrent麻烦尽量保持更新，这个软件太好用了。

Kinodi, 2019-11-10:
又不是越新越好，能稳定运行不就行了吗

outemancut, 2019-11-10:
https://github.com/qbittorrent/qBittorrent.git
开源 你自己不会修改更新么？又懒又想新！

jy3344511, 2019-11-10:
> https://github.com/qbittorrent/qBittorrent.git
> 开源 你自己不会修改更新么？又懒又想新！

如果所有人都懂编程、都懂编译，就不会有人跑到这来提交问题了，编译不通过、报错，这么简单，你不会自己修改吗？？？？

outemancut, 2019-11-10:
PKG_VERSION:=4.1.9.1
PKG_SOURCE_VERSION:=18688c4bf50eb285d5ee1bdae771f301308059b31ac9f1b4aeff117ae7a12db2
就一个版本号一个SHA256都不会修改？

jy3344511, 2019-11-10:
> PKG_SOURCE_VERSION

不是吃这碗饭的，也没人教啊。

jy3344511, 2019-11-10:
> PKG_VERSION:=4.1.9.1
> PKG_SOURCE_VERSION:=18688c4bf50eb285d5ee1bdae771f301308059b31ac9f1b4aeff117ae7a12db2
> 就一个版本号一个SHA256都不会修改？

PKG_SOURCE_VERSION的版本号从哪看？

coolsnowwolf, 2019-11-10:
4.1.9.1 目前运行会崩溃，怀疑是 GCC 版本问题

once375ml, 2019-11-14:
@jy3344511 qBittorrent 我一直在用，Windows Linux macOS。看了下更新日志，4.17的后续版本，改进不大，到是4.20 值得期待，目前我在用4.20的beta2.

@coolsnowwolf git你的源码后，如果加入qBittorrent 会编译无法通过："qvncscreen.h:76:22: error: expected ';' at end of member declaration"，排查后是 QT 5.8.0的问题，可以打补丁，我直接下载好QT 5.9.0的源码放入目录，并改了makefile，顺利编译通过。能否确认bug？如是的话，可否更新下源码。

大部分人，包括用，用你的代码，看上就是能成功编译，不用操心代码，只要专注每个人自定义的软件。

多谢了，贡献了高质量的代码。

zxlhhyccc, 2019-11-14:
> @jy3344511 qBittorrent 我一直在用，Windows Linux macOS。看了下更新日志，4.17的后续版本，改进不大，到是4.20 值得期待，目前我在用4.20的beta2.
>
> @coolsnowwolf git你的源码后，如果加入qBittorrent 会编译无法通过："qvncscreen.h:76:22: error: expected ';' at end [...]

能否提供你的makefile？

once375ml, 2019-11-14:
你好，

源码在公司电脑上，这几天出差没法提供。

我和你说思路吧：

首先，反选qbittorrent，编译成功，目的是把正确的软件都编译出来，为后面排错节约时间，同时也减少后面的编译时间。

然后选中qbittorrent，然后必然会失败，提示就是我上面说的。

进入lede/package/lean/QT5。大约是这样，你大概找找。

进入后看到makefile，找到QT的下载链接，进入找到5.9.0，同文件夹下有个文本文件是SHA1和MD5.

改version什么的为5.9。改release为0，看下MD5，把MD5复制到makefile的MD5处，

例如，你想用5.9.1，那么version处改为5.9 release处改为1，查官网的压缩包md5后 对应改packageMD5=xxxxx, 还要改个文件后缀名。 5.8.0的打包为.gz.tar; 5.9.0为.gz.xr。保存后应该可以了。

但是我的经历是下载后说是没有hash工具，之类的，不管了，直接下载好5.9.0的压缩包，放进lede/dl目录，再编译一次就过了。

QT包400MB，编译器内下载也慢，几百KB/s，我是windows下用下载工具8MB/s下完复制进虚拟机的dl目录的。

hong0980, 2019-11-17:
试试下加强版
PKG_NAME:=qBittorrent
PKG_VERSION:=4.1.9.3
PKG_RELEASE=1

PKG_SOURCE_PROTO:=git
PKG_SOURCE_URL:=https://github.com/c0re100/qBittorrent-Enhanced-Edition.git
PKG_SOURCE_SUBDIR:=$(PKG_NAME)-$(PKG_VERSION)
PKG_SOURCE_VERSION:=c523f23fd980839cfd26145a7a762784f200056e
PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.gz

zanjie1999, 2019-11-28:
我这里也是"qvncscreen.h:76:22: error: expected ';' at end of member declaration"这样的问题，然后找到这里来了

once375ml, 2019-11-28:
@zanjie1999 QT官方确认5.8 bug，并给了补丁，5.9解决，我直接用的5.9，但是编译出来的qbittorrent，不运行webgui可以，运行了就挂。

coolsnowwolf, 2019-11-28:
已解决

closed 2019-11-28 by coolsnowwolf