coolsnowwolf/lede #2041: 这是个什么报错  大佬们帮忙看看

bin000000, 2019-11-10:
[image]

Kinodi, 2019-11-10:
同错
`rm -f tmp/as/vpnserver.a
ar r tmp/as/vpnserver.a tmp/objs/Mayaqua/Cfg.o tmp/objs/Mayaqua/Encrypt.o tmp/objs/Mayaqua/FileIO.o tmp/objs/Mayaqua/Internat.o tmp/objs/Mayaqua/Kernel.o tmp/objs/Mayaqua/Mayaqua.o tmp/objs/Mayaqua/Memory.o tmp/objs/Mayaqua/Microsoft.o tmp/objs/Mayaqua/Network.o tmp/objs/Mayaqua/Object.o tmp/objs/Mayaqua/OS.o tmp/objs/Mayaqua/Pack.o tmp/objs/Mayaqua/Secure.o tmp/objs/Mayaqua/Str.o tmp/objs/Mayaqua/Table.o tmp/objs/Mayaqua/TcpIp.o tmp/objs/Mayaqua/Tick64.o tmp/objs/Mayaqua/Tracking.o tmp/objs/Mayaqua/Unix.o tmp/objs/Mayaqua/Win32.o tmp/objs/Cedar/Account.o tmp/objs/Cedar/Admin.o tmp/objs/Cedar/AzureClient.o tmp/objs/Cedar/AzureServer.o tmp/objs/Cedar/Bridge.o tmp/objs/Cedar/BridgeUnix.o tmp/objs/Cedar/BridgeWin32.o tmp/objs/Cedar/Cedar.o tmp/objs/Cedar/CedarPch.o tmp/objs/Cedar/Client.o tmp/objs/Cedar/CM.o tmp/objs/Cedar/Command.o tmp/objs/Cedar/Connection.o tmp/objs/Cedar/Console.o tmp/objs/Cedar/Database.o tmp/objs/Cedar/DDNS.o tmp/objs/Cedar/EM.o tmp/objs/Cedar/EtherLog.o tmp/objs/Cedar/Hub.o tmp/objs/Cedar/Interop_OpenVPN.o tmp/objs/Cedar/Interop_SSTP.o tmp/objs/Cedar/IPsec.o tmp/objs/Cedar/IPsec_EtherIP.o tmp/objs/Cedar/IPsec_IKE.o tmp/objs/Cedar/IPsec_IkePacket.o tmp/objs/Cedar/IPsec_IPC.o tmp/objs/Cedar/IPsec_L2TP.o tmp/objs/Cedar/IPsec_PPP.o tmp/objs/Cedar/IPsec_Win7.o tmp/objs/Cedar/Layer3.o tmp/objs/Cedar/Link.o tmp/objs/Cedar/Listener.o tmp/objs/Cedar/Logging.o tmp/objs/Cedar/Nat.o tmp/objs/Cedar/NativeStack.o tmp/objs/Cedar/NM.o tmp/objs/Cedar/NullLan.o tmp/objs/Cedar/Protocol.o tmp/objs/Cedar/Radius.o tmp/objs/Cedar/Remote.o tmp/objs/Cedar/Sam.o tmp/objs/Cedar/SecureInfo.o tmp/objs/Cedar/SecureNAT.o tmp/objs/Cedar/SeLowUser.o tmp/objs/Cedar/Server.o tmp/objs/Cedar/Session.o tmp/objs/Cedar/SM.o tmp/objs/Cedar/SW.o tmp/objs/Cedar/UdpAccel.o tmp/objs/Cedar/UT.o tmp/objs/Cedar/VG.o tmp/objs/Cedar/Virtual.o tmp/objs/Cedar/VLan.o tmp/objs/Cedar/VLanUnix.o tmp/objs/Cedar/VLanWin32.o tmp/objs/Cedar/WaterMark.o tmp/objs/Cedar/WebUI.o tmp/objs/Cedar/WinUi.o tmp/objs/Cedar/Wpc.o tmp/objs/vpnserver.o
ar: creating tmp/as/vpnserver.a
ranlib tmp/as/vpnserver.a
make[4]: Warning: File `src/bin/BuiltHamcoreFiles/unix/hamcore.se2' has modification time 8.6e+04 s in the future
x86_64-openwrt-linux-musl-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -I/home/openwrt/lede/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/include -I/home/openwrt/lede/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/include -iremap/home/openwrt/lede/build_dir/target-x86_64_musl/softethervpn/v4.29-9680:v4.29-9680 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DSSL_OP_NO_SSLv3  -I/home/openwrt/lede/staging_dir/target-x86_64_musl/usr/include -I/home/openwrt/lede/staging_dir/target-x86_64_musl/include -I/home/openwrt/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/usr/include -I/home/openwrt/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/include/fortify -I/home/openwrt/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/include -I/home/openwrt/lede/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/include -I/home/openwrt/lede/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/include -L/home/openwrt/lede/staging_dir/target-x86_64_musl/usr/lib -L/home/openwrt/lede/staging_dir/target-x86_64_musl/lib -L/home/openwrt/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/usr/lib -L/home/openwrt/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib -L/home/openwrt/lede/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib -Wl,-rpath-link=/home/openwrt/lede/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib -L/home/openwrt/lede/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/lib -Wl,-rpath-link=/home/openwrt/lede/staging_dir/target-x86_64_musl/usr/lib/libintl-stub/lib -znow -zrelro -DNDEBUG -DVPN_SPEED -DUNIX -DUNIX_LINUX -DCPU_64 -D_REENTRANT -DREENTRANT -D_THREAD_SAFE -D_THREADSAFE -DTHREAD_SAFE -DTHREADSAFE -D_FILE_OFFSET_BITS=64 -I./src/ -I./src/Cedar/ -I./src/Mayaqua/ -O2 -fsigned-char tmp/objs/Mayaqua/Cfg.o tmp/objs/Mayaqua/Encrypt.o tmp/objs/Mayaqua/FileIO.o tmp/objs/Mayaqua/Internat.o tmp/objs/Mayaqua/Kernel.o tmp/objs/Mayaqua/Mayaqua.o tmp/objs/Mayaqua/Memory.o tmp/objs/Mayaqua/Microsoft.o tmp/objs/Mayaqua/Network.o tmp/objs/Mayaqua/Object.o tmp/objs/Mayaqua/OS.o tmp/objs/Mayaqua/Pack.o tmp/objs/Mayaqua/Secure.o tmp/objs/Mayaqua/Str.o tmp/objs/Mayaqua/Table.o tmp/objs/Mayaqua/TcpIp.o tmp/objs/Mayaqua/Tick64.o tmp/objs/Mayaqua/Tracking.o tmp/objs/Mayaqua/Unix.o tmp/objs/Mayaqua/Win32.o tmp/objs/Cedar/Account.o tmp/objs/Cedar/Admin.o tmp/objs/Cedar/AzureClient.o tmp/objs/Cedar/AzureServer.o tmp/objs/Cedar/Bridge.o tmp/objs/Cedar/BridgeUnix.o tmp/objs/Cedar/BridgeWin32.o tmp/objs/Cedar/Cedar.o tmp/objs/Cedar/CedarPch.o tmp/objs/Cedar/Client.o tmp/objs/Cedar/CM.o tmp/objs/Cedar/Command.o tmp/objs/Cedar/Connection.o tmp/objs/Cedar/Console.o tmp/objs/Cedar/Database.o tmp/objs/Cedar/DDNS.o tmp/objs/Cedar/EM.o tmp/objs/Cedar/EtherLog.o tmp/objs/Cedar/Hub.o tmp/objs/Cedar/Interop_OpenVPN.o tmp/objs/Cedar/Interop_SSTP.o tmp/objs/Cedar/IPsec.o tmp/objs/Cedar/IPsec_EtherIP.o tmp/objs/Cedar/IPsec_IKE.o tmp/objs/Cedar/IPsec_IkePacket.o tmp/objs/Cedar/IPsec_IPC.o tmp/objs/Cedar/IPsec_L2TP.o tmp/objs/Cedar/IPsec_PPP.o tmp/objs/Cedar/IPsec_Win7.o tmp/objs/Cedar/Layer3.o tmp/objs/Cedar/Link.o tmp/objs/Cedar/Listener.o tmp/objs/Cedar/Logging.o tmp/objs/Cedar/Nat.o tmp/objs/Cedar/NativeS [... cut ...]
make[3]: *** [/home/openwrt/lede/build_dir/target-x86_64_musl/softethervpn/v4.29-9680/.built] Error 2
make[3]: Leaving directory `/home/openwrt/lede/feeds/packages/net/softethervpn'
time: package/feeds/packages/softethervpn/compile#27.89#1.14#28.55
make[2]: *** [package/feeds/packages/softethervpn/compile] Error 2
make[2]: Leaving directory `/home/openwrt/lede'
make[1]: *** [/home/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/openwrt/lede'
make: *** [world] Error 2
`

outemancut, 2019-11-10:
rm -rf ./tmp && rm -rf .config
make clean
重新试试

Waterman1997, 2019-11-10:
> rm -rf ./tmp && rm -rf .config
> make clean
> 重新试试

同上
[image]

L-vivi, 2019-11-10:
同样的错误，今天更新源码后，一直报错，make clean，make dirclean，编译都会报错

Waterman1997, 2019-11-10:
怀疑还是openssl的问题，已回退98f345b17ce4440aa5b0c81cb2fde1281ea315a1
正在编译...

Waterman1997, 2019-11-10:
> 怀疑还是openssl的问题，已回退98f345b17ce4440aa5b0c81cb2fde1281ea315a1
> 正在编译...

回退后改了KERNEL_PATCHVER 终于好了

L-vivi, 2019-11-10:
> > 怀疑还是openssl的问题，已回退98f345b17ce4440aa5b0c81cb2fde1281ea315a1
> > 正在编译...
> 
> 回退后改了KERNEL_PATCHVER 终于好了

怎么整

Waterman1997, 2019-11-10:
回退98f345b17ce4440aa5b0c81cb2fde1281ea315a1
修改[https://github.com/coolsnowwolf/lede/issues/2009#issue-518175076](https://github.com/coolsnowwolf/lede/issues/2009#issue-518175076)

L-vivi, 2019-11-10:
今天全是报错的，哭了！

kktt007, 2019-11-11:
看到每天各种编译错误，看来要使用涡轮增雕法编译了

closed 2019-12-24 by coolsnowwolf