coolsnowwolf/lede #2042: x86_64去ssl编译qb成功并可以使用，但是种子链接如果是https无法下载，必须改成http或者下种子上传下载

zcluo, 2019-11-10:
修改方法：
package/lean/qt5/Makefile
39行修改如下：
  DEPENDS:=+libatomic +libstdcpp +zlib +icu @!LINUX_2_6
去掉118行：-openssl-runtime \
该处增加：
-no-ssl \
-no-openssl \

coolsnowwolf, 2019-11-10:
已经可以了

closed 2019-11-10 by coolsnowwolf

zcluo, 2019-11-10:
> 已经可以了

ssl/qsslsocket_openssl_symbols.cpp: In function 'bool q_resolveOpenSslSymbols()':
ssl/qsslsocket_openssl_symbols.cpp:465:11: error: '_q_SSL_set_alpn_protos' was not declared in this scope
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \
           ^
ssl/qsslsocket_openssl_symbols.cpp:1008:5: note: in expansion of macro 'RESOLVEFUNC'
     RESOLVEFUNC(SSL_set_alpn_protos)
     ^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:465:11: note: suggested alternative: 'SSL_set_alpn_protos'
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \
           ^
ssl/qsslsocket_openssl_symbols.cpp:1008:5: note: in expansion of macro 'RESOLVEFUNC'
     RESOLVEFUNC(SSL_set_alpn_protos)
     ^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:465:23: error: '_q_PTR_SSL_set_alpn_protos' was not declared in this scope
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \
                       ^
ssl/qsslsocket_openssl_symbols.cpp:1008:5: note: in expansion of macro 'RESOLVEFUNC'
     RESOLVEFUNC(SSL_set_alpn_protos)
     ^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:465:23: note: suggested alternative: 'SSL_set_alpn_protos'
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \
                       ^
ssl/qsslsocket_openssl_symbols.cpp:1008:5: note: in expansion of macro 'RESOLVEFUNC'
     RESOLVEFUNC(SSL_set_alpn_protos)
     ^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:465:11: error: '_q_SSL_CTX_set_alpn_select_cb' was not declared in this scope
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \
           ^
ssl/qsslsocket_openssl_symbols.cpp:1009:5: note: in expansion of macro 'RESOLVEFUNC'
     RESOLVEFUNC(SSL_CTX_set_alpn_select_cb)
     ^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:465:11: note: suggested alternative: 'SSL_CTX_set_alpn_select_cb'
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \
           ^
ssl/qsslsocket_openssl_symbols.cpp:1009:5: note: in expansion of macro 'RESOLVEFUNC'
     RESOLVEFUNC(SSL_CTX_set_alpn_select_cb)
     ^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:465:23: error: '_q_PTR_SSL_CTX_set_alpn_select_cb' was not declared in this scope
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \
                       ^
ssl/qsslsocket_openssl_symbols.cpp:1009:5: note: in expansion of macro 'RESOLVEFUNC'
     RESOLVEFUNC(SSL_CTX_set_alpn_select_cb)
     ^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:465:23: note: suggested alternative: 'SSL_CTX_set_alpn_select_cb'
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \
                       ^
ssl/qsslsocket_openssl_symbols.cpp:1009:5: note: in expansion of macro 'RESOLVEFUNC'
     RESOLVEFUNC(SSL_CTX_set_alpn_select_cb)
     ^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:465:11: error: '_q_SSL_get0_alpn_selected' was not declared in this scope
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \
           ^
ssl/qsslsocket_openssl_symbols.cpp:1010:5: note: in expansion of macro 'RESOLVEFUNC'
     RESOLVEFUNC(SSL_get0_alpn_selected)
     ^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:465:11: note: suggested alternative: 'SSL_get0_alpn_selected'
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \
           ^
ssl/qsslsocket_openssl_symbols.cpp:1010:5: note: in expansion of macro 'RESOLVEFUNC'
     RESOLVEFUNC(SSL_get0_alpn_selected)
     ^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:465:23: error: '_q_PTR_SSL_get0_alpn_selected' was not declared in this scope
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \
                       ^
ssl/qsslsocket_openssl_symbols.cpp:1010:5: note: in expansion of macro 'RESOLVEFUNC'
     RESOLVEFUNC(SSL_get0_alpn_selected)
     ^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:465:23: note: suggested alternative: 'SSL_get0_alpn_selected'
     if (!(_q_##func = _q_PTR_##func(libs.first->resolve(#func)))     \
                       ^
ssl/qsslsocket_openssl_symbols.cpp:1010:5: note: in expansion of macro 'RESOLVEFUNC'
     RESOLVEFUNC(SSL_get0_alpn_selected)
     ^~~~~~~~~~~
make[7]: *** [Makefile:25424: .obj/qsslsocket_openssl_symbols.o] Error 1
make[7]: Leaving directory '/root/lede/build_dir/target-x86_64_musl/qt-everywhere-opensource-src-5.8.0/qtbase/src/network'
make[6]: *** [Makefile:240: sub-network-make_first] Error 2
make[6]: Leaving directory '/root/lede/build_dir/target-x86_64_musl/qt-everywhere-opensource-src-5.8.0/qtbase/src'
make[5]: *** [Makefile:47: sub-src-make_first] Error 2
make[5]: Leaving directory '/root/lede/build_dir/target-x86_64_musl/qt-everywhere-opensource-src-5.8.0/qtbase'
make[4]: *** [Makefile:64: module-qtbase-make_first] Error 2
make[4]: Leaving directory '/root/lede/build_dir/target-x86_64_musl/qt-everywhere-opensource-src-5.8.0'
make[3]: *** [Makefile:229: /root/lede/build_dir/target-x86_64_musl/qt-everywhere-opensource-src-5.8.0/.built] Error 2
make[3]: Leaving directory '/root/lede/package/lean/qt5'
time: package/lean/qt5/compile#683.81#85.98#993.27
reset --hard到最新代码编译qb还是报ssl错误