coolsnowwolf/lede #2046: qb ssl还是不行

zcluo, 2019-11-10:
ssl/qsslsocket_openssl_symbols.cpp: In function 'bool q_resolveOpenSslSymbols()':
ssl/qsslsocket_openssl_symbols.cpp:465:11: error: '_q_SSL_set_alpn_protos' was not declared in this scope
if (!(q##func = q_PTR##func(libs.first->resolve(#func)))
^
ssl/qsslsocket_openssl_symbols.cpp:1008:5: note: in expansion of macro 'RESOLVEFUNC'
RESOLVEFUNC(SSL_set_alpn_protos)
^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:465:11: note: suggested alternative: 'SSL_set_alpn_protos'
if (!(q##func = q_PTR##func(libs.first->resolve(#func)))
^
ssl/qsslsocket_openssl_symbols.cpp:1008:5: note: in expansion of macro 'RESOLVEFUNC'
RESOLVEFUNC(SSL_set_alpn_protos)
^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:465:23: error: '_q_PTR_SSL_set_alpn_protos' was not declared in this scope
if (!(q##func = q_PTR##func(libs.first->resolve(#func)))
^
ssl/qsslsocket_openssl_symbols.cpp:1008:5: note: in expansion of macro 'RESOLVEFUNC'
RESOLVEFUNC(SSL_set_alpn_protos)
^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:465:23: note: suggested alternative: 'SSL_set_alpn_protos'
if (!(q##func = q_PTR##func(libs.first->resolve(#func)))
^
ssl/qsslsocket_openssl_symbols.cpp:1008:5: note: in expansion of macro 'RESOLVEFUNC'
RESOLVEFUNC(SSL_set_alpn_protos)
^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:465:11: error: '_q_SSL_CTX_set_alpn_select_cb' was not declared in this scope
if (!(q##func = q_PTR##func(libs.first->resolve(#func)))
^
ssl/qsslsocket_openssl_symbols.cpp:1009:5: note: in expansion of macro 'RESOLVEFUNC'
RESOLVEFUNC(SSL_CTX_set_alpn_select_cb)
^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:465:11: note: suggested alternative: 'SSL_CTX_set_alpn_select_cb'
if (!(q##func = q_PTR##func(libs.first->resolve(#func)))
^
ssl/qsslsocket_openssl_symbols.cpp:1009:5: note: in expansion of macro 'RESOLVEFUNC'
RESOLVEFUNC(SSL_CTX_set_alpn_select_cb)
^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:465:23: error: '_q_PTR_SSL_CTX_set_alpn_select_cb' was not declared in this scope
if (!(q##func = q_PTR##func(libs.first->resolve(#func)))
^
ssl/qsslsocket_openssl_symbols.cpp:1009:5: note: in expansion of macro 'RESOLVEFUNC'
RESOLVEFUNC(SSL_CTX_set_alpn_select_cb)
^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:465:23: note: suggested alternative: 'SSL_CTX_set_alpn_select_cb'
if (!(q##func = q_PTR##func(libs.first->resolve(#func)))
^
ssl/qsslsocket_openssl_symbols.cpp:1009:5: note: in expansion of macro 'RESOLVEFUNC'
RESOLVEFUNC(SSL_CTX_set_alpn_select_cb)
^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:465:11: error: '_q_SSL_get0_alpn_selected' was not declared in this scope
if (!(q##func = q_PTR##func(libs.first->resolve(#func)))
^
ssl/qsslsocket_openssl_symbols.cpp:1010:5: note: in expansion of macro 'RESOLVEFUNC'
RESOLVEFUNC(SSL_get0_alpn_selected)
^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:465:11: note: suggested alternative: 'SSL_get0_alpn_selected'
if (!(q##func = q_PTR##func(libs.first->resolve(#func)))
^
ssl/qsslsocket_openssl_symbols.cpp:1010:5: note: in expansion of macro 'RESOLVEFUNC'
RESOLVEFUNC(SSL_get0_alpn_selected)
^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:465:23: error: '_q_PTR_SSL_get0_alpn_selected' was not declared in this scope
if (!(q##func = q_PTR##func(libs.first->resolve(#func)))
^
ssl/qsslsocket_openssl_symbols.cpp:1010:5: note: in expansion of macro 'RESOLVEFUNC'
RESOLVEFUNC(SSL_get0_alpn_selected)
^~~~~~~~~~~
ssl/qsslsocket_openssl_symbols.cpp:465:23: note: suggested alternative: 'SSL_get0_alpn_selected'
if (!(q##func = q_PTR##func(libs.first->resolve(#func)))
^
ssl/qsslsocket_openssl_symbols.cpp:1010:5: note: in expansion of macro 'RESOLVEFUNC'
RESOLVEFUNC(SSL_get0_alpn_selected)
^~~~~~~~~~~
make[7]: *** [Makefile:25424: .obj/qsslsocket_openssl_symbols.o] Error 1
make[7]: Leaving directory '/root/lede/build_dir/target-x86_64_musl/qt-everywhere-opensource-src-5.8.0/qtbase/src/network'
make[6]: *** [Makefile:240: sub-network-make_first] Error 2
make[6]: Leaving directory '/root/lede/build_dir/target-x86_64_musl/qt-everywhere-opensource-src-5.8.0/qtbase/src'
make[5]: *** [Makefile:47: sub-src-make_first] Error 2
make[5]: Leaving directory '/root/lede/build_dir/target-x86_64_musl/qt-everywhere-opensource-src-5.8.0/qtbase'
make[4]: *** [Makefile:64: module-qtbase-make_first] Error 2
make[4]: Leaving directory '/root/lede/build_dir/target-x86_64_musl/qt-everywhere-opensource-src-5.8.0'
make[3]: *** [Makefile:229: /root/lede/build_dir/target-x86_64_musl/qt-everywhere-opensource-src-5.8.0/.built] Error 2
make[3]: Leaving directory '/root/lede/package/lean/qt5'
time: package/lean/qt5/compile#683.81#85.98#993.27
reset --hard到最新代码编译qb还是报ssl错误

nameless3721, 2019-11-10:
有make clean吗

zcluo, 2019-11-10:
> 有make clean吗

做了

doushang, 2019-11-29:
我遇到了和你一样的问题

EnnawYang, 2019-11-29:
我用github Actions可以顺利编译（ubuntu18.04），你可以用ubuntu18试试，ubuntu14环境下我也是编译失败

zcluo, 2019-11-30:
最新的代码用ubuntu 14 lts amd64的操作系统编译没问题

closed 2019-11-30 by zcluo

pauldeng, 2021-04-29:
遇到同样问题，提供一下解决方法，希望对后面的人有用。

这个问题应该是 QT5 和 OpenWRT OpenSSL 版本有关，QT5 对这个问题有[记录](https://bugreports.qt.io/browse/QTBUG-81762)。

只需按照这个 [commit](https://codereview.qt-project.org/gitweb?p=qt/qtbase.git;a=commitdiff;h=0cf046cc7cc84ba822f43183f162989b2c078913) 修改，就可以成功编译了。不需要降级到 Ubuntu 14。