coolsnowwolf/lede #2048: x86 64固件编译报错。python要求关闭ipv6？

ggllioio, 2019-11-10:
编译固件选中了ipv6helper，不知是否有关系。
现报错如下：
Fatal: You must get working getaddrinfo() function.
       or you can specify "--disable-ipv6".
Makefile:354: recipe for target '/home/star/lede/build_dir/hostpkg/Python-2.7.17/.configured' failed
make[3]: *** [/home/star/lede/build_dir/hostpkg/Python-2.7.17/.configured] Error 1
make[3]: Leaving directory '/home/star/lede/feeds/packages/lang/python/python'
time: package/feeds/packages/python/host-compile#14.71#3.75#22.21
package/Makefile:107: recipe for target 'package/feeds/packages/python/host/compile' failed
make[2]: *** [package/feeds/packages/python/host/compile] Error 2
make[2]: Leaving directory '/home/star/lede'
package/Makefile:103: recipe for target '/home/star/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/star/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/star/lede'
/home/star/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

AmadeusGhost, 2019-11-16:
是不是使用了proxychians-ng代理来编译，如果是的话建议用```proxychains make download V=s```来先下好dl包。

ggllioio, 2019-11-16:
> 是不是使用了proxychains-ng代理来编译，如果是的话建议先用`proxychains make download V=s`下好dl包，然后编译时别用proxychains。

编译时有开启代理。
但是编译之前已经用make download下载了全部包。
代理不支持ipv6.
我关闭代理编译试试。

ggllioio, 2019-11-24:
> 是不是使用了proxychains-ng代理来编译，如果是的话建议先用`proxychains make download V=s`下好dl包，然后编译时别用proxychains。

还真是代理的问题。
我用的deepin Linux。
之前一直编译不过开的是应用代理（终端）。
试了下开全局代理一次通过。
开心~

closed 2019-11-24 by ggllioio