coolsnowwolf/lede #2049: 昨天更新源码之后，一直报错

L-vivi, 2019-11-11:
/home/vivi/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/include/sys/poll.h:1:2: warning: #warning redirecting incorrect #include <sys/poll.h> to <poll.h> [-Wcpp]
 #warning redirecting incorrect #include <sys/poll.h> to <poll.h>
  ^~~~~~~
/home/vivi/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/7.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: tmp/objs/Mayaqua/Encrypt.o: in function `NewX509.part.22':
Encrypt.c:(.text+0x22ae): undefined reference to `X509_getm_notBefore'
/home/vivi/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/7.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: Encrypt.c:(.text+0x22c2): undefined reference to `X509_getm_notAfter'
/home/vivi/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/7.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: tmp/objs/Mayaqua/Encrypt.o: in function `NewRootX509':
Encrypt.c:(.text+0x260c): undefined reference to `X509_getm_notBefore'
/home/vivi/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/7.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: Encrypt.c:(.text+0x2623): undefined reference to `X509_getm_notAfter'
/home/vivi/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/7.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: tmp/objs/Mayaqua/Encrypt.o: in function `X509ToX':
Encrypt.c:(.text+0x4b94): undefined reference to `ASN1_STRING_get0_data'
/home/vivi/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/7.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: tmp/objs/Mayaqua/Network.o: in function `StartSSLEx':
Network.c:(.text+0xbaf0): undefined reference to `SSLv3_method'
collect2: error: ld returned 1 exit status
src/makefiles/linux_64bit.mak:355: recipe for target 'tmp/hamcorebuilder' failed
make[4]: *** [tmp/hamcorebuilder] Error 1
make[4]: Leaving directory '/home/vivi/lede/build_dir/target-x86_64_musl/softethervpn/v4.29-9680'
Makefile:194: recipe for target '/home/vivi/lede/build_dir/target-x86_64_musl/softethervpn/v4.29-9680/.built' failed
make[3]: *** [/home/vivi/lede/build_dir/target-x86_64_musl/softethervpn/v4.29-9680/.built] Error 2
make[3]: Leaving directory '/home/vivi/lede/feeds/packages/net/softethervpn'
time: package/feeds/packages/softethervpn/compile#0.22#0.06#0.27
package/Makefile:107: recipe for target 'package/feeds/packages/softethervpn/compile' failed
make[2]: *** [package/feeds/packages/softethervpn/compile] Error 2
make[2]: Leaving directory '/home/vivi/lede'
package/Makefile:103: recipe for target '/home/vivi/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/vivi/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/vivi/lede'
/home/vivi/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

lzx2009vb, 2019-11-11:
我也是这里报错

shenyz, 2019-11-11:
刚才 把内核改成4.9 ，选softethervpn5 就可以编译通过了

hereisderek, 2019-11-13:
#2056

dongfangy2011, 2019-11-14:
我也是更新到最新版本然后就。。。。。。。。。。。和上面一样了

dongfangy2011, 2019-11-15:
首先梯子必须好，然后运行rm -rf ./tmp&&rm -rf .config后就可以了，

L-vivi, 2019-11-15:
> 首先梯子必须好，然后运行rm -rf ./tmp&&rm -rf .config后就可以了，
找到问题，softethervpn不要选就过了

closed 2019-11-16 by L-vivi

czjesda, 2019-11-20:
还是mac80211报错，从openwrt里拷贝过来还是一样的报错，不知道是什么原因