coolsnowwolf/lede #2051: 在docker中安装了qBittorrent，一直显示等待，下载无速度。

jy3344511, 2019-11-11:
用的这个地址的镜像：https://hub.docker.com/r/johngong/qbittorrent

设备Nanopi NEO2，在docker中安装了qbittorrent:qee_4.1.9.1_arm64v8，进入web的管理界面，添加种子文件后，任何的种子文件，都是一直显示等待，下载无速度。

以下是创建时用的代码，麻烦有知道的人帮忙看看，是哪里设置的不对，谢谢。

一：
docker pull johngong/qbittorrent:qee_4.1.9.1_arm64v8

二：
docker create  \
    --name=qbittorrent  \
    -e WEBUIPORT=8989  \
    -p 6881:6881  \
    -p 6881:6881/udp  \
    -p 8989:8989  \
    -v /mnt/mmcblk0p3:/config  \                  
    -v /mnt/sda1:/Downloads  \                       
    --restart unless-stopped  \
    johngong/qbittorrent:qee_4.1.9.1_arm64v8

备注： /mnt/mmcblk0p3是我的TF卡上，在刷完openwrt后，用DiskGenius手动分的第3个分区，ext4格式。
           /mnt/sda1是挂载的移动硬盘，用DiskGenius手动分的区，ext4格式。

Her0R0cket, 2019-11-12:
可能是文件夹权限的问题

jy3344511, 2019-11-12:
> 可能是文件夹权限的问题

之前也试过在代码中添加下面两项，不知道添加的对不对，但是下载一样一直显示等待，没速度。
-e PUID=0 \
-e PGID=0 \

nameless3721, 2019-11-14:
你看dht节点有没有东西 没的话代表没网络
我的nanopi要host模式才有网络的

jy3344511, 2019-11-14:
> host模式

host模式怎么设置的？还要做端口映射吗？

nameless3721, 2019-11-14:
就network的bridge换host模式

jy3344511, 2019-11-14:
> 就qbt network的bridge换host模式

能具体点吗？docker下Network里面有3个网络，都修改不了啊。

用docker run --add-host=johngong/qbittorrent:qee_4.1.9.1_arm64v8:192.168.0.130，又提示IP错误，在哪弄啊？

nameless3721, 2019-11-14:
[image]
点qbt的名字就有个更改配置的按钮
[image]
有个network按钮改成host就是了 因为已经卸了没图大概这个位置吧  如果已经加入其他网络可以尝试先退出或者停后再改 其实现在makeclean后已经能正常编译qbt了 因为docher和ssr 有点冲突就没用了

jy3344511, 2019-11-14:
> [image]
> 点qbt的名字就有个更改配置的按钮
> [image]
> 有个network按钮改成host就是了 因为已经卸了没图大概这个位置吧 如果已经加入其他网络可以尝试先退出或者停后再改 其实现在makeclean后已经能正常编译qbt了 因为docher和ssr 有点冲突就没用了

closed 2019-11-14 by jy3344511

jy3344511, 2019-11-15:
> > [image]
> > 点qbt的名字就有个更改配置的按钮
> > [image]
> > 有个network按钮改成host就是了 因为已经卸了没图大概这个位置吧 如果已经加入其他网络可以尝试先退出或者停后再改 其实现在makeclean后已经能正常编译qbt了 因为docher和ssr 有点冲突就没用了

可以下载了，但是下载后的文件，找不到啊，用-v /mnt/sda1:/Downloads \创建的容器，但在/mnt/sda1下面，啥都没有啊。

molodye, 2023-11-16:
网络改成 host模式。其次把你tcp udp端口号改成 16881 ，6881太小了 容易被pt官网禁止。