coolsnowwolf/lede #2052: ShadowsocksR Plus+ NOT RUNNING

daqinKnight, 2019-11-11:
１、重新编译后，ShadowsocksR Plus+ NOT RUNNING　一直报ShadowsocksR Plus+ NOT RUNNING
[image]

２、主机型号	NETGEAR WNR2200　架构	Atheros AR7241 rev 1

３、服务器情况：
[image]
４、系统状态情况
[image]
５、日志情况
[image]

６、大神，这个会是什么问题？该如何处理呢？

hgh188, 2019-11-11:
[image]
[image]
一切正常 最新源码

coolsnowwolf, 2019-11-12:
luci 版本不对

daqinKnight, 2019-11-13:
大神在看一下什么问题
1、重新编译一下（主机型号 NETGEAR WNR2200　架构 Atheros AR7241 rev 1），安装报错如下：
root@OpenWrt:/tmp# opkg install luci-app-ssr-plus_1-125_all.ipk 
Installing luci-app-ssr-plus (1-125) to root...
Configuring luci-app-ssr-plus.
uci: Entry not found
uci: Entry not found
uci: Parse error (invalid command) at line 1, byte 0
uci: Entry not found
Collected errors:
 * resolve_conffiles: Existing conffile /etc/dnsmasq.ssr/gfw_list.conf is different from the conffile in the new package. The new conffile will be placed at /etc/dnsmasq.ssr/gfw_list.conf-opkg.
2、系统还是不能运行
[image]

[image]

daqinKnight, 2019-11-15:
已经OK ，重新编译后，刷入自己编译后的openwrt 
[image]

daqinKnight, 2019-11-17:
稳定运行2天后出现这个问题？什么情况？
[image]

closed 2019-12-24 by coolsnowwolf