coolsnowwolf/lede #2056: Ubuntu18.04LTS系统第一次编译，报错如下，请指点

GitMasterArm, 2019-11-12:
ERROR: module '/home/jry/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-4.19.79-1/net/mac80211/mac80211.ko' is missing.
Makefile:561: recipe for target '/home/jry/lede/bin/targets/x86/64/packages/kmod-mac80211_4.19.81+4.19.79-1-1_x86_64.ipk' failed
make[3]: *** [/home/jry/lede/bin/targets/x86/64/packages/kmod-mac80211_4.19.81+4.19.79-1-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/jry/lede/package/kernel/mac80211'
time: package/kernel/mac80211/compile#18.87#3.06#422.77
package/Makefile:107: recipe for target 'package/kernel/mac80211/compile' failed
make[2]: *** [package/kernel/mac80211/compile] Error 2
make[2]: Leaving directory '/home/jry/lede'
package/Makefile:103: recipe for target '/home/jry/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/jry/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jry/lede'
/home/jry/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

GitMasterArm, 2019-11-12:
[image]

ksong008, 2019-11-12:
先默认跑一次，再添加删除插件

Albert-Zhan, 2019-11-12:
x86-64编译这个版本需要换内核，内核换成4.9就可以了，跟插件无关的。

Albert-Zhan, 2019-11-12:
[image]

d975xbx, 2019-11-12:
x64一直都是4.19编译的，和选那个内核没什么关系

GitMasterArm, 2019-11-12:
是下面动作吗：
1.git clone https://github.com/coolsnowwolf/lede 命令下载好源代码
2.通过WinSCP找到Makefile文件，修改保存
3.再进入PuTTY，执行./scripts/feeds update -a以后的命令吗？

Albert-Zhan, 2019-11-12:
已经有代码了

Albert-Zhan, 2019-11-12:
已经有代码了

Albert-Zhan, 2019-11-12:
已经编译过了先

greennyreborn, 2019-11-12:
> x64一直都是4.19编译的，和选那个内核没什么关系.
> 默认编译一遍就好

是说 make menuconfig 后不改参数先编译一遍，然后再重新再改插件，参数就行了？

Albert-Zhan, 2019-11-12:
> 是下面的动作吗：
> 1.git clone https://github.com/coolsnowwolf/lede命令下载好源代码
> 2.通过WinSCP找到Makefile文件，修改保存
> 3.再进入PuTTY，执行。/scripts/feedsupdate -a以后的命令吗？

已经编译过的:
rm -rf ./tmp
rm -rf ./staging_dir
rm -rf ./build_dir
rm -rf ./bin
然后修改Makefile，在编译就可以。

GitMasterArm, 2019-11-12:
https://github.com/coolsnowwolf/lede/issues/2056#issuecomment-552712540
按你的办法再试一次，完成后汇报哈

GitMasterArm, 2019-11-12:
> 已经编译过的:
> rm -rf ./tmp
> rm -rf ./staging_dir
> rm -rf ./build_dir
> rm -rf ./bin
> 然后修改Makefile，在编译就可以。

按你的办法试一次，完成后汇报哈

GitMasterArm, 2019-11-12:
> 按你的办法试一次，完成后汇报哈

这次报错是
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://downloads.openwrt.org/sources/shadowsocks-libev-3.3.1.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:02 --:--:--     0curl: (7) Failed to connect to downloads.openwrt.org port 443: Network is unreachable
Download failed.
我能自己下载shadowsocks-libev-3.3.1.tar.gz，拷贝到dl目录下吗？

Albert-Zhan, 2019-11-12:
> 这次报错是
>
> * curl -f --connect-timeout 20 --retry 5 --location --insecure https://downloads.openwrt.org/sources/shadowsocks-libev-3.3.1.tar.gz
>   % Total    % Received % Xferd  Average Speed   Time [...]

可以的，然后在执行，make -j1 V=s

GitMasterArm, 2019-11-12:
> make -j1 V=s
成了，谢谢！！
[image]

GitMasterArm, 2019-11-12:
> 可以的，然后在执行，make -j1 V=s

[image]
成了，谢谢！！！

Albert-Zhan, 2019-11-12:
> [image]
> 成了，谢谢！！！

看了一下今天L大的代码貌似已经修复了4.19内核编译mac80211错误的问题了，你可以更新最新代码编译4.19的试下。

GitMasterArm, 2019-11-12:
> 看了一下今天L大的代码貌似已经修复了4.19内核编译mac80211错误的问题了，你可以更新最新代码编译4.19的试下。

好，我试一下。

GitMasterArm, 2019-11-13:
> x64一直都是4.19编译的，和选那个内核没什么关系.
> 默认编译一遍就好

默认编译一次成功了，我要再添加或删除组件怎么做，请高手指点一下呀

TerryWang159, 2019-11-13:
> > x64一直都是4.19编译的，和选那个内核没什么关系.
> > 默认编译一遍就好
> 
> 默认编译一次成功了，我要再添加或删除组件怎么做，请高手指点一下呀

1.  命令  make menuconfig    
2  选择你要的插件或者组件， 
3 网络如果不好的话，请预下载一下文件（在dl 文件夹下），命令 make download V=s ,失败的文件手动下载 放到dl文件下 。 如果你的网络足够好，也可以忽略这一步直接进入第四步.
4.  命令 make -j1 V=s

iamggs, 2019-12-05:
经典mac80211错误，解决办法看：
https://github.com/coolsnowwolf/lede/issues/2254

closed 2019-12-24 by coolsnowwolf