coolsnowwolf/lede #2061: NPS运行一段时间，会自动停止，需要重新启动

smsource, 2019-11-12:


njones5011, 2019-11-14:
出现同样的情况，建议增加断线重连的选项：auto_reconnection=true

Thu Nov 14 14:40:04 2019 daemon.info npc[30882]: 2019/11/14 06:40:04.516 [1;31m[E][0m [client.go:71]  Accept server data error read tcp 192.168.2.2:41044->X.X.X.X:8024: read: connection reset by peer, end this service
Thu Nov 14 14:40:04 2019 daemon.info npc[30882]: 2019/11/14 06:40:04.517 [1;33m[W][0m [client.go:144]  accpet error,the conn has closed
Thu Nov 14 14:40:15 2019 daemon.info npc[20730]: 2019/11/14 06:40:15.067 [1;31m[E][0m [control.go:111]  EOF
Thu Nov 14 14:40:25 2019 daemon.info npc[20769]: 2019/11/14 06:40:25.572 [1;31m[E][0m [control.go:111]  EOF
Thu Nov 14 14:40:36 2019 daemon.info npc[20797]: 2019/11/14 06:40:36.067 [1;31m[E][0m [control.go:111]  EOF
Thu Nov 14 14:40:46 2019 daemon.info npc[20825]: 2019/11/14 06:40:46.565 [1;31m[E][0m [control.go:111]  EOF
Thu Nov 14 14:40:57 2019 daemon.info npc[20868]: 2019/11/14 06:40:57.014 [1;31m[E][0m [control.go:111]  EOF
Thu Nov 14 14:41:07 2019 daemon.info npc[20896]: 2019/11/14 06:41:07.433 [1;31m[E][0m [control.go:111]  EOF
Thu Nov 14 14:41:07 2019 daemon.info procd: Instance nps::instance1 s in a crash loop 6 crashes, 5 seconds since last crash

smsource, 2019-11-18:
> 出现同样的情况，建议增加断线重连的选项：auto_reconnection=true
>
> Thu Nov 14 14:40:04 2019 daemon.info npc[30882]: 2019/11/14 06:40:04.516 [1;31m[E][0m [client.go:71] Accept server data error read tcp [...]

修改那个文件？op重启后，该修改是否仍然有效？

closed 2019-12-24 by coolsnowwolf